ev3dev/ev3dev #828: cannot use static ip

urkarthiksg, 2017-01-29:
i am using a MAC OS X 10.11 with EV3DEV via bluetooth. I go the latest ev3dev

i am trying to use a static ip to be able to connect. so i go into ipv4 config and use the OS X defaults. it comes up with 
192.168.2.3, 
255.255.255.0, 
192.169.2.1

the DNS also has the 2.1

I am able to connect to the bluetooth on MAC (I restarted mac, ev3dev, etc)... but it seems to only connect if i use DHCP. can someone help

dlech, 2017-01-29:
How did you configure the Mac?

urkarthiksg, 2017-01-29:
Hi David:

Thanks for your help. There was a bluetooth pan already existing. So.did not change anything there. Should I need to do something? 

Karthik

dlech, 2017-01-29:
Not that I know of. I just tried using the OS X defaults to connect to my Mac and it worked as expected. I'm not sure what you might be doing differently. Are you enabling Internet Sharing on your Mac?

urkarthiksg, 2017-01-29:
i did not. i can try that.

The issue is if i use OS X defaults, i do connect on bluetooth. but i just cannot open a ssh or even ping to the IP displayed on the brick.

BTW when i flashed it, i had to comment out a line in the image for MAC OS X 10.11. hope thats not causing anything funny.

urkarthiksg, 2017-01-29:
okay seems to be working now! tried twice and both cases i get the same brick IP… i now just use ev3dev.local instead of using a IP. 

its interesting, i am still using the DHCP option instead of the OSX defaults…

closed 2017-03-10 by dlech