ev3dev/ev3dev #829: Fatcatlab EVB 1.0 doesn't boot

Bujtar, 2017-02-01:
I have an Fatcatlab EVB 1.0 card with BBB.

I downloaded the EVB image (ev3dev-jessie-bone-generic-2016-12-21.img) copied it onto SD card
with (sudo dd bs=1M if=ev3dev-jessie-bone-generic-2016-12-21.img of=/dev/mmcblk0
) and uncommented in uEnv.txt file the  "cape=evb" and then I tried to boot from SD card
When I powered the board for a second all 4 LEDS turned on, and then D3 and D2 remained on. No booting logo on screen, nothing happens.

Same situation with other images:
snapshot-ev3dev-jessie-bone-generic-2017-01-25.img
nightly-ev3dev-jessie-bone-generic-2016-10-16.img

When I removed the EVB board, the BBB seemed to boot, at least the LEDs started to flash like during normal boot.

I tried Fatcatlab's image also, this boots correctly.
Any help appreciated !

Janos

dlech, 2017-02-01:
What is the exact model of your BeagleBone and how long have you had it?

I have used the ev3dev-jessie-bone-generic-2016-12-21 image, so I know that it is a working image. Snapshots, on the other hand, are not usually tested.

Bujtar, 2017-02-02:
Hi David,

Thanks for the quick reply.
You are right the issue was BBB card itself. As far as I can tell this card came from China (Banggood), although there is "element14" text  but no other information about type/model. The TI chip is AM3358BZC100.

I had another BBB with Beaglebone sign on pcb, it works perfectly with the latest snapshot image too.
If it would help I can try to gather more information about not working BB, just tell me how.

Thanks for your help!

dlech, 2017-02-02:
If the not working one is older, it could be that the bootloader needs to be updated. I can't find any good info on this at the moment, but there are some hints about it [on this page](https://eewiki.net/display/linuxonarm/BeagleBone+Black#BeagleBoneBlack-SetupmicroSDcard). Somewhere there is a "flasher" image floating around that updates this for you.

You could also try adding the "Dealing with old Bootloader in eMMC" from the link above to uEnv.txt, but I kind of remember that when I tried it before, it wasn't quite right, but it might work.

dlech, 2017-08-12:
Is this still a problem? Can this issue be closed?

closed 2017-10-03 by dlech