ev3dev/ev3dev #830: Getting mindsensor serial console adapter working

pipothy, 2017-02-03:
I'm trying to get the mindsensor serial console adapter working. I read the other thread but that seemed to be more of a hardware issue.

I was looking at the following guide and couldn't get it working with the original EV3Sources code:
http://icube-avr.unistra.fr/en/index.php/Hacking_the_Lego_EV3#Summary

So, I thought I'd try getting it working with EV3Dev's code. According to the previous thread, the serial console should work with the standard EV3Dev code and require no modification (is that correct?)

I followed up to step 4 of the getting started guide for EV3Dev, then simply plugged my serial console adapter into S1 and to my PC via USB. The serial console adapter is installed correctly, according to the instructions. I then opened up PuTTy, started a serial connection (Baudrate 115200) and got nothing on the screen when the robot was booting or when the robot had past boot.

Any idea why this might not be working? Anything obvious I've missed? Sorry, I am not that technically experienced!

bmegli, 2017-02-03:
@pipothy serial console on ttyS1 (port 1) was disabled, it occasionally led to boot problems with UART (serial) sensors plugged when booting. It's now set to debug to screen.

See #623 if you are interested.

You can re-enable serial console on ttyS1 (input port 1) by editting:
`/usr/share/flash-kernel/bootscript/bootscr.ev3` or probably `cmdline.txt` on boot partition
 
Replace 
```
console=${console}
```
with
```
console=ttyS1
```

You can also specify parameters, like `console=ttyS1, some_baudrate`. Note that if you use different baudrate from putty side and EV3 you may get garbage. I don't remember which was the default (9600? 115200?)
 
You can use both ttyS1 (port 1) and tty0 (screen) at the same time. See [here](https://kernel.org/doc/html/latest/admin-guide/serial-console.html) if you are interested.

Keep in mind that with serial console enabled on ttyS1 (input port 1) and UART sensor plugged on port 1 you may experience occasional boot failures.

Also changes in `bootscr.ev3` are updated when flash-kernel package is updated so the change there is not permanent (but will last as long as you don't update the package)

pipothy, 2017-02-03:
Hi,

Thanks, immensely helpful. Pardon my incompetence, but hit an issue. SSH'd in and found the file, edited it with nano but when I tried to exit it wouldn't save and said permission denied. Have I missed something?

P.S. it's also currently set to console=tty1, so I assume tty1 is the screen? So I put console=ttyS1,115200 console=tty1 which I assume should give output to both?

Thanks for your help!

WasabiFan, 2017-02-03:
[image]
_https://www.xkcd.com/149/_

You need to run nano with `sudo` (`sudo nano <file>`) so that you have permission to edit that file.

Also, I believe you can make this change using `ev3dev-config` without editing the file manually. Run that command and use the menu that appears in the terminal to enable the serial console.

pipothy, 2017-02-03:
Thanks WasabiFan, sorry, inexperienced user here. I can't seem to find a command called ev3dev-config under help. I'll edit it with nano for now though :)

Update: Amended the line in the file, restarted the EV3. Then connected it with the cable from S1 to the serial adapter to USB in my laptop. I went to putty, connected over serial (com3, rate of 115200) and just got a blank terminal so it isn't working. I checked the comport is right, that the file saved, I even tried changing the file to say tty0 instead of tty1 for the second console. Any idea why it might not be working?

Thanks again guys!

[image]

Attached image of the code in bootscr.ev3 in case there's anything glaringly wrong.

dlech, 2017-02-03:
> I'm trying to get the mindsensor serial console adapter working.

What exactly are you trying to accomplish? Do you want to log kernel messages? Do you want a login prompt? Do you want to do something else?

> by editting: /usr/share/flash-kernel/bootscript/bootscr.ev3

This is bad advice. All files in `/usr/` on debian should be treated as read-only because they are owned by packages. If you change this file, any changes will be written over when the `flash-kernel` package is upgraded. The correct place to make changes to the kernel command line is in `/etc/default/flash-kernel`.

But, as @WasabiFan said, you should run `sudo ev3dev-config` and it will automatically change all of the files needed. Before you do this, please answer the questions about what exactly you are hoping to accomplish.

pipothy, 2017-02-03:
Embarrassingly, you've now made me realise I don't need the serial connection!

The problem was I needed to be able to run the runtime command so I could get the computational load, which I can get without the serial connection. I was stuck in the mindset of the original ev3sources code where you can't easily run unix commands. As such, no help needed. Sorry for the bother!

dlech, 2017-02-03:
No problem. :wink: There is good information here, so hopefully it will help someone else who really does need a serial console.

closed 2017-02-03 by dlech