ev3dev/ev3dev #832: ssh default password problem

PatrickSalle, 2017-02-04:
### What did you do?
I had flashed a sd card  with ev3dev-jessie-ev3-generic-2016-12-21.img
I use a PC with Ubuntu 16.04 
After a boot on a LEGO Mindstorm, I can connect on it either in bluetooth or with a USB cable.

### What did you expect to happen?
To have a ssh connection with the default password maker on the login robot

### What actually happened?
I can ping on 10.42.0.1
but ev3dev.local is unreachable
and after ssh robot@10.42.0.1 
robot@10.42.0.1's password:     maker
Permission denied, please try again.

I have tried two times to reflash my sdcard.
I have also tried to suppress manually the password in the /etc/password on the sdcard
without success

### What hardware and software are you using (including version numbers)?

mob41, 2017-02-04:
What is the type (e.g. SDHC) and size (e.g. 8GB) of your SD card? SD Cards with 32 GB or higher seems to have problems with EV3. Try to flash the image with a lower capacity (e.g. 8GB microSDHC) SD card.

In another case, you might downloaded a bad image, try downloading a new copy.

@dlech is there any possibility to put a MD5 sum and hasher for users to validate the downloaded file?

Similar: https://github.com/ev3dev/ev3dev/issues/804

PatrickSalle, 2017-02-04:
Thank you.
I used a SDHC 32GB. I had a 16GB in a Raspberry and will swap the images.
I had token the image on the EV3DEV site (and two times).
I will look also at the #804

dlech, 2017-02-04:
> @dlech is there any possibility to put a MD5 sum and hasher for users to validate the downloaded file?

I don't see an easy way to do this since we are using GitHub releases for image downloads. There may be something in the works with the Etcher software though.

PatrickSalle, 2017-02-05:
I have tried with a 16GB sdcard (which was previously in a Raspberry Pi)
The result is the same 
On the EV3 Wired Connected a strange IP address  169.254.38.107
this address changes at each reboot
and just a ping possible on 10.42.0.1 (which is the only active IPV4 connection on my PC)
The mac address is 12:16;53:52:F5:FF

dlech, 2017-02-05:
OK, I see what is going on now. You don't have a network connection. 169.254.38.107 is what is called a "local link address" meaning that the EV3 could not get an IP address from your computer. 10.42.0.1 is the IP address of your computer, which is why you can ping it, but not connect with SSH.

Which instructions did you follow to setup a network connection?

PatrickSalle, 2017-02-05:
Bon sang mais c'est bien sûr (a french expression when we have not understood
an evidence).
I followed the instructions  on http://www.ev3dev.org/docs/getting-started/

mob41, 2017-02-05:
You might missed some steps.

Have you taken a look of http://www.ev3dev.org/docs/networking/?

You mentioned you connected it with USB on a Ubuntu. You should go to http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/ and click on the "Ubuntu" tab. And follow the instructions.

PatrickSalle, 2017-02-05:
It was exactly what I have do.
But in fact, in the first step Ethernet network (LEGO group EV3...)
does not appear in the list. I have tried to edit connections and add the connection.
It seems to recognize the right mac address, but it's not functional.

I wiill try with an old Ubuntu on another old laptop...

mob41, 2017-02-05:
I just pick my free time to try to answer this but seems I really don't know how to guide you to continue... Wait for @dlech, he is so professional on this :smile:

PatrickSalle, 2017-02-05:
Than you very much and sorry to take a part of your time.
I will wait an answer of @dlech.

dwalton76, 2017-02-05:
@PatrickSalle can you cut-n-paste the output from "ip addr show" on the PC that the ev3 connects to via usb.

PatrickSalle, 2017-02-05:
With a very old laptop and a so old Ubuntu version (Lucid) it works!
Perhaps due to an error at the first test some bad information has been registered
on my recent laptop?

dlech, 2017-02-05:
Some screenshots of your configuration on the new laptop could be helpful to see what is wrong.

PatrickSalle, 2017-02-06:
On the recent laptop 

>lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 064e:9700 Suyin Corp. Asus Integrated Webcam
Bus 001 Device 003: ID 8087:0a2a Intel Corp. 
Bus 001 Device 009: ID 0694:0005 Lego Group 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

An extract of >dmesg

[  553.759752] wlan0: associated
[  553.759811] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 1096.794324] usb 1-1: new high-speed USB device number 7 using xhci_hcd
[ 1101.905027] usb 1-1: device descriptor read/64, error -110
[ 1117.117249] usb 1-1: device descriptor read/64, error -110
[ 1117.333182] usb 1-1: new high-speed USB device number 8 using xhci_hcd
[ 1122.444000] usb 1-1: device descriptor read/64, error -110
[ 1137.300474] usb 1-1: new high-speed USB device number 9 using xhci_hcd
[ 1137.430235] usb 1-1: New USB device found, idVendor=0694, idProduct=0005
[ 1137.430244] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1137.430249] usb 1-1: Product: EV3+ev3dev
[ 1137.430252] usb 1-1: Manufacturer: LEGO Group
[ 1137.430256] usb 1-1: SerialNumber: 00165352f5ff
[ 1137.432728] cdc_ether 1-1:2.0 usb0: register 'cdc_ether' at usb-0000:00:14.0-1, CDC Ethernet Device, 12:16:53:52:f5:ff

Result of a nm-configuration which proves that the ev3 is not seen
[image]

PatrickSalle, 2017-02-06:
I have forgotten a part of the dmesg

 1137.450481] cdc_ether 1-1:2.0 enp0s20u1c2: renamed from usb0
[ 1137.474728] IPv6: ADDRCONF(NETDEV_UP): enp0s20u1c2: link is not ready
[ 1137.474906] cdc_ether 1-1:2.0 enp0s20u1c2: kevent 12 may have been dropped
[ 1137.474914] cdc_ether 1-1:2.0 enp0s20u1c2: kevent 12 may have been dropped
[ 1137.521216] cdc_ether 1-1:2.0 enp0s20u1c2: kevent 12 may have been dropped
[ 1828.990331] cdc_ether 1-1:2.0 enp0s20u1c2: kevent 12 may have been dropped
[ 1874.383589] cdc_ether 1-1:2.0 enp0s20u1c2: kevent 12 may have been dropped
[ 1874.383596] cdc_ether 1-1:2.0 enp0s20u1c2: kevent 12 may have been dropped
[ 1874.383616] cdc_ether 1-1:2.0 enp0s20u1c2: kevent 12 may have been dropped
[ 1874.383619] cdc_ether 1-1:2.0 enp0s20u1c2: kevent 12 may have been dropped
[ 1874.384180] cdc_ether 1-1:2.0 enp0s20u1c2: kevent 12 may have been dropped
[ 1874.392315] cdc_ether 1-1:2.0 enp0s20u1c2: kevent 12 may have been dropped
[ 1884.032423] cdc_ether 1-1:2.0 enp0s20u1c2: kevent 12 may have been dropped
[ 1884.032431] cdc_ether 1-1:2.0 enp0s20u1c2: kevent 12 may have been dropped
[ 1884.032455] cdc_ether 1-1:2.0 enp0s20u1c2: kevent 12 may have been dropped
[ 1884.032459] cdc_ether 1-1:2.0 enp0s20u1c2: kevent 12 may have been dropped
[ 2626.617383] perf interrupt took too long (2569 > 2500), lowering kernel.perf_event_max_sample_rate to 50000

mob41, 2017-02-06:
In my understanding, you need to add the "ev3dev" yourself manually  in the <code>nm-configuration</code> right?

PatrickSalle, 2017-02-06:
Thank you it works.

dlech, 2017-02-06:
Don't forget to close the issue if it is resolved.

closed 2017-02-06 by PatrickSalle