ev3dev/ev3dev #833: SSH to multiple EV3 bricks via USB using a Raspberry Pi

craigsammisutherland, 2017-02-05:
I am trying to connect multiple EV3 bricks to a single Raspberry Pi using USB. I can physically plug in them and see they are connected (using ip add) but I can only SSH into one of the bricks.

The process I normally use to connect to a brick is to SSH into the Raspberry Pi, then type in:
```
SSH robot@ev3dev.local
```
But this only works for the first connected EV3 brick.

I have seen #300 but this involves installing extra software onto the EV3 brick which I would rather not do.

So my question is, how do I connect to multiple EV3 bricks from a Raspberry Pi command line, using just the pre-existing software on the EV3?

WasabiFan, 2017-02-05:
I see a couple options, but they may not be the best (I'm not an expert):

- Use the IP address that is displayed on the brick display to connect. This changes every time, so it isn't ideal.
- Change the hostname of each brick. If you plug just one in, you can SSH into it, run `ev3dev-config`, and use the menu to change the hostname. Historically I've named embedded devices using Greek letters (so `ev3dev-alpha`, `ev3dev-beta`, etc.) but you can call them whatever you want. Then you can connect another brick and run through the same process.
- There might be a way to make mDNS return multiple IP addresses for a given hostname, but I'm not sure.

dlech, 2017-02-05:
:+1: for changing the hostname with `sudo ev3dev-config`.

craigsammisutherland, 2017-02-05:
Thanks for the suggestion. 

I changed the hostnames with ```sudo ev3dev-config``` on each EV3 brick and then tried connecting to multiple bricks. With the hostnames I can now specify which brick to connect to, but only if it was the most recently plugged in brick.

So, I tried pinging the bricks using the new hostnames. Each hostname gives me the same IP address (192.168.0.1). It appears the the bricks are set to the same IP address, is there some way to change this IP address?

dlech, 2017-02-05:
Are you running ev3dev on the Raspberry Pi as well? If so, enable tethering using `connmanctl` on the Raspberry Pi.

If not, you can assign manual IP addresses on each EV3 using the Wireless and Networks > All Network Connections > *name* > IPV4.

craigsammisutherland, 2017-02-05:
No, I am using Raspbian on the Raspberry Pi. 

Is it possible to assign manual IP addresses from the command line? Apologies if this is obvious, I don't have a Linux background and everything I've googled is about changing the IP address of the Pi rather than the EV3 brick.

WasabiFan, 2017-02-05:
I think there's something more basic going on here. Can you explain what you meant by the following?

> With the hostnames I can now specify which brick to connect to, but only if it was the most recently plugged in brick.

Can you copy-paste the terminal output of repeatedly running `ping` with the different hostnames? I believe that the hostname approach is correct and that there's something procedurally wrong; it should "just work".

craigsammisutherland, 2017-02-05:
With both bricks plugged in I get:
```
pi@raspberrypi:~/python/ev3_srv $ cd ../..
pi@raspberrypi:~ $ ping ev3dev-alpha.local
ping: unknown host ev3dev-alpha.local
pi@raspberrypi:~ $ ping ev3dev-beta.local
PING ev3dev-beta.local (192.168.0.1) 56(84) bytes of data.
64 bytes from 192.168.0.1: icmp_seq=1 ttl=64 time=0.627 ms
```

If I then uplug ev3dev-beta, I get:
```
pi@raspberrypi:~/python/ev3_srv $ cd ../..
pi@raspberrypi:~ $ ping ev3dev-alpha.local
PING ev3dev-alpha.local (192.168.0.1) 56(84) bytes of data.
64 bytes from 192.168.0.1: icmp_seq=1 ttl=64 time=0.627 ms
```

WasabiFan, 2017-02-05:
Ah, I see now; that definitely isn't what I would expect. What networking option do you have selected on your bricks? Are they set to use a static IP address (in which case, you'll need to change the addresses as @dlech said) or to use DHCP?

craigsammisutherland, 2017-02-06:
I am running the bricks with the default settings.

dlech, 2017-02-06:
You can set a static IP address on each EV3 using `connmanctl`

http://manpages.ubuntu.com/manpages/xenial/en/man1/connmanctl.1.html

craigsammisutherland, 2017-02-06:
I have tried changing the IP address using connmanctl. I use the following commands:
```
robot@ev3dev-alpha:~$ connmanctl services
    Wired                gadget_021653459872_usb
robot@ev3dev-alpha:~$ connmanctl config gadget_021653459872_usb --ipv4 manual 192.168.0.10 255.255.255.0 0.0.0.0
```

But the IP address does not change. Does anyone know what I am doing wrong?

dlech, 2017-02-06:
I just tried it and it worked for me. Although one thing I did differently is that I set the gateway to something other than 0.0.0.0. In your case, the gateway is the IP address of your Raspberry Pi. (Another thing I did differently was typed in a few things wrong the first time and completely lost the connection and had to reboot, so maybe rebooting is a factor).

craigsammisutherland, 2017-02-07:
I just tried again using the IP address of the Raspberry Pi and rebooting, but when the brick reboots, it is still using 192.168.0.1.

Would you be able to post the commands you used?

dlech, 2017-02-07:
connmanctl config gadget_0216533f4838_usb --ipv4 manual 10.42.0.5 255.255.255.0 10.42.0.1

I'm connecting to my Ubuntu desktop, so I used `10.42.0.*` because I didn't feel like messing up the configuration on my desktop, but it is essentially the same.

Perhaps you have enabled tethering on the EV3 bricks or somehow changed some other configuration in your previous attempts to get this working?

dlech, 2017-02-07:
After you try to configure, what is the output of 

    connmanctl services gadget_021653459872_usb

closed 2020-08-17 by dlech