ev3dev/ev3dev #836: Reusing ssh connections

Dave-Baum, 2017-02-09:
I'm not sure if this is anywhere in the docs, but I found that reusing ssh connections was very helpful.  Establishing a new ssh connection typically takes a few seconds, which makes things like scp'ing a single file annoyingly slow.  However, ssh does provide the ability for new ssh connections to reuse and share previously established connections.  The key is to add the following to your .ssh/config file (assuming Linux or MacOS host):

Host ev3
  HostName 192.168.1.10
  User robot
  ControlMaster auto
  ControlPath ~/.ssh/master-%r@%h:%p

In this case I defined a new host called ev3, mapped it to the static address of my ev3 brick, told ssh to use the "robot" account, and reuse a connection if possible.  If your ev3 already has a valid dns name then use that in the "Host" line and don't use a "HostName" line.

Now start an initial connection (i.e. "ssh ev3") and leave this shell open.  It will serve as the master connection for any future ssh/scp commands.

You can open new shells ("ssh ev3"), issue single commands ("ssh ev3 ls"), or copy files ("scp foo.py ev3:).  All of these will be much faster now because they don't have to establish a new connection.

There is another option you can use to leave the master connection around for a limited time even after you close the shh process that initiated it (i.e. "ControlPersist 600").  I prefer not to use this and just keep the original ssh shell open while doing my development and then explicitly closing it when I'm finished.

dlech, 2017-02-11:
This is a pretty neat trick. Maybe something for an "Advanced Networking" [tutorial](http://www.ev3dev.org/docs/tutorials/)?

Dave-Baum, 2017-02-14:
Sure - any guidelines on writing a tutorial or should I just fork the repo, edit, and send a pull request?

WasabiFan, 2017-02-14:
I'd recommend looking at some existing similar pages to get a sense of formatting, but other than that there isn't much I think you need to know.  Keep things concise while making sure that even a beginner can follow, and I think it'll be fine.

The flow you have is right -- fork and edit, then open a PR for review.

closed 2017-02-19 by Dave-Baum