ev3dev/ev3dev #837: Using FTDI adapter as a USB IR Transmitter (some toughts)

JorgePe, 2017-02-10:
Hi all!

While reviewing my old notes about using LIRC and a soundcard as an IR Transmitter I found that it is also possible to use LIRC and a FTDI adapter for the same purpose. With less than €20 it has a short range (~1 meter) but adding a FET, some LEDs and resistors, I believe it can reach much more.

I made a long and somewhat confusing [howto](http://ofalcao.pt/blog/2017/using-ftdi-as-ir) but I have some doubts about further developments.

A big advantage I see on this hack is the possibility of having several independent transmitters since LIRC can address each FTDI adapter by its S/N. This could allow some crazy setups like using 10 transmitters to control 20 LEGO PF motors or spreading several transmitters across a room to extend the control range of a PF train or a car.

Instaling LIRC on ev3dev when there are already some C and python libraries that can do the same is somewhat overkill but it was the easiest way (for me). I didn't use LIRC as a daemon yet but I suppose it will use some memory and CPU that could be used somewhere else. And it will also contribute to some latency.

I considered the idea of creating a driver (although I have no idea on how to do that and, honestly, I think that I am to lazy and lack some programmers background to do that).
 
Since it is not exactly a Mindstorms device like HiTechnic IRLink do you think this hack may have some interest? I can improve my ideas a bit further and then write a Tutorial based on my blog's post but I am not so sure it that fits on the idea of ev3dev's Tutorials.

So fellows... ideas, suggestions, critics, requests for even more hacks?

dlech, 2017-02-11:
> I considered the idea of creating a driver

We have an interface for "userspace drivers" that I hope to improve in the future that would be good for this. No kernel hacking required.

> So fellows... ideas, suggestions, critics, requests for even more hacks?

Since it requires quite a bit of work to get up and running, I would suggest making a project page for it instead of a tutorial page. If LIRC in stretch works better and you don't have to compile your own, then we can look at making a tutorial page.

JorgePe, 2017-02-12:
> No kernel hacking required
That would be great! Not just for my IR Transmitter but for all the bunch of gadgets I've been connecting to my EV3. 

LIRC version in stretch is 0.9.4c-7, it already includes ftdix (was included in 0.9.4a) but I didn't test it, I went directly to sourceforge last version, it was also 0.9.4c, had a few problems compiling in Ubuntu 16.10 but two days later a first version of 0.9.4d was released that fixed my problems.

If I was a smart person I would spend some time with docker or any other tool that allows me to compile for EV3 in my i7 laptop :)

Thanks for the 'project page' sugestion. I've been postponing my first project but it really makes sense to present something more than just an hack, I was already wondering about possible uses for several IR transmitters.

dlech, 2017-10-24:
Anything else to do here?

JorgePe, 2017-10-24:
Nop.
And I'm still far from writing drivers.

closed 2017-10-24 by JorgePe