ev3dev/ev3dev #838: Website doesn´t show some images:

jabrena, 2017-02-11:
Hi, 

this morning for some reason, I didn´t show some traditional images on the website.
http://www.ev3dev.org/

```
<div class="col-lg-4">
    <img class="img-responsive" id="splash-thumbnail" src="images/logo_ev3dev_mono.png" alt="ev3dev boot logo">
</div>
```

I tested with FF & Safari on OSX and I see the image but for Chrome the logo disappeared. It is like some responsive media query was enabled.

Juan Antonio

dlech, 2017-02-11:
I can see the image in Chrome (Chromium 55 on Ubuntu). It disappears if the window is < 1200px wide.

WasabiFan, 2017-02-11:
Yes, it will be hidden when the window becomes small enough that things start to collapse into mobile view.

jabrena, 2017-02-11:
Maybe, it is necessary to check that media query. If the window enter in a Tablet dimension, in that case, it should be hidden. Isn´t it?

Cheers

dlech, 2017-03-10:
Can you post a screenshot?

jabrena, 2017-03-12:
I link a image:

[image]

Take a look dimensions from DOM. If you observe, although my laptop is 1024, if I reduce a bit the screen, the media query disappear. I am not sure if this behaviour is natural from Bootstrap or not.

Cheers

WasabiFan, 2017-03-12:
Your display is small enough that it's being identified as a tablet, and as a result the unnecessary images are being hidden and content is wrapping.

jabrena, 2017-03-12:
My comment is if the website for tablet should maintain the image about Linux. The essence of EV3Dev is the development of a complete Linux inside of a EV3Brick, so the image is significative.

WasabiFan, 2017-03-16:
Although that image is technically valid, it isn't very useful for a new user and I don't believe that it is a good visual element to have on that page. We plan to replace that image with something that is more relevant to the project now that the old boot screen depicted there has been replaced, and in the process I can make it flow properly in tablet/mobile view. See https://github.com/ev3dev/ev3dev.github.io/issues/238.

jabrena, 2017-03-17:
Yeahhh!!

I like so much this pic:
https://cloud.githubusercontent.com/assets/3310349/21585636/55456b4a-d092-11e6-9746-557d54c6600c.png

jabrena, 2017-03-17:
This weekend, I will give my personal comment.

Cheers

closed 2017-10-24 by dlech