ev3dev/ev3dev #840: file executes but does nothing

thepinecone461, 2017-02-12:
I tried to run a program from file explorer
#!/usr/bin/env python3

from ev3dev.ev3 import *
from time import sleep

m = LargeMotor('outD')

m.run_to_rel_pos(position_sp=360, speed_sp=900, stop_action="hold")

sleep(5)   

### What did you expect to happen?
It was supposed to run the motor

### What actually happened?
the leds turned orange  for 2 sec then it went back to the normal screen

### What hardware and software are you using (including version numbers)?

Image file:         ev3dev-jessie-ev3-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3

WasabiFan, 2017-02-12:
Try this: [ev3dev-lang-python FAQ: My script works when launched as python3 script.py but exits immediately or throws an error when launched from Brickman or as ./script.py](http://ev3dev-lang.readthedocs.io/projects/python-ev3dev/en/latest/faq.html#my-script-works-when-launched-as-python3-script-py-but-exits-immediately-or-throws-an-error-when-launched-from-brickman-or-as-script-py).

thepinecone461, 2017-02-21:
The page doesn't work anymore
can you make another one

WasabiFan, 2017-02-21:
I updated my comment with the new link.

dlech, 2017-10-24:
Can this be closed?

FYI, in ev3dev-stretch, errors will be logged to a file when running from Brickman or using `brickrun` from a terminal or shown in the output window in Visual Studio Code when using the ev3dev extension.

closed 2018-04-22 by dlech