ev3dev/ev3dev #841: Realtek 8152 usb ethernet adapter not working

MaxNoe, 2017-02-14:
### What are you trying to do?
Using a realtek 8152 USB to ethernet adapter

### What did you expect to happen?
A working ethernet link

### What actually happened?
The device is recognised by udev:

```
[   27.325547] configfs-gadget gadget: high-speed config #1: c
[  215.806554] usbcore: registered new interface driver rtl8150
[  220.094436] usb 1-1: USB disconnect, device number 2
[  223.015096] usb 1-1: new full-speed USB device number 3 using ohci
[  223.246344] usb 1-1: New USB device found, idVendor=0bda, idProduct=8152
[  223.246507] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  223.246566] usb 1-1: Product: USB 10/100 LAN
[  223.246611] usb 1-1: Manufacturer: Realtek
[  223.246652] usb 1-1: SerialNumber: A0CEC8052DDF
```

But `ip link` does not show any new ethernet link.
On my arch notebook, the usb2ethernet adapter is recognised immediately and a new network link is created using the `r8152` kernel module.

This kernel module is not available on the ev3dev

### What hardware and software are you using (including version numbers)?

```
Image file:         ev3dev-jessie-ev3-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

MaxNoe, 2017-02-14:
For comparision, here the `dmesg` output on my arch notebook:
```
[Feb14 00:01] usb 1-3: new high-speed USB device number 15 using xhci_hcd
[  +0.357323] usb 1-3: reset high-speed USB device number 15 using xhci_hcd
[  +0.240799] r8152 1-3:1.0 eth1: v1.08.7
[  +0.568797] r8152 1-3:1.0 enp0s20f0u3: renamed from eth1
[  +0.027458] IPv6: ADDRCONF(NETDEV_UP): enp0s20f0u3: link is not ready
[  +0.028067] IPv6: ADDRCONF(NETDEV_UP): enp0s20f0u3: link is not ready
[  +2.170547] IPv6: ADDRCONF(NETDEV_CHANGE): enp0s20f0u3: link becomes ready
```

dlech, 2017-02-15:
It looks like we don't have `CONFIG_USB_RTL8152` turned on in the EV3 kernel. I will enable it in the next kernel release.

MaxNoe, 2017-02-15:
Awesome

dlech, 2017-02-15:
You could compile your own kernel. See <https://github.com/ev3dev/ev3dev-buildscripts>.

MaxNoe, 2017-02-15:
How long do you expect it to be till the next release with the updated kernel?

dlech, 2017-02-15:
Probably about 2 months.

MaxNoe, 2017-02-15:
I downloaded the driver from realtek here:
http://www.realtek.com.tw/Downloads/downloadsView.aspx?Langid=1&PNid=14&PFid=55&Level=5&Conn=4&DownTypeID=3&GetDown=false

I installed the `linux-headers-ev3dev-ev3` package. 
I also had to run:
```
sudo ln -s /usr/src/linux-headers-4.4.47-19-ev3dev-ev3/ /lib/modules/4.4.47-19-ev3dev-ev3/src
sudo ln -s /usr/src/linux-headers-4.4.47-19-ev3dev-ev3/ /lib/modules/4.4.47-19-ev3dev-ev3/build
```

But then the compilation fails with
```
make -C /lib/modules/4.4.47-19-ev3dev-ev3/build SUBDIRS=/home/robot/r8152-2.08.0 modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.47-19-ev3dev-ev3'
  CC [M]  /home/robot/r8152-2.08.0/r8152.o
In file included from ./arch/arm/include/asm/hardirq.h:6:0,
                 from include/linux/hardirq.h:8,
                 from include/linux/interrupt.h:12,
                 from include/net/flowcache.h:4,
                 from include/net/netns/xfrm.h:9,
                 from include/net/net_namespace.h:28,
                 from include/linux/netdevice.h:43,
                 from /home/robot/r8152-2.08.0/r8152.c:15:
./arch/arm/include/asm/irq.h:7:23: fatal error: mach/irqs.h: No such file or directory
 #include <mach/irqs.h>
                       ^
compilation terminated.
```

dlech, 2017-02-15:
The kernel has a built-in driver. I would not use the ones from realtek.

MaxNoe, 2017-02-15:
After downloading the file `irqs.h` from the linux source repo, I could build the module, the `r8152.ko` file exists, but `/sbin/modprobe r8152` says undefined symbol:
```
$ dmesg
...
[  891.296447] r8152: Unknown symbol mii_nway_restart (err 0)
```

dlech, 2017-02-15:
Did you run `depmod`?

MaxNoe, 2017-02-15:
There is no mii module

dlech, 2017-02-15:
Try this...

[package.zip](https://github.com/ev3dev/ev3dev/files/778395/package.zip)

MaxNoe, 2017-02-15:
It's working ;)

MaxNoe, 2017-02-15:
What's that?

I got it working anyway, after running `depmod` and `modprobe r8152`

dlech, 2017-02-15:
a kernel package with the mainline kernel driver enabled

MaxNoe, 2017-02-15:
Ok, will try it out tomorrow, thanks for the support!

dlech, 2017-04-03:
Fix has been released in kernel 4.4.59-19-ev3dev-ev3. To upgrade, make sure your EV3 is connected to the Internet, then...

```
sudo apt-get update
sudo apt-get install linux-image-ev3dev-ev3
```

closed 2017-04-03 by dlech