ev3dev/ev3dev #843: Install deb package from ubuntu with support armhf

jabrena, 2017-02-16:
Hi,

I would like to know if a debian package from Ubuntu, could be installed on jessie if this package has armhf.

https://launchpad.net/ubuntu/+source/openjdk-9

Many thanks in advance.

Juan Antonio

dlech, 2017-02-16:
It will only run on Raspberry Pi 2/3 and BeagleBone. `armhf` == armv7 instruction set plus hard float.  The EV3 is armv5 with soft float.

jabrena, 2017-02-16:
Hi @dlech, the idea is to test Java 9 on Raspi boards.

dlech, 2017-02-16:
There is a debian package too: https://tracker.debian.org/pkg/openjdk-9

jabrena, 2017-02-16:
Hi, but I try to install using sudo:

```
robot@ev3dev:~$ sudo apt-get install openjdk-9
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package openjdk-9
```

I couldn´t do it. Any alternative? the package has armhf support.

dlech, 2017-02-16:
it is only available in `unstable`, so you have to add `unstable` to your apt package sources in `/etc/apt/sources.list`

jabrena, 2017-02-16:
Hi @dlech, 

In my source.list file I have the following content:

``` 
deb http://httpredir.debian.org/debian jessie main  contrib non-free
#deb-src http://httpredir.debian.org/debian jessie main contrib non-free

deb http://security.debian.org/ jessie/updates main contrib non-free
#deb-src http://security.debian.org/ jessie/updates main contrib non-free

deb http://archive.ev3dev.org/debian jessie main
#deb-src http://archive.ev3dev.org/debian jessie main

```

I tried to add unstable in the repo: http://archive.ev3dev.org/debian & http://httpredir.debian.org/debian but in both cases, when I execute:

```  
sudo apt-get update
``` 

I receive an error downloading packages from unstable. I suppose that it is something elemental but I don`t know how to add the address from:

https://tracker.debian.org/pkg/openjdk-9

With the parameter unstable.

What is the address to add in /etc/apt/sources.list in order to download.

Sorry, I suppose that it is a basic question.

Cheers

Juan Antonio

WasabiFan, 2017-02-16:
Wouldn't it be this? I really don't know what I'm doing, but that seems correct :grin:

```
deb http://httpredir.debian.org/debian sid main contrib non-free
```

jabrena, 2017-02-16:
I tried but I continue with the same problem:

``` 
robot@ev3dev:~$ sudo apt-get install openjdk-9
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package openjdk-9
```

BertLindeman, 2017-02-16:
Just to be sure (sorry, a bit naive) did you do 
`sudo apt-get update`
after changing /etc/apt/sources.list ?

jabrena, 2017-02-16:
Hi, I couldn´t install in an automatic way, but... with some patience... I installed all packages in a manual way... :D 

``` 
robot@ev3dev:~$ java -version
openjdk version "9-Ubuntu"
OpenJDK Runtime Environment (build 9-Ubuntu+0-9b155-1)
OpenJDK Server VM (build 9-Ubuntu+0-9b155-1, mixed mode)
```

jabrena, 2017-02-16:
``` 
robot@ev3dev:/usr/lib/jvm/java-9-openjdk-armhf/bin$ jshell -v
Feb 16, 2017 11:14:50 PM java.util.prefs.FileSystemPreferences$1 run
INFO: Created user preferences directory.
/imports
|  Welcome to JShell -- Version 9-Ubuntu
|  For an introduction type: /help intro

jshell> /imports
|    import java.io.*
|    import java.math.*
|    import java.net.*
|    import java.nio.file.*
|    import java.util.*
|    import java.util.concurrent.*
|    import java.util.function.*
|    import java.util.prefs.*
|    import java.util.regex.*
|    import java.util.stream.*

```

jabrena, 2017-02-16:
Anyway, although, I could install in a manual way, @dlech how to indicate debian to download that package?

```
sudo apt-get install openjdk-9
```

What is the source to add in order to install?

In my case, I downloaded every package from here:
https://launchpad.net/ubuntu/+source/openjdk-9

and for dependencies, here:
https://packages.debian.org/search?suite=default&section=all&arch=any&searchon=names&keywords=libpng16-16

jabrena, 2017-02-20:
Hi @dlech, do you think that using the sources, is it possible to compile to ARM5?

dlech, 2017-02-21:
The debian package page says that it should be compiling for armel, so I am guessing that there is some problem that it is not working on armel currently. I would suggest open a bug with debian and hopefully someone who knows what is going on will tell you why there is not a binary armel package already.

jabrena, 2017-02-21:
I wil create it

closed 2020-08-17 by dlech