ev3dev/ev3dev #844: Working programs will not run if I unplug the USB cable.

clplaneguy, 2017-02-16:
I have

The Hewlett-Packard has 
Windows 7
Lego MindstormSEV3 Programmable Brick Revision 0006
Brickman v0.8.0 2014-2015
ev3dev? Kernel 4.4.32-17-ev3dev-ev3
MobaXterm Personal Edition V9.4
ev3dev Python3.4.2((defalt,oct 8 2014:47:30
Java 8.01210.13: 8 update 121 (8u121) (64-bit)
Python 3.5.1 (64 bit)
Python 3.6.0 (32 bit)
leJOS EV3 0.9.1-beta
leJOS EV3 0.9.0-beta (win 32)
jdk1.8.0_121
jre1.8.0_121
leJOS_EV3_0.9.0-beta_win32_setup
Java 8 (Computer)
Java 7 (EV3)
Mon Nov 16, 2015 5:29 pm   leJOS EV3 0.9.1-beta release is now out - see https://lejosnews.wordpress.com/2015/11
leJOS_EV3_0.9.0-beta_win32_setup (2)			Brick card
leJOS:0.9.1-beta	From brick
Menu:  0.9.1		From brick
Eclipse 4.6.2 (Neon)

### What are you trying to do?

I am using EV3/EV3dev/PyCharm.  Last week I was using MobaXterm.  I have written a PID program for the EV3 in PyCharm.  It performs in a very believable manner.  I am satisfied with the program and am ready to get the bot off my desktop and onto the floor.  But, when I unplug the USB cable, the program will not execute.  I had this problem last weak with MobaXterm and the editor was sending the file in DOS format and I needed UNIX format.  Is this the same problem?  How do I set PyCharm to save a file in UNIX format?

```
#!/usr/bin/env python3

from ev3dev.ev3 import *
from time import sleep

import ev3dev.ev3 as ev3

print('Auto Driver PID 6')
# File EV3 Python PID 8  PID8.py
ev3.Sound.speak('Hello ryan.  Hi Hunter. Jacob.  Give me five.  I do not have any hands. '
                'Uncle Michael wants me to follow a line.  But, I do not see a line.  I will go this way.  '
                'Ready. Set. Go.').wait()
LeftWheel = ev3.LargeMotor('outA')
RightWheel = ev3.LargeMotor('outB')
cl = ColorSensor()
cl.mode = 'COL-REFLECT'

Max = 0
Min = 100
Average = 50
Error = 0
Integral = 0
Derivative = 0
LastError = 0
Kp = 12
Ki = .4
Kd = 1
PID = 0
Time = 100
LeftPower = 1
RightPower = 1
MaxStraightPower = 300
MaxTurnPower = 700
RuningSum = 0
Denominator = 1
Iteration = 0
Seed = 200
SlowStart = 4
Light = 0
Push = 0
SpeedFactor = 0
Temp = 0
Start = True

Spread = Max - Min
Begining = '{:3d}    Light {:>4.0f}    Min {:3d}   Max {:3d}      Spread {:>4d}     Average {:3.0f}'.format(Iteration,
                                                                                                            Light,
                                                                                                            Min, Max,
                                                                                                            Spread,
                                                                                                            Average)
Ending = '    Error {:>4.0f}    Integral {:>5.0f}    Derivative {:>5.0f}    Denominator {:>5.0f}    Seed {:>3d}    Temp {:>3d}     PID {:>5.0f}    LeftPower {:>5.0f}    RightPower {:>5.0f}'.format(

    Error, Integral, Derivative, Denominator, Seed, Temp, PID, LeftPower, RightPower)
# print(Begining, Ending)

for x in range(1, 600):
    Iteration += 1
    Light = cl.value()
    if Light < Min:
        Min = Light

    if Light > Max:
        Max = Light
    Spread = Max - Min

    Average = int((Average + (Min + Max) / 2) / 2)
    Error = int(Light - Average)
    Integral += Error
    Derivative = Error - LastError

    if Integral < -MaxTurnPower:
        Integral = -MaxTurnPower
    if Integral > MaxTurnPower:
        Integral = MaxTurnPower

    if Derivative < -MaxTurnPower:
        Derivative = -MaxTurnPower

    if Derivative > MaxTurnPower:
        Derivatine = MaxTurnPower

    PID = Kp * Error + Ki * Integral + Kd * Derivative

    if PID < -MaxTurnPower:
        PID = -MaxTurnPower

    if PID > MaxTurnPower:
        PID = MaxTurnPower

    NormalizedPID = PID / 900
    SpeedCoefficient = 1 - abs(NormalizedPID)
    SpeedFactor = SpeedCoefficient * MaxStraightPower
    TurnFactor = NormalizedPID * MaxTurnPower
    Denominator = Max - Min
    if Denominator < 1:
        Denominator = 1

    LeftPower = (SpeedFactor + TurnFactor) * 100 / Denominator + Push
    RightPower = (SpeedFactor - TurnFactor) * 100 / Denominator + Push
    Temp = SlowStart * Iteration + Seed

    if Spread < 20:
        Derivative = 0
        Integral = 0

        if Temp < LeftPower:
            LeftPower = Temp
        else:
            LeftPower = Temp

        if RightPower > 0:
            if Temp < RightPower:
                RightPower = Temp
        else:
            RightPower = Temp

    if Spread > 50:
        if Start is True:
            LeftWheel.run_timed(time_sp=100, speed_sp=-100)
            RightWheel.run_timed(time_sp=100, speed_sp=-100)
            ev3.Sound.speak(
                'I found an edge. Light on the left and dark on the right.  I will follow this edge.').wait()
        Start = False

    if LeftPower < -900:
        LeftPower = -901

    if LeftPower > 900:
        LeftPower = 902

    if RightPower < -900:
        RightPower = -903

    if RightPower > 900:
        RightPower = 904

    Begining = '{:3d}    Light {:>4.0f}    Min {:3d}   Max {:3d}      Spread {:>4d}     Average {:3.0f}'.format(
        Iteration, Light,
        Min, Max, Spread,
        Average)
    #    Ending = '    Error {:>4.0f}    Integral {:>5.0f}    Derivative {:>5.0f}    Denominator {:>5.0f}    Seed {:>3d}    Temp {:>3d}     PID {:>5.0f}    LeftPower {:>5.0f}    RightPower {:>5.0f}'.format(

 #   Error, Integral, Derivative, Denominator, Seed, Temp, PID, LeftPower, RightPower)
    #   print(Begining, Ending)

    LeftWheel.run_timed(time_sp=Time, speed_sp=LeftPower)
    RightWheel.run_timed(time_sp=Time, speed_sp=RightPower)

    LastError = Error
    UninitilizedVeriabale = 0

Sound.beep()

```

WasabiFan, 2017-02-16:
I suspect it is the same issue. See the page [here](https://www.jetbrains.com/help/pycharm/2016.3/configuring-line-separators.html), specifically under the heading "To change line separator for a file, currently opened in the editor"; that explains how to change the line endings in PyCharm.

clplaneguy, 2017-02-16:
SOLVED   

THANK YOU

WasabiFan, 2017-02-16:
Please close the issue now that your problem has been resolved.

closed 2017-02-17 by clplaneguy