ev3dev/ev3dev #846: User group problems

heidashuai, 2017-02-20:
### What are you trying to do?
Sorry,but I met a problem with the user groups. I wanted to use sound function so I add robot to audio group,but I find that robot is out of ev3dev group!
And I do not know how to make it back,since I have no right to do the operations...

### What did you expect to happen?
I want to be ev3dev member again

### What actually happened?
I have no right to change the group

### What hardware and software are you using (including version numbers)?

heidashuai, 2017-02-20:
@dlech

dlech, 2017-02-20:
This sounds like yet another case of SD card corruption. Please make sure you are using an SD card smaller than 32GB. Also try a different SD card if you have one and flash a new image.

heidashuai, 2017-02-20:
@dlech Thank you David. It is more complicated than SD card problems,but I think I have already solved the it.
And for the sound function.Could you please tell how can I remotely control the ev3 to make a sound?

dlech, 2017-02-21:
This page hasn't really been maintained, but I think most of the information there is still good: <https://github.com/ev3dev/ev3dev/wiki/Using-Sound>. You should prefer commands to sysfs attributes since the sysfs attributes might not be around forever.

heidashuai, 2017-02-22:
@dlech Thank you very much! I will try it

heidashuai, 2017-02-22:
@dlech It works great!Thank you again and maybe I should close the issue.

closed 2017-02-22 by heidashuai