ev3dev/ev3dev #848: Can't connect EV3 with the Internet.

LuckyChicken91, 2017-02-20:
### What are you trying to do?
I tried to connect my EV3 with the Internet with the http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/ Tutorial. I have maked all right but at the end he showed me in the EV3 at "State:" not "Online". He showed me only "Connected"
[image]

And so it dont work to connect my EV3 with putty/ssh

At next I tried to make internet with thus toturial:
http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/
It dont work because he show me only "HELP WANTED! See issue #287." (I use windoows)

### What did you expect to happen?
That he show me "State: Online" so that
I finally can use SSH/Putty to use my EV3 with ev3dev

### What actually happened?
It doesnt work.

### What hardware and software are you using (including version numbers)?
Hardware:
 LEGO Mindstorms EV3
Software: ev3dev.
Version number: I think I dont need to show it for this problem.

I cant copy and paste the output of "ev3dev-sysinfo -m" because I cant go into the ssh/putty console of my EV3

(Sorry for bad english by the way)

alesolano, 2017-02-20:
You can still go with the USB option. I have not tried to connect the EV3 via bluetooth, but it works perfect with USB on Windows. I followed this tutorial: http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/

LuckyChicken91, 2017-02-20:
Have you readed the full issue..? I have already tested exactly this tutorial but it doesnt work

LuckyChicken91, 2017-02-20:
Dont work he still show "Connected"

alesolano, 2017-02-20:
I use Windows 10 too, I don't think that's the problem.

I'm afraid I can't help you more, I'm just an user. A lucky one. I recommend to start all over again, take screenshots of what you're doing (the same pictures as the use in the tutorials) and share them here. Maybe it's just a tiny issue.

LuckyChicken91, 2017-02-20:
But what if microsoft make a change at the network settings etc. ?

Its not a tiny issue for me. For me its a ***BIG*** issue! Because I cant use my EV3 and this making me sad.
:(

Is it correct that I have connected it with my WIFI? With my normaly Internet WiFi?

Here are some screenshots but I have taken this screenshot where I haven't connected my EV3 to my Computer with the USB Cabel and thats why there stands "Select a private network connection" and not normaly "EV3", ignore this.

[image]
[image]

Does this screenshots help you? 
What should I do now?

WasabiFan, 2017-02-20:
That looks right to me. On the EV3, did you follow the part of the instructions where you configure"Windows defaults"? And is there an IP address in the upper-left corner? What is that address?

LuckyChicken91, 2017-02-21:
Yes I have setted it "Windows defaults". Yes theres is an IP address in the upper-left corner.
I hope its not dangerous to share this IP address. Its: 192.168.137.3

by the way I have now maked screenshots where my EV3 is connected with my computer.
Is the problem that there stands "No network access"?

[image]
[image]

dlech, 2017-02-21:
Everything looks correct to me. Often, Windows Internet Connection Sharing gets in a bad state and you have to turn if off and set everything up again.

dlech, 2017-02-21:
Also, try using DHCP on the EV3 instead of Windows Defaults.

LuckyChicken91, 2017-02-21:
He still show "State: Connected" but it works! It works with DHCP!

But I cant use my Internet anymore if I have plugged the USB Cabel in my Computer. Only putty works then. I think the EV3 consume too many Internet so thats it so slow in the browser that I cant use the internet anymore.

closed 2017-02-21 by LuckyChicken91

cavenel, 2017-02-22:
I think you played too much with the settings and now your computer is
trying to connect to internet via the EV3. That would explain why you can
not access internet anymore, because you should. I do not believe in the
theory that the EV3 is taking too many internet :-)

LuckyChicken91, 2017-02-22:
I dont think I have too much played with the settings.. how can I change that the EV3 is not taking so much internet?

dlech, 2017-02-22:
The EV3 should not be using the Internet unless you run a command like `apt`.

xslendix, 2018-08-06:
@dlech I am not too sure about that, because there are commands like ping, git and wget that require information in order to get and send information.

AlphaPranav9102, 2019-08-06:
@dlech What do you mean using DHCP? How do you do that? I am a beginner.

dhrumilp15, 2019-08-08:
@Cavenel I'm having that exact issue right now and I'm almost pulling my hair out over it. I'm basically in the same boat as @LuckyChicken91 and it's really frustrating. How do i make my laptop still connect to the wifi after the ev3 is plugged in through usb?

R2D2Hud, 2020-04-13:
@dhrumilp15 I have the same thing

R2D2Hud, 2020-04-13:
I just tried reflashing the sd card. no luck. :(