ev3dev/ev3dev #849: Cant install ruby gems

LuckyChicken91, 2017-02-21:
### What are you trying to do?
I try to install the ev3dev_ruby gem but it dont work he show me that he cant find it but if I install it on my normal PC it works. So I was sure that I must first make "sudo apt-get update" but it too dont works.
What should I do now?

[image]

### What hardware and software are you using (including version numbers)?

LEGO Mindstorms EV3, ev3dev

JorgePe, 2017-02-21:
You do need an internet connection. If you are using USB, you should get internet connection through your PC  if you followed that tutorial (as long as your PC has internet connection and there is no firewall or other security program on your PC blocking it).

dlech, 2017-10-24:
Can this be closed?

closed 2017-11-12 by LuckyChicken91