ev3dev/ev3dev #854: Included ROS packages in stretch images

dlech, 2017-02-28:
According to https://github.com/ev3dev/ev3dev/issues/526#issuecomment-282947974, there are ROS packages in official debian stretch.

@bidaian, do you know which packages specifically need to be installed and how big they are?

bidaian, 2017-03-01:
I think ros-core is enough for python.

Is there an ev3dev stretch docker image somewhere? I could do some more testing and let you know about the total size and what it works there. Because right now I'm doing it on top of the jessie image.

WasabiFan, 2017-03-01:
@osmado has gotten ROS running using a custom Docker script, I believe: http://www.ev3dev.org/projects/2017/02/05/ROS-and-wiimote/

P.S. In general, I'd prefer to keep core ev3dev images as light as possible, because most users won't use these extra packages.

bidaian, 2017-03-01:
Yes, I'd like to keep them light too (though I did appreciate openrobertalab being shipped). I'm asking for the stretch image because it's a lot easier to do "sudo apt-get install ros-core" than it is to modify sources.list, do some package pinning and only then install ros-core. And since stretch is going to be the default anyway (see #522), this will allow for some additional testing from those willing to do it.

dlech, 2017-03-01:
No stretch image yet. Maybe in the next month or so.

moriarty, 2017-03-01:
👍

@bidaian the ros-core meta package brings in some things that are likely not used the ev3. http://www.ros.org/reps/rep-0142.html#metapackages
And recently they've added some more supported msg generation like gennodejs which if I've seen other issues requesting newer versions of nodejs might fail.

It would be really nice to have the robot metapackage, which is designed to be the package you need to install on hardware. for the control msgs and robot state publisher. But it's likely too big for the EV3.

It is easy enough to define a custom ros-metapackage stripped down even further for the EV3.
https://github.com/ros/metapackages/blob/kinetic-devel/ros_core/package.xml
With the bare essentials, I assume most ros users if using the ev3 hardware, they are connecting over wifi or bluetooth to a machine running the ros-core, in which case, they should just need the ros_comm package, roscpp or rospy package for their language, and the package containing the messages they need. On the BrickPi, or stronger platform, none of this is an issue as they can run full ros.

I was hoping ROS2 would come out soon and make it easier to run on the lower spec ev3.

bidaian, 2017-03-01:
@moriarty I agree, a stripped down list of packages is the way to go. I'm a complete newbie both in ev3dev and ROS, so it wasn't an issue for me to go overboard with the packages. I just wanted to know if it would work. I mean, I know I will not be able to run roscore (as you expected, it's now running on a desktop, connected via bluetooth to ev3dev), but it doesn't hurt to have it taking space there at this stage.

The good thing about your suggestion, going for ros_comm and roscpp, is that it's also an easier target to backport, so that I might be able start testing it with the official jessie image.

osmado, 2017-03-01:
I just wanted to remark that it is possible to run full ROS in ev3 brick.

It is not for stretch, but I share a [project](https://github.com/osmado/ev3dev_ros_distribution) that create a ev3 docker image (based on jessie) that includes ros_comm.  I made a little modification on the ROS code that allows to execute "roscore" into ev3 brick (not host computer needed).  

The project does not explain howto obtain the ev3dev image from the docker (it is in progress).

Beside, this other [project](https://github.com/osmado/ev3_wiimote) contains two little ROS nodes:
- First node receive information from wii remote control and plublish it to ROS.
- Second node subscribes to the previous information an drive the ev3 motors.
	Note:  this project has a very dirty code.  I use it only to learn about ROS.

Finally, I created a systemd service that start ROS during booting process. (This file is not shared yet).

These two project (plus systemd service) running together allows me to use an ev3 brick with ROS and ev3dev (without external computer).  Honestly, the starting period is slow (one and a half minute after bluetooth is started). My personal opinion is that python based nodes use a lot of resources (CPU and memory usage), but c++ works more lightly.

As you say, it's a lot easier to do "sudo apt-get install ros-core" than a complex docker script :).

dlech, 2017-03-25:
The very first stretch images are available for testing: https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2017-03-25/

I have included the `ros-robot` package. It looks like it is about 70MB uncompressed with all of the dependencies (`--no-install-recommends`), so it is not *too* bad.

WasabiFan, 2017-03-25:
At 70MB, I would prefer to leave it out. I don't believe that the number of users which would take advantage of ROS is sufficient to offset the additional size.

jabrena, 2017-03-25:
Hi @WasabiFan, I think that ROS is an Advanced topic, but we should not lost the train. 

`+1 to add ROS Packages in stretch images`

Cheers

WasabiFan, 2017-03-26:
Keep in mind that this is a very large package... 70MB of space used on every ev3dev SD card for something that isn't useful for most people is a pretty large waste. If we take the position that some people may want ROS so it should be added, that sets a precedent which is severely damaging for the usability of the platform. We've already done this with a few packages and I'd caution against doing it more.

I think it would be great to have the "base" ev3dev image be as small as possible -- maybe only a hundred MB -- and then allow installation of other packages as easily as possible. I wouldn't be against offering a separate set of ROS images, but we should minimize the number of separate choices that users have to make... the Eclipse IDE is a great example of how this can go wrong.

jabrena, 2017-03-26:
Oki, another alternative more light weight, could be add in some place a doc with the instruction to add that packages. But in an automatic or manual way, ROS should run on EV3Dev. :)

dlech, 2017-03-26:
FWIW, here are all of the debian packages pulled in by `ros-robot` that aren't required by other packages:

```
  catkin cmake cmake-data docutils-common googletest graphviz
  joint-state-publisher libactionlib0d libann0 libapr1 libaprutil1
  libarchive13 libbondcpp0d libboost-atomic1.62.0 libboost-date-time1.62.0
  libboost-filesystem1.62.0 libboost-program-options1.62.0
  libboost-regex1.62.0 libboost-signals1.62.0 libcdt5 libcgraph6
  libclass-loader0d libconsole-bridge0.2v5 libcpp-common0d libcurl3 libgd3
  libgtest-dev libgts-0.7-5 libgvc6 libgvpr2 libjsoncpp1 liblog4cxx10v5
  libltdl7 libmessage-filters1d libnodeletlib-tools libnodeletlib0d
  libpathplan4 libpocofoundation46 librosbag-storage1d librosbag2d
  librosconsole2d libroscpp-serialization0d libroscpp1d libroslib0d
  libroslz4-1d librospack0d librostime0d libtf0d libtf2-0d libtf2-ros0d
  libtinyxml2.6.2v5 libtopic-tools1d libuv1 libxaw7 libxdot4 libxmlrpcpp1d
  libxmu6 libxpm4 libxt6 python-catkin-pkg python-cffi-backend
  python-cryptography python-dateutil python-defusedxml python-docutils
  python-empy python-enum34 python-funcsigs python-genmsg python-genpy
  python-geometry-msgs python-idna python-ipaddress python-message-filters
  python-mock python-netifaces python-paramiko python-pbr python-pyasn1
  python-roman python-rosbag python-rosclean python-roscpp-msgs
  python-rosgraph python-rosgraph-msgs python-roslaunch python-roslib
  python-roslz4 python-rosmaster python-rosmsg python-rosnode python-rosparam
  python-rospkg python-rospy python-rosservice python-rostopic python-roswtf
  python-sensor-msgs python-std-msgs python-tf python-yaml ros-base ros-core
  ros-robot rosout tf-tools tf2-tools topic-tools
```

`ros-core` pulls in about 50MB worth of packages

```
  catkin cmake cmake-data docutils-common googletest libapr1 libaprutil1
  libarchive13 libboost-atomic1.62.0 libboost-date-time1.62.0
  libboost-filesystem1.62.0 libboost-program-options1.62.0
  libboost-regex1.62.0 libboost-signals1.62.0 libconsole-bridge0.2v5
  libcpp-common0d libcurl3 libgtest-dev libjsoncpp1 liblog4cxx10v5
  librosbag-storage1d librosbag2d librosconsole2d libroscpp-serialization0d
  libroscpp1d libroslz4-1d librostime0d libtopic-tools1d libuv1 libxmlrpcpp1d
  python-catkin-pkg python-cffi-backend python-cryptography python-dateutil
  python-defusedxml python-docutils python-empy python-enum34 python-funcsigs
  python-genmsg python-genpy python-idna python-ipaddress
  python-message-filters python-mock python-netifaces python-paramiko
  python-pbr python-pyasn1 python-roman python-rosbag python-rosclean
  python-roscpp-msgs python-rosgraph python-rosgraph-msgs python-roslaunch
  python-roslib python-roslz4 python-rosmaster python-rosmsg python-rosnode
  python-rosparam python-rospkg python-rospy python-rosservice python-rostopic
  python-roswtf python-std-msgs python-yaml ros-core rosout topic-tools
```

fferri, 2017-04-24:
are those pulled packages for Python 2.x?

dlech, 2017-04-24:
Yes, they are all python 2. The official debian packages do not seem to have a python 3 version of anything for ROS.

osmado, 2017-04-24:
Hi,
I downloaded new ev3dev-strech docker image and I run it on my computer (docker).  I executed "roscore" and it started properly. After that I made an SD card image and run it into EV3 brick.  In this case "roscore" didn't work.  As I made in my previous project, I modified a _TIMEOUT_MASTER_START and _TIMEOUT_MASTER_STOP from 10  to 100 (/usr/lib/python2.7/dist-packages/roslaunch/launch.py).
After that modification "roscore" started.

Thanks.

dlech, 2017-05-02:
It sounds like we have enough reasons (installing only required packages, python3 support, timeout bug) to consider maintaining a fork of the official debian packages like we do with a number of other packages. If someone is willing to do this for ROS, will can setup the initial repositories. You can read more about debian packaging and being an ev3dev package maintainer at <http://www.ev3dev.org/docs/devtools/packaging-for-ev3dev/>.

dlech, 2017-05-02:
FYI, someone started working on python3 support in the official debian packages, so we should just have to cherry-pick those commits.

dlech, 2017-08-12:
It doesn't seem that anyone is making time to work on this so I am going to drop the ROS package from the default image for now.

jabrena, 2017-08-12:
For EV3 Brick, I think that running ROS core is so much for the small CPU. 
Is right @moriarty, @osmado @bidaian ?

MadsFox, 2017-10-02:
I have been struggling with the complexity of setting up a ROS node to run on my EV3 by trying to follow these instructions:
http://www.ev3dev.org/projects/2017/02/05/ROS-and-wiimote/
http://www.ev3dev.org/projects/2014/11/17/ROS-on-EV3/
and would love to find a image that contains the necessary dependencies to run a ROS node. 
I am new to ROS and EV3 and Linux and Docker, therefore i would love to have a way to work with the EV3 and ROS without to much time spend on setting it up. 
I have also tried to follow this instruction to getting ROS on EV3:
https://github.com/moriarty/ros-ev3, 
but when you follow the link to ev3dev.org, you are told that cross-compilation with Brickstrap is obsolete and you should use Docker. When I tried to use docker and gdb to just debug a "hello world" example it gave me an error, telling me that it couldn't read "arm" in line 1. This I interpret as a problem with the host and target settings in docker(both x86_64).
If you follow the instruction here:
http://hacks4ros.github.io/h4r_ev3_ctrl/
When unpacking the root files I get a lot of errors and if I try to create a bootable SD card anyway, it doesn’t boot. 
Going in circles and dreaming of an image for the SD card with what’s needed to run a ROS node on the EV3. Can anybody alternativly help me or point me to a community that might be able to?
Also new to the comunity part of github. If i'm the wrong place please redirect me. Thanks

dlech, 2017-10-02:
Have you tried and ev3dev-stretch image? There are official debian packages for ROS in stretch, so you don't have to compile your own.

bidaian, 2017-10-02:
Don't struggle! Follow dlech advice and try the [stretch image](https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2017-09-30/snapshot-ev3dev-stretch-ev3-generic-2017-09-30.img.xz). Then you can simply do an apt-get to add whatever is missing. I'm using the stretch image (not for ROS) on my EV3 and it works great. And I'm not getting the 'bluetooth not  found' (#855) I used to get after every other reboot. I recommend it.

About ROS, I gave up on it. They have fixed endiannes on the wire so that the protocol only works with certain hardware combinations. I could not make the very simple, kind of "hello world", add_int service work with ev3dev. [This is my report](https://answers.ros.org/question/256754/error-deserializing-at-the-client/) and [this is another one](https://github.com/ros-drivers/rosserial/issues/109), on MIPS. It might be  endianness or data type size, who knows, but it does not work.

The answer I've seen from them is "change endianness", because some chipsets allow for that. Anyway, until they fix that, I'd say ROS has a pretty big design issue right there (it's hard to believe, I mean, thinks like ntohs and ntohs have been there forever). If you use it, some things will work (I was succesfully reporting battery status on a topic for example), and some will not (like the service that adds two integers), so I wouldn't count on ROS on EV3 right now. Although you might make something work if you carefully choose your data types.

If you succeed, I hope you post your project to ev3dev projects page. Good luck!

MadsFox, 2017-10-03:
@dlech: Isn't it possible to create the bootable SD card image with docker or is it only possible with Brickstrap?

MadsFox, 2017-10-03:
First thank you for the quick reply and giving me hope again :) but got stuck again :(

I have tried to follow this instruction to install ros_comm on ev3dev. Tried to do it on a docker container based on a ev3-stretch-generic image, but bumped into the issues i described in this issue: https://github.com/moriarty/ros-ev3/issues/16

I also tried to install ROS in the same container following these instructions: http://wiki.ros.org/lunar/Installation/Debian

But when setting up the source.list i encountered this problem:
<pre><code>
$ sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
/bin/sh: 1: lsb_release: not found
</code></pre>

moriarty, 2017-10-03:
the ```$(lsb_release -sc)``` command is just printing the version of debian, so you can try to replace it with:

```$ sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu stretch main" > /etc/apt/sources.list.d/ros-latest.list" ```

Sorry I haven't had time to update my instructions and I'm still not near a computer which is setup to test any of this out.

MadsFox, 2017-10-03:
Great thank you @moriarty. 

Trying to follow the "Debian install for ROS Lunar" to install ROS on the ev3dev-stretch-ev3-generic docker image.

Issues:
Needed GnuPG before i could set up keys(1.3).

Can't find a ROS-lunar package to be installed on the ev3dev-stretch-ev3-generic image. Tried to search for the different distributions for a package using apt-cache search ros-_distribution_. 
Didn't find any distributions from Lunar-Fuerte for this os: 
Linux 5a9dfd206680 4.9.49-moby #1 SMP Wed Sep 27 23:17:17 UTC 2017 armv7l GNU/Linux
Unless there are any recommendations, i will go back to the jessie image and try to install Kinetic on that...

MadsFox, 2017-10-03:
To clarify, i only want a ros node to run on the brick and not ros core, so in my understanding i should only need to packages that makes it possible to communicate via service calls or topics to and from the EV3 brick. 

Thank you so fare for all your help in supporting my project. Hope it's okay i am also using this forum to get support and hopefully gather experience to support others introduction to ROS on EV3.

moriarty, 2017-10-03:
Yes, the brick isn't powerful enough to run the ROS core. I've tried... it's slow, you'll need a large swap partition and that will kill your sd card.

When I made the original ros-ev3 instructions (2014) everything for the ```ros-comm``` package needed to be cross compiled manually because the armel packages weren't available... following the instructions at that time, would give you the ability to pass basic messages to and from other nodes either on the brick or on the master computer... Unfortunately I see another message above mentioning it wouldn't work with ros services linked to from @bidaian, I only tested if I could teleop from my computer to the brick, just running a small node on the brick to listening.

 So, what you're trying to achieve is possible.

Also, I will try to find the time to dust off my EV3 and spend a weekend updating the instructions. I was just at ROSCon2017 two weeks ago, and FroSCon before that, were I met other people who knew of the ROS on EV3 attempt... I backed the BrickPi on Kickstarter years ago, but there are still a lot of EV3s in classrooms.

bidaian, 2017-10-03:
I know it used to be complicated, but I think it is no longer so. The steps I would take:
- download the [stretch image](https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2017-09-30/snapshot-ev3dev-stretch-ev3-generic-2017-09-30.img.xz)
- add the **Debian** official repository:
`sudo sh -c 'echo "deb http://ftp.br.debian.org/debian stretch main contrib non-free" > /etc/apt/sources.list.d/ros-latest.list`
- do something like sudo a `sudo apt-get update` followed by `sudo apt-get install ros-base` (full navigable list of options [here](https://packages.debian.org/source/stretch/ros-metapackages))

That should be it. And maybe the second step is not necessary.

MadsFox, 2017-10-03:
@moriarty Do you know of anybody in Copenhagen, Denmark, who also have tried to work with ROS and EV3 successfully, that might be able to give in person support?

moriarty, 2017-10-03:
@MadsFox, no sorry, but this might be interesting for you:
https://discourse.ros.org/t/invitation-to-the-second-danish-ros-meetup/2442
The date has passed, Aug 30 2017, but it might be a good lead...

Also, when the first danish ROS meetup happened in 2016, I did forward the email to LEGO just to let them know. LEGO said they're interested whenever cool projects make it up on the projects page http://www.ev3dev.org/projects/

dlech, 2017-10-03:
> @dlech: Isn't it possible to create the bootable SD card image with docker or is it only possible with Brickstrap?

Brickstrap takes a Docker image and turns it into a bootable image, so you need both.

dlech, 2017-10-03:
> EDIT: does ROS really package the armel binaries required for the EV3? I only see armhf packages. http://packages.ros.org/ros/ubuntu/dists/stretch/main/

You are looking at ubuntu. We use Debian. https://packages.debian.org/source/stretch/ros-ros

dlech, 2017-10-03:
More useful link for debian packages: https://packages.debian.org/source/stretch/ros-metapackages

dlech, 2017-10-03:
To recap, all you need to do to install ROS should be...

1. Flash an SD card with ev3dev-stretch
2. Boot it
3. Connect the EV3 to the Internet
4. Open an SSH terminal
5. `sudo apt update`
6. `sudo apt install ros-robot`

gavanderhoorn, 2017-10-04:
I was the "ROS contributor" who commented on https://github.com/moriarty/ros-ev3/issues/14.

@MadsFox: if you have a regular EV3 install, then @dlech's comment (https://github.com/ev3dev/ev3dev/issues/854#issuecomment-333884901) should work. I'm using it myself, although I haven't done much with my brick as of late.

@bidaian wrote:
> The answer I've seen from them is "change endianness", because some chipsets allow for that.

I don't think that's fair: Jackie Kay replied to your comments on https://github.com/ros-drivers/rosserial/issues/109 and she *asked* whether switching endianness on your MIPS platform would be an option. Potentially a good suggestion, as it would avoid having to update any of the infrastructure or your own code. That is definitely not the same as "change endianness" (which sounds like an imperative).

I do agree with you that any potential endianness issues should be addressed, but you have to remember that ROS is an open-source project - just as ev3dev - and we depend on contributions for these kind of things (ie: use-cases outside the regular ones). ROS1 was never really meant to be used on such small embedded platforms, which is why things like this probably weren't noticed.

I am a bit confused though, as IIRC, ROS on Stretch uses the `armel` pkgs from Debian, which would suggest little endian, but perhaps your MIPS platform is different.

@moriarty wrote:
> EDIT: does ROS really package the `armel` binaries required for the EV3? I only see `armhf` packages. http://packages.ros.org/ros/ubuntu/dists/stretch/main/

No, there are no packages for `armel` in the main ROS package repositories. That is the reason why you *have* to use the upstream Debian packages for those platforms if you want to avoid having to build from sources.

As I also commented on https://github.com/moriarty/ros-ev3/issues/16 though, be aware of the consequences: no mixing with binaries from `packages.ros.org`, not following the normal ROS release cycle and only a subset of packages that are released in 'regular ROS' are part of the Debian releases. Refer to [ros/wiki/UpstreamPackages](http://wiki.ros.org/UpstreamPackages) for some more info on that.

That being said, it *is* the quickest way to get a barebones ROS installed on the EV3.

@MadsFox wrote:
> I also tried to install ROS in the same container following these instructions: http://wiki.ros.org/lunar/Installation/Debian

As should now be clear, there is no point in doing that, as the ROS buildfarm does not build any packages for `armel` Debian (or for any ARMv5 arch for that matter).

MadsFox, 2017-10-04:
Thank you all @moriarty @dlech @bidaian @gavanderhoorn, got it up and running with a stretch image and succeeded with this tutorial:
http://wiki.ros.org/ROS/NetworkSetup
and is currently going through this one: 
http://wiki.ros.org/ROS/Tutorials/MultipleMachines

A last question relating to my further work:  
Is there a forum or website where it is possible to get as good support as I got from you here? or what would you recommend me going forward?

gavanderhoorn, 2017-10-04:
> Is there a forum or website where it is possible to get as good support as I got from you here? or what would you recommend me going forward?

I can't guarantee that you'll get "as good support as I got from you here", but the main support channel for ROS is [answers.ros.org](http://answers.ros.org).

For more info on where to get support on ROS related matters, see [ros/wiki/Support](http://wiki.ros.org/Support).

MadsFox, 2017-10-09:
Back again
Tested the setup using Python:

- Tried the listener talker test - everything worked.
- Tried to print values from a infrared sensor - everything worked. 
- Tried to write a scripts that takes values from a infrared sensor and publishes them on a topic - didn’t work.
Turns out, that ROS dosen't work with Python3 and EV3 dosen't work with Python2, that made me switch to C++, but missing some pieces to getting started. Would you recommend me setting up a ROS workspace on my master computer and then transfer the build files to the EV3? and is that even possible? 
Tried to compile the ev3dev-lang-demo file with the ev3dev.h in the same folder. Can you help me with what is missing in the logic or where can a find help to fill out the gaps? 
I see a issue with building a ROS workspace on the EV3 as I predict a heavy compilation for the EV3.
Hope you can help me again.

bidaian, 2017-10-09:
I had that python2/python3 problem with opencv and finally I installed python-evdev (instead of python-ev3dev). I had to change a couple of things in the code because the motor class was missing an attribute, but other than that, everything worked. Here you find the packages http://archive.ev3dev.org/debian/pool/main/ and you might have to install a dependency, I don't remember.

Please open new issues for new topics, for example, something like "please backport latest ev3dev to python2". It's easier to track and easier to find for other people with the same problem.

MadsFox, 2017-10-09:
I dont care how, just want this to work now.

MadsFox, 2017-10-09:
@bidaian Will that work on the stretch image that i am working on now? and how do i install that package?

bidaian, 2017-10-09:
Yes, it will work on stretch. I downloaded the debs and installed them manually.
Try this: (I don't have the brick at hand now but I think this is it)

```
$ wget http://archive.ev3dev.org/debian/pool/main/p/python-evdev/python-evdev_0.4.7-0ubuntu4ev3dev1_armel.deb
$ sudo dpkg -i python-evdev_0.4.7-0ubuntu4ev3dev1_armel.deb
```

Basically, you are downgrading ev3dev but still run stretch. Maybe @dlech can tell you the minimum  packages you need to install (it's no more than a couple).

Then, remember to use the documentation for the older version of ev3dev, because some APIs changed.

WasabiFan, 2017-10-09:
I'm a bit confused about what you are trying to do here. The `evdev` package isn't specific to ev3dev -- they just have names with similar letters. `evdev` _does_ allow you to interface with device drivers, but that's just a raw means of setting internal values generic to Linux's driver style as a whole. I don't think that is really what you are looking for. I suspect you actually referring to the Python 2 version of the python-ev3dev package; that is entirely unsupported and a fair amount of functionality likely won't work. There is also no proper documentation for that.

We dropped support for Python 2 a while ago and won't be backporting or reverting. Debian moved to primarily Python 3 with Stretch and we are following along; at some point, hopefully ROS will catch up. Until then, you can either access the drivers manually or use C++.

bidaian, 2017-10-09:
Yes, sorry, that's what I meant. Thanks for pointing it out! The python2 version of python-ev3dev package.
If the user can do without the missing functionality, then I think that using the python2 is worth a look. I mean, I tried for hours to find a way for opencv to work on python3 and the solutions were so complicated that I ended up downgrading. And it was ok for what I had to do (move motors and read sensors, a line follower using a camera).

MadsFox, 2017-10-09:
Can you answer me if its the same docker image i can use to compile the code for the brick?

dlech, 2017-10-10:
> Maybe @dlech can tell you the minimum packages you need to install (it's no more than a couple).

minimum packages for what?

dlech, 2017-10-10:
> @dlech Can you answer me if its the same docker image i can use to compile the code for the brick? Now that i have installed the stretch image?

You probably want the ev3dev/debian-stretch-armel-cross Docker image for cross-compiling for stretch. (ev3dev/debian-stretch-cross will only work if all of the dependencies are multiarch capable)

dlech, 2017-10-10:
Discussion for opencv should be directed to #803 (this thread is about ROS)

bidaian, 2017-10-10:
>>  Maybe @dlech can tell you the minimum packages you need to install (it's no more than a couple).

> minimum packages for what?

For the python 2 version of the python-ev3dev package.

bidaian, 2017-10-10:
> Discussion for opencv should be directed to #803 (this thread is about ROS)

I mentioned it because I run into the same mismatch (python2 library / python3 ev3dev) and solved with the python2 version of the package. Since @MadsFox was just testing things, I though that could help him for the moment. Sorry fo the noise then.

dlech, 2017-10-10:
There has not been a python2 compatible library for ev3dev for a long time. I doubt that there is any debian package for it floating around. There are discussions here and in the opencv issue about how we can get python3 compatible versions of ROS and and OpenCV as an alternative to a python2 compatible ev3dev library. If we want a solution, someone will have to step up and spend some time on it.

MadsFox, 2017-10-10:
Right now I have the Linux-debian-stretch up and running and concluded that Python is not an option when combining ROS and EV3. 

I have successfully installed:

- `apt-get install ros-robot`
- ev3dev-lang-cpp as explained [here](https://github.com/moriarty/ros-ev3/blob/master/brickstrap-build-status.md) and changed the CMakelists file manually. 
- Created a catkin_ws, but `echo $ROS_PACKAGE_PATH` gave nothing.

I proceeded to downloade @moriarty ev3dev_ros_demo package, but when trying to run `catkin_make` on the brick it can't find roscpp:

 CMake Warning at /usr/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "roscpp" with any
  of the following names:

    roscppConfig.cmake
    roscpp-config.cmake

  Add the installation prefix of "roscpp" to CMAKE_PREFIX_PATH or set
  "roscpp_DIR" to a directory containing one of the above files.  If "roscpp"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  ev3dev_ros_demo-master/CMakeLists.txt:6 (find_package)

 Could not find the required component "roscpp". The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /usr/share/catkin/cmake/catkinConfig.cmake:82 (find_package):
  Could not find a package configuration file provided by "roscpp" with any
  of the following names:

    roscppConfig.cmake
    roscpp-config.cmake

  Add the installation prefix of "roscpp" to CMAKE_PREFIX_PATH or set
  "roscpp_DIR" to a directory containing one of the above files.  If "roscpp"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  ev3dev_ros_demo-master/CMakeLists.txt:6 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/robot/catkin_ws/build/CMakeFiles/CMakeOutput.log".
Makefile:364: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

I know that the tutorial says to use brickstrap, but havnt found any instructions on how to install brickstrap, that's why in trying to make is on the EV3 itself. 
Am I making it harder for myself then I should? Or is this an impossible task I have started?

bidaian, 2017-10-11:
> There has not been a python2 compatible library for ev3dev for a long time. I doubt that there is any debian package for it floating around.

Just for the record, I'm using [this one](http://archive.ev3dev.org/debian/pool/main/p/python-ev3dev/python-ev3dev_0.6.0_all.deb) one, from the ev3dev repository, it works for me in the current project. But I do understand it probably isn't good enough for other people.

dlech, 2017-10-14:
> I know that the tutorial says to use brickstrap, but havnt found any instructions on how to install brickstrap, that's why in trying to make is on the EV3 itself.
> Am I making it harder for myself then I should? Or is this an impossible task I have started?

The EV3 is very limited on memory, so trying to build any large project on the EV3 is going to take a very long time if it even works at all. The current recommendation for cross-compiling is to use Docker. See <http://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/>. Except use the `ev3dev/debian-stretch-armel-cross` Docker image.

Brickstrap still exists, but it is now only used for converting a Docker image into a bootable SD card image. It can't be used for compiling.

MadsFox, 2017-10-17:
I finally got the ev3dev-lang-cpp library to work and now only need to send the data via a ROS topic. Following the directions @dlech gave me:

1. Flash an SD card with ev3dev-stretch
2. Boot it
3. Connect the EV3 to the Internet
4. Open an SSH terminal
5. `sudo apt update`
6. `sudo apt install ros-core-dev`

and setting up a docker image with `ros-core-dev` installed and using it to compile the code in a catkin_ws, where i have compiled the ev3dev-lang-cpp library as a project library. 
Small changes needed to be made to the library, to connect to the right ports in the stretch image [Making is work with ROS](https://github.com/ddemidov/ev3dev-lang-cpp/issues/29#issuecomment-335443486)
After compiling i transferred the project code(with compiled files) to the EV3 brick using a mounted folder(sshfs) and then ran the executable file on the EV3 brick via a ssh connection. 
Next step is to send the sensor data via topics and controle the motors via topics. 
So fare thank you all for helping: @dlech @moriarty @ddemidov don't hope i forgot anybody.

MadsFox, 2017-10-17:
I also created a docker image with the ros-core-dev installed and the ev3dev_test_node from ev3dev-lang-cpp in a package in a catkin workspace called compiler_ws in the compiler user folder: 
250684/ev3-stretch-ros