ev3dev/ev3dev #855: Bluetooth 'not available'

funcmap, 2017-03-04:
### What are you trying to do?
turn on _Bluetooth_

### What did you expect to happen?
_Bluetooth_ on and ev3 device discoverable

### What actually happened?
no _Bluetooth_ working; device says 'not available'.
using ssh over _USB_ tried to restart _BT_ stack; but unsuccessful.
```lsusb``` does not list any device except root hub. Isn't _BT_ on _USB_?

### What hardware and software are you using (including version numbers)?
```
Image file:         ev3dev-jessie-ev3-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2017-03-04:
This seems to happen sometimes on the first boot. Have you tried rebooting? Also make sure you don't have any sensors plugged in.

funcmap, 2017-03-04:
Thanks, so now it is visible. But pairing does not work. Using ```btmon``` I see a lot of messages being exchanged even a "connected", but a "disconnect" at the end. This was from Linux; will retry using Win.

StapleIT, 2017-08-06:
What was the outcome of this issue?  I had a similar issue using 'ev3dev-jessie-ev3-generic-2017-06-09' .  Everything worked fine over USB.  Then I got tired of being tethered (physically) to the EV3 by the USB cable, so tried to connect via bluetooth from my Windows10 PC.  Initially I could see the Bluetooth settings in the ' wireless & networks' menu on EV3, but I couldn't get the connection to work when following the instructions on ev3dev site.  Status kept coming up as 'failed' when trying to connect.  I was getting an additional message (can't remember wording) and asked to accept or reject it.  I had 'accepted' this a few times but then tried 'rejecting' to see if this was my issue . Still no joy.  Rebooted EV3 brick.  Suddenly bluetooth came up as ' unavailable' .  Tried rebooting PC and EV3 a few tiems; no change.  To check it wasn't some other issue, I rebooted without the SD card and found I could still connect via BT using the standard Lego Mindstorm program on my PC.  Thinking I must have inadvertently ' damaged'  the ev3dev install on the SD, I re-installed the same ev3dev  zip file onto the SD.  Rebooted.  Bluetooth still unavailable.  Is there s known way I can get bluetooth back again on my ev3dev boot?  Thanks  Addtional info:  kernel 4.4.68-20-ev3dev-ev3,  EV3 programmable brick, revision 0006

dlech, 2017-08-06:
There is no outcome. Bluetooth is still not working well. e.g. #925.

dlech, 2017-08-06:
It helps if you unplug all sensors during boot, the most of the time, Bluetooth will be available.

StapleIT, 2017-08-07:
Thanks, dlech.  I'm new to all this and assumed i'd broken it.  You guys have done a great job developing ev3dev such that the likes of me can use python to program ev3.

dlech, 2017-09-02:
Please try the latest ev3dev-jessie snapshot:

https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2017-09-02/

dlech, 2018-12-07:
ev3dev-jessie is a good as it is going to get and we have #925 for the same problem in ev3dev-stretch.

closed 2018-12-07 by dlech

wizard1066, 2019-07-02:
Having the same issue with bluetooth, using ev3dev-stretch. Tried rebooting multiple times. No sensors attached, no bluetooth. Tried unplugging everything, powering off and then on again. No bluetooth. Tried rebooting. No bluetooth. Nothing. Wanted to rule out H/W tried different SD card in the same brick. No BT. Suspecting hardware. Removed ev3dev SD card. Booted on Lego OS. Tried to enable BT. Doesn't work. Tried SD card in a different brick. It worked. Sending brick with BT back to LEGO!