ev3dev/ev3dev #856: Raspi B+ V1.2 + BrickPi3 cannot boot

jert47, 2017-03-06:
I'm trying to boot with kernel 19 on Raspi B+ v1.2 and BrickPi3.
I got stuck on "Started Update UTMP about System Runlevel Changes" during boot.
The only thing I've uncommented in config.txt is dtoverlay=brickpi3 but here is my entire config.txt.
Should I do anything more?

[config.txt](https://github.com/ev3dev/ev3dev/files/821048/config.txt)

dlech, 2017-03-06:
This is not stuck. We currently don't support using an HDMI display with Raspberry Pi. Although I am told that you can get a command prompt by pressing <kbd>ALT</kbd>+<kbd>CTRL</kbd>+<kbd>F6</kbd>.

jert47, 2017-03-06:
Thanks David. Now I see the HDMI limitation in the documentation.Thank you.
 
Jerry T
Carpe diem!!!

jert47, 2017-03-07:
The issue is closed but the problem remains. I tried to connect a monitor via a HDMI-to-VGA converter with the same result: "stuck" on  "Started Update UTMP about System Runlevel Changes".
With Raspi B+ v1.2 and BrickPi 1, how should config.txt be modified?

WasabiFan, 2017-03-07:
What @dlech was saying is that it _isn't_ stuck: it finished booting and is waiting for you to do something. Try [connecting with SSH](http://www.ev3dev.org/docs/tutorials/connecting-to-ev3dev-with-ssh/) as a first step.

jert47, 2017-03-07:
I see. What @dlech was saying was also that ev3dev currently cannot handle HDMI. That's why I assumed that I should go for VGA.
But this presents next question: what  does "do something" mean? I cannot find any info as what to expect from a EV3DEV/BrickPi system. There is no GUI, as I understand. There are some raspi files and some brickpi files. What can one do with them?Of course, people can use trial and error method, nut some enlightment wouldn't hurt.
 Jerry T

WasabiFan, 2017-03-07:
The point that he was making was generic to all display interfaces, not just HDMI. Ev3dev doesn't use the display ports because having your mobile robot show stuff on a wired display isn't particularly useful.

Ev3dev is designed to let you use LEGO motors, sensors and other devices from your own code. Pick a language from our "programming languages" page on our website and follow the instructions from there to learn how to write code in your preferred language to use motors and sensors.

jert47, 2017-03-07:
OK, thank you, for now.
Jerry T

dlech, 2017-03-07:
> The issue is closed but the problem remains.

True story.

> how should config.txt be modified?

On your host computer, right after you flash the SD card, before you put it in the RPi.

> But this presents next question: what  does "do something" mean?

Not very many people have a BrickPi3 yet, so no libraries support it out of the box. Fixing it is quite simple though. It is just a matter of the names of ports. Most people use [ev3dev-lang-python](http://python-ev3dev.readthedocs.io/en/stable/?badge=stable) to write programs. It works with BrickPi3, but like I said, you have to use the port names from [this table](http://www.ev3dev.org/docs/platform-comparison/) since there are not built-in constants for BrickPi3.

I recently made a simplified version of [this useless machine](http://jkbrickworks.com/luuma-ev3) using BrickPi3 with ev3dev and wrote a [simple program](https://gist.github.com/dlech/37d4a4aca664c38458d85512390a521b) for it.

dlech, 2017-03-07:
This has come up several times before, but apparently not in the main ev3dev issues. As I have said in the past, I don't have a display to hook to my RPi, nor am I particularly interesting in buying one for myself. So, someone else is going to have to fix this.

There are couple of options:
* Figure out how to detect if there is no compatible display on RPi and don't run Brickman.
* Make Brickman work on the HDMI display.
* Send me an HDMI display and I will do something about it.

dlech, 2017-09-10:
Starting with Brickman 0.10.0 (ev3dev-stretch only), you should see Brickman come up on the display, so it won't appear as stuck anymore.

https://github.com/ev3dev/ev3dev/issues/956#issuecomment-328325190

closed 2017-09-10 by dlech