ev3dev/ev3dev #859: Lego Boost, a threat to EV3?

ndward, 2017-03-13:
As you probably know, a new Lego robotics kit, Lego Boost, will be released in August 2017. See this preview from CES: http://www.huffingtonpost.com/dan-nessel/lego-boost-first-look-a-r_b_14060182.html. Relative to EV3, it seems to offer a similar range of sensors and motors but makes it possible to build robots that are twice as large for about half the cost of EV3. Lego Boost seems to target a younger age range (7-12), with a focus more on building and less on coding. Lego Boost programming is block-based and I assume is not compatible with EV3-G, is not Linux-based, and is not compatible with EV3dev. Being so much cheaper than EV3, this kit is bound to negatively impact sales of the EV3, which can only be bad news for the EV3dev project. But perhaps one can still hope that one day there will be a replacement for the EV3 aimed at older kids which WILL be Linux-based and WILL maintain compatibility with EV3-G and EV3dev? Any thoughts?

moriarty, 2017-03-13:
I think the target audience being younger kids is nice to get them into robotics, but no threat to the EV3... which is more powerful and well used by universities all over the globe. Universities by NXTs and EV3s in huge classroom sets, and the Lego Boost is no option for university courses.

It may negatively impact sales of the EV3, and I'm interested how it works and may recommend it when people with kids ask me about robotics platforms, because the EV3 is the most reliable, but it's expensive.

On my campus, it was hard to get them to switch from the NXT to the EV3 because they already have 100 NXT kits, and in fact, the NXT still dominates campus :-(

And, with BrickPis available, Universities are more likely to switch their class sets to BrickPis and use their original NXT Motors and Sensors, while they wait for the next EV3 replacement. That's a bigger threat to Lego, but as long as EV3dev supports the RaspberryPi or BeagleBone it's a strength for EV3dev.

Lego is a major sponsor of RoboCup Junior, and they are aware of the challenges which the RoboCup Organisation sets in the competition, and I don't see the Lego Boost being used for these leagues. I do see it as a great less expensive option to younger kids or lower level classrooms, getting them hooked on robotics sooner.

I think there exists enough room in the market for the Boost, and the EV3. There is even more room in the market on the other side of the EV3, between the EV3 and the next reliable classroom robot... 

And if Lego is reading this: it would be nice to see another option above the EV3 with the classroom tested reliability of Lego, but the power of an NVIDIA Jetson, and price tag similar to a TurtleBot ;-)

markyi370, 2017-05-16:
Do you think it will be possible to run ev3dev on the Boost (obviously porting required)?

dlech, 2017-05-16:
No, I don't think it is possible. I don't know the exact specs, but I think it is a sure bet that it is not capable of running Linux. Remote controlling it on the other hand is #521.

ndward, 2017-12-26:
I note that some people are already programming the Boost with Python: https://youtu.be/oqsmgZlVE8I
In the presenter's words (https://github.com/undera/pylgbst):
Python library to interact with Move Hub. Move Hub is central controller block of LEGO® Boost Robotics Set. In fact, Move Hub is just Bluetooth hardware, all manipulations are done with commands passed through Bluetooth Low Energy (BLE) wireless protocol. One of ways to issue these commands is to write Python program using this library. Best way to start is to look into demo.py file, and run it (assuming you have installed library). If you have Vernie assembled, you might run scripts from vernie directory.

**Features**

- auto-detect and connect to Move Hub device
- auto-detects peripheral devices connected to Hub
- constant, angled and timed movement for motors, rotation sensor subscription
- color & distance sensor: several modes to measure distance, color and luminosity
- tilt sensor subscription: 2 axis, 3 axis, bump detect modes
- LED color change
- push button status subscription
- battery voltage subscription available
- permanent Bluetooth connection server for faster debugging

closed 2020-08-17 by dlech