ev3dev/ev3dev #860: 180 degree turn using gyro sensor

agentS, 2017-03-13:
### What are you trying to do?

I am trying to make a robot do a 180 degree as shown by the image below.

[image]

I am using the gyro sensor for measuring the degrees the robot has driven.

### What did you expect to happen?

The robot should be able to detect the line with his colour sensor again after the reversal.

### What actually happened?

The robot accomplishes the curve A successfully. However, the robot is unable to drive the second curve, as the robot is not able to both drive a certain amount of degrees and to drive in reverse direction simultaneously.

### What hardware and software are you using (including version numbers)?

EV3 Brick
ev3dev Kernel 17
(C++ bindings)[https://github.com/ddemidov/ev3dev-lang-cpp]

### Source code

    void Robot::turn_around(turn_around_direction direction)
    {
        int target_degrees = DEGREES_TURN_AROUND;
        if (direction == TA_RIGHT)
        {
            target_degrees *= (-1);
        }

        do
        {
            // TODO: replace by function pointer
            if (direction == TA_RIGHT)
            {
                turn_right();
            }
            else if (direction == TA_LEFT)
            {
                turn_left();
            }
        } while (rotation_sensor.value() < target_degrees);
        
        turn_around_drive_straight();
        int rotation_sensor_value = 0;
        do
        {
            // TODO: replace by function pointer
            if (direction == TA_RIGHT)
            {
                turn_left_reverse();
            }
            else if (direction == TA_LEFT)
            {
                turn_right_reverse();
            }
            rotation_sensor_value = abs(rotation_sensor.value());
        } while (rotation_sensor_value > DEGREES_TURN_AROUND_SECOND_FORWARD_CURVE);
    }

    void Robot::turn_left()
    {
        motor_left.set_position(0);
        motor_right.set_position(0);
        motor_left.set_position_sp(DEGREES_TURN);
        motor_right.set_position_sp(-DEGREES_TURN);
        motor_left.set_speed_sp(SPEED_MAX);
        motor_right.set_speed_sp(SPEED_MAX);

        motor_right.run_to_rel_pos();
        motor_left.run_to_rel_pos();

        while
        (
            (motor_left.state().count("running"))
            ||
            (motor_right.state().count("running"))
        )
        {
            this_thread::sleep_for(chrono::milliseconds(10));        
        }
    }

    void Robot::turn_right_reverse()
    {
        motor_left.set_position(0);
        motor_right.set_position(0);
        motor_right.set_position_sp(DEGREES_TURN);
        motor_left.set_position_sp(-DEGREES_TURN);
        motor_right.set_time_sp(DEGREES_TURN);
        motor_left.set_time_sp(-DEGREES_TURN);
        motor_right.set_speed_sp(-SPEED_MAX);
        motor_left.set_speed_sp(-SPEED_MAX);

        // Can a motor make both a curve and drive reverse simultaneously?
        motor_right.run_timed();
        motor_left.run_timed();

        while
        (
            (motor_left.state().count("running"))
            ||
            (motor_right.state().count("running"))
        )
        {
            this_thread::sleep_for(chrono::milliseconds(10));        
        }
    }

dlech, 2017-03-14:
> motor_left.set_time_sp(-DEGREES_TURN);

You cannot have a negative time setpoint. Also, time is not measured in degrees. It is likely that you need a different constant here.

WasabiFan, 2017-03-14:
I think you want to use `run_to_rel_pos()` instead of `run_timed()`. You aren't trying to spin for a certain amount of time -- instead, you want the motors to rotate a certain distance, correct? If so, you can switch that command and remove the `set_time_sp` calls. The time setpoint ("sp") is only used in `run_timed` mode.

agentS, 2017-03-14:
Sorry, the `motor_left.set_time_sp(-DEGREES_TURN);` was a copy&paste error.

I already tried to accomplish that behaviour using `run_to_rel_pos()`. However, as the documentation says when using a `run_to_*_pos` method the sign of the speed is ignored so that I can't make the robot drive reverse. When trying to accomplish the required behaviour using `run_to_rel_pos()` the robot moves as shown by the image below.

[image]

WasabiFan, 2017-03-14:
The sign of the speed is ignored because it's the sign of the `position_sp` that is used. It wouldn't make much sense to spin to a forward position at a reverse speed, so the sign of the speed is ignored and it will run at the given absolute speed in the necessary direction.

agentS, 2017-03-21:
Am Mon, 13 Mar 2017 21:57:42 -0700
schrieb Wasabi Fan <notifications@github.com>:

> I think you want to use `run_to_rel_pos()` instead of `run_timed()`.
> You aren't trying to spin for a certain amount of time -- instead,
> you want the motors to rotate a certain distance, correct? If [...]

I had to work on other issues. Sorry for the delayed response.

I have tried to run both motors to a negative relative point. However,
when trying instructions listed below the robot just continued it's
half circle.

    motor_left.set_position_sp(-DEGREES_TURN);
    motor_right.set_position_sp(-DEGREES_TURN);
    motor_left.set_speed_sp(SPEED_TURN);
    motor_right.set_speed_sp(SPEED_TURN);

    motor_right.run_to_rel_pos();
    motor_left.run_to_rel_pos(); 

So, is there any possibility to reverse the robot the way I illustrated
it?

closed 2020-08-17 by dlech