ev3dev/ev3dev #861: AutoShutdown when low power

FlopGithub, 2017-03-15:
Hi,

I use a 7.4V LiPo to power my EV3.

Yesterday, I leave my LiPo plugged and it reached a very low voltage level that is known to break the LiPo.
A very simple and interesting features ( for my usage ) would be to be able to set a low voltage limit that causes the EV3 to shutdown.

I don't know if here is where I can post this request !

WasabiFan, 2017-03-15:
@dlech is there a prebuilt way of doing this or would you need to put a power check on a timer?

dlech, 2017-03-15:
The is a program called `upowerd` that does power monitoring sorts of thing. However, when I tried it in the early days of ev3dev, I found that it uses up way too many system resources. So, I was thinking it is best to build our own solution into brickman. But maybe it would be better to build something more generic that could be used without brickman.

But yeah, whatever we come up with will have to have a timer to poll the power state.

dlech, 2017-09-14:
There is a new program in ev3dev-stretch called [birckd](https://github.com/ev3dev/brickd) that will shut down ev3dev devices on low battery.

closed 2017-09-14 by dlech