ev3dev/ev3dev #862: Coast mode when controlling motor with duty_cycle?

naimen, 2017-03-22:
### What are you trying to do?
I'm trying to reproduce the mountain car problem using a EV3, which requires the robot to build and perserve the momentum from driving down a slope.  I'm using the python library and controls the motor with 
"LargeMotor.run_direct(duty_cycle_sp=30, stop_action='coast')"

### What did you expect to happen?
The robot gains a lot of momentum from driving down a slope with power applied, and the momentum are perserved to help the robot drive up a hill. 

### What actually happened?
The motor brakes, the actual speed for driving down a slope with power applied is much lower than just coast down the slope. Even if I start the motor at the end of the slope, the momentum it have gained from coasting won't be perserved as the motor immediately brakes. 

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
Image file:         ev3dev-jessie-ev3-generic-2016-12-21
Kernel version:     4.4.47-19-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3

dlech, 2017-03-22:
Yes, this is expected. A duty cycle of 30 is turning the motor very slowly.

Duplicate of #327.

naimen, 2017-03-22:
Thank you for replying

Is there any workaround then? My supervisor tells me that for the old Lejos NTX, it's possible to set motor in "float" mode while power is applied, I wonder if that's possible for the EV3s as well.

If I just give more power to the motor, the robot could just do the climbing by it self, which kinda destory the point of my research.

dlech, 2017-03-22:
You would have to compile your own kernel to do this. See the discussion in #327.

naimen, 2017-03-22:
Ok, at least it's possible then.

So if I want to compile my own kernel and make the changes, how do I do it? where should I start and what files should I change?
The disussion in #327 seems overwhelming, can't really grasp what's going on.

dlech, 2017-03-22:
See https://github.com/ev3dev/ev3dev-buildscripts/ to build your own kernel.

The key to the rest of your questions is all in #327. I understand that it is quite a bit, but you can't expect to understand it all in 5 or 10 minutes. Just take it one piece at a time.

* https://github.com/ev3dev/ev3dev/issues/327#issuecomment-102716335 confirms your suspicion that there was "float" mode for the NXT - at least in 3rd party software.
* https://github.com/ev3dev/ev3dev/issues/327#issuecomment-102662057 helps us translate this to the more technical term "fast decay"
* https://github.com/ev3dev/ev3dev/issues/327#issuecomment-102638843 shows us where the code in question is and that we are using "slow decay"
* https://github.com/ev3dev/ev3dev/issues/327#issuecomment-102636291 has a truth table that should look suspiciously similar to the code. (One thing that is not obvious without looking at the schematics is that when a GPIO is set to input (floating) the PWM is passed through to the motor driver).

So, to recap so far, we have seen that when the motor is running forward, we pull pin 1 high and there is a PWM on pin 2, so according to the truth table, that means we are toggling between "Forward" and "Brake", which matches the definition of slow decay.

If you only need fast decay for your application and never slow decay, then the changes should be easy. We just need to make it toggle between "Forward" and "Standby" instead of "Brake". The truth table tells us we need to toggle pin 1 and keep pin 2 low. Something like this...
    
    gpio_direction_input(data->gpio[GPIO_PIN1].gpio);
    gpio_direction_output(data->gpio[GPIO_PIN2].gpio, 0);

A similar change will be needed for reverse.

If you need to add a switch where you can use both fast decay and slow decay, then the changes get a bit more complicated.

naimen, 2017-03-22:
Thank you very much, I'll try to modify this :)

dlech, 2017-03-22:
It is possible, but if you don't already know how to do that, IMHO it is much easier to compile the whole kernel.

laurensvalk, 2017-04-02:
Instead of modifying the software, you may also reconsider whether you need to.

The existing behavior mimics a simple DC motor controlled by a voltage in the range [-Vmax, Vmax], which you can control using a number [-100, 100]. Just like a DC motor controlled at 0V (a short circuit), it will brake rather than coast. 

How are your actuator constraints and optimization cost function specified? I'm asking because duty cycle (voltage) is not directly representative of power/energy/torque, in which such criteria are often specified.

naimen, 2017-04-03:
I'm using the tacho counters in the motor as oboserversion variables, i,e using tacho count to determin position and tacho count per secound to determin it's velocity. 
The algorithm I'm using is SARSA[https://webdocs.cs.ualberta.ca/~sutton/book/ebook/node89.html](url),  so it's discrete actions of forward, back and neutral with a predefined duty cycle`[30, coast, -30]`. 

The cost function are  the weighted sum of parameters for each action given it's current oberservation, and chose the action with lowest cost. I'm using a grid-tiling to extract features from observations.

Currently my solution to the problem is to increase the duty cycle when going down a slope, and then let it decay. So the robot still needs to build "energy" up to solve the original problem.

laurensvalk, 2017-04-03:
With that set of actions, I don't think you'd need to modify the kernel either. 

You can set it to either -30/30 in "run-direct" mode during the power modes, and send the stop command with stop mode "coast" when you want coast.

I'm not sure what the according Python library commands are, but you can try out the various configurations [using the command line first](http://www.ev3dev.org/docs/tutorials/tacho-motors/).

I don't have an EV3 near me to try, but you can try the following
```
echo run-direct > $MC/command
echo 30 > $MC/duty_cycle_sp
# wait a while...
echo coast > $MC/stop_action
echo stop > $MC/command
```
It might not be necessary to reconfigure the stop action each time.

naimen, 2017-04-03:
Thanks for replying, the problem that I'm experiencing is when power applied to the motor, the motor immediately changes it's rotation speed and disregarding whatever "momentum" it have gained beforehand. I have made a graph that shows how the speed and power are related in the experimenting world. Position 1000 is the bottom of the valley.
[image]
So coasting can get the a maximal speed compared to 50 duty cycle, while 30 duty cycle is around half of that, so nothing is gained by driving down a slope with power applied, the speed actually decrease when driving down. 
On the other hand, I've tried to start the motor when the robot is at it's maximal coasting speed, but the speed just jumps down to 30's level and the robot tilts forward.

laurensvalk, 2017-04-03:
> The problem that I'm experiencing is when power applied to the motor, the motor immediately changes it's rotation speed and disregarding whatever "momentum" it have gained beforehand.

As mentioned above, applying a duty cycle (voltage) does not necessarily apply torque/power in the desired direction. When the applied voltage is less than the back EMF due to the motor speed, you're actively slowing the motor down.

This is not a limitation of the EV3/ev3dev, but it is due to the way a DC motor works. Have a look at the standard equations of a DC motor, which demonstrate how torque, speed and voltage are related.  

Perhaps similar to the mountain car problem, I've seen several demonstrations of the pendulum swing up problem. Typically, the maximum control torque is limited, rather than the voltage.

You can do this as well, by changing your SARSA action space to [-Tau_max, 0, Tau_max], and at a lower level map this to the corresponding duty cycle, based on the motor speed.

naimen, 2017-04-03:
Thank you for this information, I have looked at the torque equation, it's seems that the torque `t` is calculated as
`t=k_t*(V-k_i*w)/R`
where `k_t`,`k_i` are some constant, `V` is the voltage supply, `w` is speed, and `R` is the resistance of the motor. Do you perhaps know how much is `R` for an EV3 large motor?

laurensvalk, 2017-04-03:
Have a look [here](http://nxt-unroller.blogspot.nl/2015/03/mathematical-model-of-lego-ev3-motor.html), and perhaps [here](http://philohome.com/motors/motorcomp.htm).

dlech, 2018-04-15:
Good information, but nothing to do here, so closing.

closed 2018-04-15 by dlech