ev3dev/ev3dev #863: How to execute get sound by the minijack output from a raspberry pi?

jabrena, 2017-03-23:
### What are you trying to do?

I am trying to execute beep with a PiStorms/Brickpi+ and listen the sound with a speaker connected with the mini jack.

### What did you expect to happen?

Listen a beep

### What actually happened?

I don´t listen nothing.

### What hardware and software are you using (including version numbers)?

I read this post:
https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=40872

but when I execute this command, I receive an error.

``` 
robot@ev3dev:~$ sudo amixer cset numid=3 1
amixer: Control default open error: No such file or directory
``` 

How to solve the problem?

Many thanks in advance.

``` 
robot@ev3dev:~$ uname -r
4.4.47-19-ev3dev-rpi2
```

Juan Antonio

dlech, 2017-03-24:
There is nothing special about ev3dev here. There is a line in `config.txt` to enable the sound driver.

The `beep` command won't work though. The driver doesn't support it.

jabrena, 2017-03-24:
What linux method could I use on a raspi with BrickPi+/PiStorms to play a tone or play a sound?

dlech, 2017-03-24:
https://lmddgtfy.net/?q=raspberry%20pi%20beep

jabrena, 2017-03-25:
Hi @dlech, many thanks I re read the comment to update config.txt and it works.

Maybe using some sounds from here, it is possible to replace native beep:
https://www.soundjay.com/beep-sounds-1.html

Many thanks

Cheers

closed 2017-03-25 by jabrena