ev3dev/ev3dev #864: Motors not detected ( Sensor to, I guess )

mrpierrot, 2017-03-24:
Hi!

### What are you trying to do?
I'am trying to run NXT Motors on my BrickPI3

### What did you expect to happen?
A Motor is detected in /sys/class when I plug in it.

### What actually happened?
I have plug the motor in MA port.
```
robot@ev3dev:~$ for f in /sys/class/tacho-motor/*; do echo -n "$f: "; cat $f/address; done
/sys/class/tacho-motor/motor0: spi0.1:MA
/sys/class/tacho-motor/motor1: spi0.1:MB
/sys/class/tacho-motor/motor2: spi0.1:MC
/sys/class/tacho-motor/motor3: spi0.1:MD
```

Here the dmesg : https://gist.github.com/mrpierrot/dd27f60a1ceb77a1ceb376d9f69cf1e7
I tried plug/unplug the motor but nothing changed.

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-rpi2-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-rpi2
Board:              Raspberry Pi 3 Model B Rev 1.2
Revision:           a02082
Brickman:           0.8.1
ev3devKit:          0.4.3
```

Note : I can run the motor when I use Raspberry For Robot OS of DexterIndustries.

Can you help me?
Thank!!!

mrpierrot, 2017-03-24:
In the doc, I read this : 

> The BrickPi3 has four output ports, labeled M1, M2, M3 and M4, for driving motors or other devices. The ports are similar to the output ports on the EV3, except that they cannot automatically detect when a motor is connected. By default, the lego-nxt-motor driver is loaded, so you don’t need to manually set the mode or device unless you want to use something else.

cf : http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/brickpi3.html#brickpi3-out-port-modes

Motors are not recognized by default? 
I should change the driver **lego-nxt-moto**r by **brickpi3-out-port** ?
How I can do that?

Thank you.

dlech, 2017-03-24:
It looks like everything is working as expected. You have shown in your first comment that the motor drivers have loaded. If you have a motor connected to port MA, then this should run the motor:

    echo 500 > /sys/class/tacho-motor/motor0/speed_sp
    echo run-forever > /sys/class/tacho-motor/motor0/command
    echo stop > /sys/class/tacho-motor/motor0/command

mrpierrot, 2017-03-24:
It's work!
I use ev3dev-lang-js and It doesnt work when Motor not flag as "connected"
Not a problem, I can code a small js drivers interface.

Thank you for this reply!

WasabiFan, 2017-03-24:
The `connected` property is `true` if the library was able to find a motor to use. The fact that it's `false` means that you likely haven't passed the correct parameters to the constructor or that you are calling the incorrect one.

If your problem is resolved, please close this issue.

mrpierrot, 2017-03-24:
If I do not say nonsense, the library mark `connected` at `true` if `/sys/class/tacho-motor/motor0/address` is `outA` ( For the motor A ) But with my setting and a motor on port A, the actual value of `address` is `spi0.1:MA` not `outA`

currently I defined the motor like this : 
`const motor = new ev3dev.Motor(ev3dev.OUTPUT_A);`
I made a mistake?

I will understand, I will close the issue after :-)
Thank you.

dlech, 2017-03-24:
Use `'spi0.1:MA'` instead of `ev3dev.OUTPUT_A`

mrpierrot, 2017-03-24:
Thank you. 
Maybe specified this case in the doc? :-)

closed 2017-03-24 by mrpierrot

raphmc, 2018-05-31:
Had the same problem. Thanks for asking this question before me!
And YES I totally support adding this part to the documentation ;)