ev3dev/ev3dev #865: Can't get any Sound to play (NameError on Sound)

pipothy, 2017-03-27:
### What are you trying to do?

I am trying to use the beep command (or tone, neither work). When I use any Sound command, I get the error "NameError: name 'Sound' is not defined". P.S. I heard something about audio user groups but I couldn't find any info on what this is or how to do anything. Any tips if this is likely it?

### What did you expect to happen?

I expected a beep to play.

### What actually happened?

I get the error "NameError: name 'Sound' is not defined".

### What hardware and software are you using (including version numbers)?

The Ev3dev flash file currently available, Python 3.4.2.

ddemidov, 2017-03-27:
It looks like this is an [ev3dev-lang-python](https://github.com/rhempel/ev3dev-lang-python) issue. Usually you get `"NameError: name '...' is not defined".` kind of errors when you forgot to import a module. Or may be you should add `ev3.` in front of the `Sound` as in
```py
import ev3dev.ev3 as ev3
ev3.Sound.beep()
```

pipothy, 2017-03-27:
That fixed it.

I am so sorry, I am a mess.

Thanks a lot!

closed 2017-03-27 by dlech