ev3dev/ev3dev #866: Using pyserial do connect to USB port

pablosampaio, 2017-03-30:
I am trying to connect serially to the USB type-A port of the EV3 using pyserial, but after trying a lot, I was not able. My first tests consisted in receiving data (not sendind) from an external device connected in the USB port.

So, this is my first question: is it possible to access the USB port to do (ad-hoc) serial communication?

I also don't know what device name ("/dev/???") I should use to access the port. I tried all of them with pyserial. To some of them, I was able to connect only after giving super-user privilege ("sudo"), but none of them received the data. (I called "inWaiting()" to check if there was any data available).

This is my second question: if it is possible, how to do that?

dlech, 2017-03-30:
What is the "external USB device" you are using? What is the output of `lsusb` on the EV3?

pablosampaio, 2017-03-30:
I am using a clone of Arduino UNO, programmed to simply output sequential values each 0.5 second. 

I don't have the EV3 here to copy its output now, but "lsusb" does show the microcontroller.

dlech, 2017-03-30:
Is there anything in `demsg` as well?

jabrena, 2017-03-30:
Hi @pablosampaio, 

Take a look this examples, maybe the scripts could help you a bit.
https://github.com/jmccormack200/RPLidar/blob/master/LidarPrint.py
https://github.com/SkRobo/rplidar/blob/master/rplidar.py

In both cases, the scripts communicates with a USB Device using serial library from Python.

Another example more specific for your scenario could be this one:
https://github.com/toddstavish/Python-Arduino-Serial-Text-Send-Receive/blob/master/test_serial_text.py

Anyway, I am not an expert with Python but maybe those links could guide you a bit.

Cheers

Juan Antonio

pablosampaio, 2017-03-31:
@dlech these are the outputs of lsub and demsg:

`**demsg**`, after I plug the Arduino-clone, these lines are added:
[  657.517113] usb 1-1: new full-speed USB device number 3 using ohci
[  657.736339] usb 1-1: New USB device found, idVendor=1a86, idProduct=7523
[  657.736425] usb 1-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[  657.736576] usb 1-1: Product: USB2.0-Serial

`**lsusb**` :
Bus 001 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

`**lsusb -v**` :
Bus 001 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0x7523 HL-340 USB-Serial adapter
  bcdDevice            2.54
  iManufacturer           0
  iProduct                2
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               96mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      1
      bInterfaceProtocol      2
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

pablosampaio, 2017-03-31:
Hi @jabrena , thank you for the reply. However, I didn't see how the scripts you linked could help me, sorry...

I have a very simple Python script working on Windows, and I would like to make it run on linux. At first, I think I only needed to know the correct port name ("\dev\???") of the USB plug. 

Now I am wondering if I need to install any driver for Arduino in the ev3dev, or something like that...

dlech, 2017-03-31:
It looks like we need to enable a driver in the kernel configuration:

```
CONFIG_USB_SERIAL=m
CONFIG_USB_SERIAL_GENERIC=y
CONFIG_USB_SERIAL_CH341=m
```
http://blog.crox.net/archives/34-Using-the-HL-340-USB-serial-adapter-with-Linux-1a867523,-driver-ch341.html

dlech, 2017-04-03:
Fix has been released in kernel 4.4.59-19-ev3dev-ev3. To upgrade, make sure your EV3 is connected to the Internet, then...

```
sudo apt-get update
sudo apt-get install linux-image-ev3dev-ev3
```

closed 2017-04-03 by dlech

pablosampaio, 2017-04-04:
Well,  I still had no success. These are the outputs to the commands:

```
robot@ev3dev:~$ **sudo apt-get update**
Err http://security.debian.org jessie/updates InRelease
Err http://httpredir.debian.org jessie InRelease
Err http://archive.ev3dev.org jessie InRelease

<etc.>

W: Failed to fetch http://httpredir.debian.org/debian/dists/jessie/Release.gpg  Temporary failure resolving 'httpredir.debian.org'
W: Failed to fetch http://archive.ev3dev.org/debian/dists/jessie/Release.gpg  Temporary failure resolving 'archive.ev3dev.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.

``` 

```
robot@ev3dev:~$ **sudo apt-get install linux-image-ev3dev-ev3**
Reading package lists... Done
Building dependency tree
Reading state information... Done
linux-image-ev3dev-ev3 is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 22 not upgraded.

```

pablosampaio, 2017-04-04:
At first, I had some problems because my EV3 was not connected to the internet. I repeated the tutorial in the site, and then I was able to connect (using USB), and to install the updates. (That's why I posted a message then deleted it).

Then, I was finally able to plug my Arduino-clone in EV3 and read its data This opens many possibilities of (cheaper) sensors and modules to be plugged in EV3.

Thank you very much!

MogMiner, 2024-11-10:
What did you do to receive the data, the Serial library doesn't work on my EV3.
Can you please post your program? :-)

pablo-sampaio, 2024-11-13:
https://github.com/pablo-sampaio/swarm_ev3/blob/master/Common/RfidReader/__init__.py

MogMiner, 2024-11-13:
Thanks, I needed help because I'm making a Lego 3d_printer, I was using a ev3 and I found some servos knocking around, I found out I can control them with an Arduino connected to an my project.
Thanks again 😁😁😁👍👍!

pablo-sampaio, 2024-11-14:
Absolutely! Arduino is a great way to connect non-LEGO components (and usually much cheaper ones) to the EV3. In my project, I also used an Arduino to interface with an RFID reader. Good luck with your LEGO 3D printer project!

MogMiner, 2024-11-17:
my program is still running into issues:

import serial
ser = serial.serial("dev/ttyACM0", 9600)
while True:
    print(ser.readline())
    wait(1)

MogMiner, 2024-11-17:
I know I should probably start my own issue, but There is barely anyone else using the ev3dev port like this.

MogMiner, 2024-11-17:
Traceback (most recent call last):
  File "SerialPrint.py", line 2, in <module>
    import serial
ImportError: No module named serial