ev3dev/ev3dev #867: Connecting 3G modem to ev3dev

alexissusset, 2017-04-04:
### What are you trying to do?
Trying to connect a 3G Dongle and SIM card to ev3dev

### What did you expect to happen?
When connecting a 3G Dongle, a /dev/ttyUSB0 device should be created after running usb_modeswtich

### What actually happened?
usb_modeswtich runs correctly and lsusb reports the new device type but devices in /dev/ttyUSB* are not created, same behaviour by manually adding in udev rules

### What hardware and software are you using (including version numbers)?
Image file:         ev3dev-jessie-ev3-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3

### Notes
Would be great to build in support for 3G Dongles and dialup, so far added the following packages but without success:
apt-get install usb-modeswitch wvdial gsm-utils

we also have a udev+wvdial config script that didn't work:
https://soracom-files.s3.amazonaws.com/setup_air.sh

dlech, 2017-04-04:
:smiley: This is something I have always wanted to try. What is the output of `lsusb`?

alexissusset, 2017-04-05:
Hi @dlech, here's the lsusb output from before and after:
root@ev3dev:~# lsusb
Bus 001 Device 004: ID 12d1:14fe Huawei Technologies Co., Ltd. 
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

root@ev3dev:~$ usb_modeswitch -v 12d1 -p 14fe -J
[...]
 Found devices in default mode (1)
[...]
-> Run lsusb to note any changes. Bye!

robot@ev3dev:~$ lsusb
Bus 001 Device 003: ID 12d1:1506 Huawei Technologies Co., Ltd. E398 LTE/UMTS/GSM Modem/Networkcard
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

It looks like de device is switching correctly but it doesn't create /dev/ttyUSB0

dlech, 2017-04-05:
It looks like we need to enable `CONFIG_USB_NET_QMI_WWAN` in the kernel.

dlech, 2017-04-15:
OK, here is where we are at this point:

* I have the required kernel modules enabled in the EV3 kernel (both 4.4.x/jessie and 4.9.x/stretch)
* I have patched ofono in stretch to include a recent upstream fix for these modems and backported this to jessie.

So to get something working, for now, here is what you need to do...

* Either grab the [latest snapshot image](https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/) or [upgrade the kernel](http://www.ev3dev.org/docs/tutorials/upgrading-ev3dev/) on an existing image.

* `modprobe tun` (this is needed by `ofonod`). `sudo sh -c 'echo tun > /etc/modules-load.d/ofono.conf'` should make this happen automatically at boot.

* Install ofono.

        sudo apt-get update
        sudo apt-get install ofono ofono-scripts mobile-broadband-provider-info

* To get a Soracom Air SIM working:

        sudo /usr/share/ofono/scripts/create-internet-contrex soracom.io
        sudo /usr/share/ofono/scripts/set-roaming-allowed
        sudo /usr/share/ofono/online-modem

* Then you should be able to use the `brickman` user interface to connect and disconnect. Or, you can use `connmanctl` from the command line.

closed 2017-07-25 by dlech