ev3dev/ev3dev #868: Allow different user

MaxNoe, 2017-04-11:
### What are you trying to do?
Using another username than `robot`

### What did you expect to happen?
Stuff works

### What actually happened?

Filebrowser does now work because `/home/robot` is hardcoded there

The Filebrowser is just the first thing that i noticed not working if there is no `robot` user.

dlech, 2017-04-11:
What is the advantage of using a different user name?

MaxNoe, 2017-04-11:
It's kind of the point of installing a debian on these machines, right? To be able to do such things.

Settings hostnames works fine for example (and is needed for us since we use multiple bricks in the same project). 

So why not choosing a different username (or default path for the Filebrowser for that matter).

The issue could at the moment also be rephrased to `Allow setting default path for filebrowser`.

MaxNoe, 2017-04-11:
On a site note: It's very unusual to hard code things like file paths to me. This always screams for config files in my opinion.

kmuehlbauer, 2017-04-14:
Why not let the filebrowser start at `/home`? We have different users on the brick, and have to copy the python-scripts from their home-folder to `/home/robot` just to get them accessible from filebrowser. 

How is the workflow to be done right?

markyi370, 2017-05-16:
https://github.com/ev3dev/brickman/blob/abc25a5f1e920f4f109adcf0deaf3848e798137f/src/controller/FileBrowserController.vala#L30

That line decides the initial directory, from what I can tell.

MaxNoe, 2017-05-16:
Would be sane to use $HOME and $USER there.

markyi370, 2017-05-16:
Someone will have to try it.

(I can't, because all my microSD cards are busy at the moment)