ev3dev/ev3dev #869: "Host does not exist" PuTTY SSH on Windows 10

pipothy, 2017-04-13:
### What are you trying to do?

Connect using PUTTY SSH to my Lego Brick on Windows 10. The brick is connected via USB to my laptop.

### What did you expect to happen?

A terminal to appear asking me for login details.

### What actually happened?

"unable to open connection to ev3dev. Host does not exist". Pinging it works successfully, and the robot indicates that it is connected to the laptop and online via the USB connection from the brick to the laptop. Tried restarting everything also, just in case.

### What hardware and software are you using (including version numbers)?

Current release of image file of EV3Dev, Windows 10 fully updated.

dlech, 2017-04-13:
What are you using for the hostname? What command did you use to ping? Have you tried using the IP address instead of the hostname?

pipothy, 2017-04-13:
Hostname is ev3dev. I went into cmd and typed ping <ip address> (local IP address). It worked when I used the IP address instead so that's good enough for me. Not sure why ev3dev stopped working though!

roterBulle, 2017-07-07:
Hi,

same issue here, after upgrading from Server 2012R2 to Server 2016.
Before it was working without any issues, now i can only connect via IP.
I did a `ipconfig /flushdns` but that didn't help.
Also `ping hostname.domain.xy` and `ping hostname ` is working.
So it seems not to be related to DNS resolution. 
Does putty cache DNS somewhere?

Thanks

EverstormHDY, 2018-10-12:
How did you connect your brick to your PC?
If it´s via Bluetooth, make sure, you choose DHCP as IPv4 Method in the Settings. That always worked for me

dlech, 2018-10-12:
> Does putty cache DNS somewhere?

Not sure about this, but .here was a discussion on Gitter recently that seems to indicate that setting PuTTY to IPv4 instead of Automatic may help.

https://gitter.im/ev3dev/chat?at=5bad506e8196693171c5d126

iwillneverstop, 2019-06-30:
I have the same problem. I'm using `ev3dev-stretch-ev3-generic-2019-03-03.zip`. 

However, I can connect to my ev3dev using `ev3dev device browser` extension in Visual Studio Code. What's a strange problem :(

Purvi-modi, 2021-02-23:
> What are you using for the hostname? What command did you use to ping? Have you tried using the IP address instead of the hostname?

Using Ip address, it is working

AI-madman, 2021-04-20:
I have the same problem, however, I am trying to send a whole folder over the pscp -r command from my Windows 10 machine to the Ubuntu server in my uni.  This worked before, but after a windows update, it seems to have stopped working may be due to changes in the internal windows settings.

SamSirry, 2025-08-11:
It seems that this is a bug with PuTTY. If a host name does have an IPv6 DNS record (AAAA), and does not have an IPv4 DNS record (A), PuTTY seems to fail right away after attempting the IPv4 lookup and finding no A record.

P.s.: I've searched the currently reported bugs (https://www.chiark.greenend.org.uk/~sgtatham/putty/wishlist/#pending) and this issue does not appear reported.