ev3dev/ev3dev #870: not able to install packages on ev3dev

Ephesians, 2017-04-20:
### What are you trying to do?
install mosquitto, paho, on ev3dev OS

### What did you expect to happen?
I expected that sudo apt-get install mosquitto would find the package mosquitto

### What actually happened?
mosquitto is not a found package

Then I tried and got an error

sudo wget http://repo.mosquitto.org/debian/mosquitto-jessie.list--2017-04-20 01:27:25--  http://repo.mosquitto.org/debian/mosquitto-jessie.list
Resolving repo.mosquitto.org (repo.mosquitto.org)... 85.119.83.194, 2001:ba8:1f1:f271::2
Connecting to repo.mosquitto.org (repo.mosquitto.org)|85.119.83.194|:80... failed: Connection timed out.
Connecting to repo.mosquitto.org (repo.mosquitto.org)|2001:ba8:1f1:f271::2|:80... failed: Address family not supported by protocol.

### What hardware and software are you using (including version numbers)?
I am trying it all on MAC OSX and Windows 10 ... neither are working, and I am getting the same error. I am connected to the internet using usb on both OS. 

sudo apt-get update
Err http://archive.ev3dev.org jessie InRelease

Err http://archive.ev3dev.org jessie Release.gpg
  Unable to connect to archive.ev3dev.org:http:
0% [Connecting to httpredir.debian.org (128.31.0.62)] [Connecting to security.d
Err http://security.debian.org jessie/updates InRelease

Err http://security.debian.org jessie/updates Release.gpg
  Unable to connect to security.debian.org:http: [IP: 128.31.0.63 80]
Err http://httpredir.debian.org jessie InRelease

Err http://httpredir.debian.org jessie Release.gpg
  Unable to connect to httpredir.debian.org:http: [IP: 5.153.231.4 80]
Reading package lists... Done
W: Failed to fetch http://httpredir.debian.org/debian/dists/jessie/InRelease

W: Failed to fetch http://security.debian.org/dists/jessie/updates/InRelease

W: Failed to fetch http://archive.ev3dev.org/debian/dists/jessie/InRelease

W: Failed to fetch http://archive.ev3dev.org/debian/dists/jessie/Release.gpg  Unable to connect to archive.ev3dev.org:http:

W: Failed to fetch http://security.debian.org/dists/jessie/updates/Release.gpg  Unable to connect to security.debian.org:http: [IP: 128.31.0.63 80]

W: Failed to fetch http://httpredir.debian.org/debian/dists/jessie/Release.gpg  Unable to connect to httpredir.debian.org:http: [IP: 5.153.231.4 80]

W: Some index files failed to download. They have been ignored, or old ones used instead.
robot@ev3dev:~/mosquitto/mosquitto-1.4.11$

dlech, 2017-04-20:
There is something wrong with your Internet connection. It looks like DNS resolution is working, but something is happening when trying to connect to the site directly.

We have had some reports of universities blocking access to debian repositories, so this might be a possibility. Can you access other web pages from your EV3 (e.g. `wget google.com`)? Or even just ping these sites?

dlech, 2017-10-24:
`mosquitto` is installed by default in ev3dev-stretch starting with the 2017-10-24 snapshot

https://github.com/ev3dev/docker-library/commit/9dce29b9e7cebf57a963f5e96e29e6005d3effbb

closed 2017-10-24 by dlech