ev3dev/ev3dev #872: PiStorms V2, simple shape drawing in C++

floydbloke, 2017-04-24:
**### What are you trying to do?**
Draw simple graphics, e.g. rectangles, circles on the PiStorms screen in C or C++.
I have installed AJ Starks  OpenVG, compiled and made the sample programs successfully.
[https://github.com/ajstarks/openvg](url)

**### What did you expect to happen?**
E.g. the hellovg program to show a blue half-circle with hello world displayed

**### What actually happened?**
The program appears to execut fine, no errors but the screen stays in text mode, with a flashing cursor, until I hit a key and it returns to brickman.

**### What hardware and software are you using (including version numbers)?**

PiStorms V2 
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-rpi-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-rpi
Board:              Raspberry Pi Model B Plus Rev 1.2
Revision:           0010
Brickman:           0.8.1
ev3devKit:          0.4.3
```
As above I installed the OpenVG library.  To do so successfully I first 'installed' VC from here:
[https://github.com/ev3dev/raspberrypi-bootloader](url)
_When I say installed, I simply copied the VC folder and subfolders into /opt  .   Did I do that wrong?  I'm very much a beginner in most things Linux and Git_

as well installing libfreetyp6-dev and libjpeg9-dev

dlech, 2017-04-24:
FYI, we include the video core library in ev3dev. It is just in `/usr/lib/` instead of `/opt/`.

However, as far as I know, the video core stuff only works with HDMI (or composite) displays attached to the RPi. The PiStorms screen is basically an Adafruit 2.4" screen. You can check out the "Learn" page on the Adafruit site for more info.

Also, in ev3dev, we have a graphics library called GRX that works well with these displays. ev3dev-jessie uses [GRX 2.4.9](http://grx.gnu.de/grx249um.html) and ev3dev-stretch uses [GRX 3](http://docs.ev3dev.org/projects/grx/en/ev3dev-jessie/c-api/index.html).

floydbloke, 2017-04-25:
Thanks David
Looking at the GRX documentation it appears to do everything that I need it to.
I have been able to locate GRX3 for EV3DEV here:
[https://github.com/ev3dev/grx](url)

but I'm running ev3dev-jessie and I can't seem to find GRX 2.4.9.
Will GRX 3 work on jessie?  (I have been trying, but unsuccessfully so far.)

Also, beginner's question, do I simply clone these types of libraries and point the compiler at the rerlevant headers, or do I need to configure and 'make' them somehow first?

dlech, 2017-04-25:
You should be able to install the development files with `sudo apt-get install libgrx-dev` and include the header file `grx20.h` in your source files.

If you find a need to look at the v2.4.9 source code, it is on [this branch](https://github.com/ev3dev/grx/tree/grx2-ev3dev-jessie), but there is no need to clone it.

Technically, there is GRX3 on jessie, but I would not recommend using it. It is not really stable yet.

floydbloke, 2017-04-25:
Partial success.
I now have GRX2.4.9 installed and compiled the sample code in the documentation you linked to.
Unfortunately the program crashes back out to brickman on 
`GrSetMode( GR_default_graphics );`
I have tried to set the video driver in code by using:
`int GrSetDriver(char *drvspec);`
and trying _svgalib_, _linuxfb_ and _memory_ (as per the documentation).
The GrSetDriver command appears to execute OK but using the first two variables crash back to brickman (I think it flashes up an error, but it doesn't stay long enough for me to read it).  Using _memory_ the program doesn't crash.  It waits for a keypress but doesn't go into graphics mode.

Do I need to install a video driver of some sort, or have I perhaps missed a configuration parameter somewhere?

dlech, 2017-04-25:
`GrSetMode( GR_default_graphics );` should work. And as you have seen, we don't have a VGA card in the EV3, so no need for anything VGA-related. We do have a Linux framebuffer device. And of course, memory is just in memory, so nothing id displayed.

However, Linux consoles are tricky. You have to be running on a (local) virtual console to use graphics mode. I tend to use `sudo conspy` to be able to control the (local) virtual consoles using a (remote) SSH session. More explicitly....

    sudo chvt 6
    sudo conspy
    <login as robot:maker>
    <run graphics program>
    <press ESC three times to exit>
    sudo chvt 1 # back to brickman

There are also some other issue here that discuss using `sudo openvt ...`. It is quite a long sting of commands and options, so I will let you search for it if you are interested.

floydbloke, 2017-04-26:
The results are in, and...we have a winner.
Thank you so much David, it works as desired now.
One more small question please.  
With a keyboard attached to the Pi, once I have done `sudo chvt 6` to open a terminal I can type straight from the keyboard.  Is there any way to open a terminal from Brickman with the keyboard, rather than having to SSH in?

WasabiFan, 2017-04-26:
I requested such a feature in #173 but it was closed as _wontfix_.

dlech, 2017-04-26:
> Is there any way to open a terminal from Brickman with the keyboard, rather than having to SSH in?

Sure. Just press <kbd>ALT</kbd>+<kbd>CTRL</kbd>+<kbd>F6</kbd> on the keyboard to switch to a login console and <kbd>ALT</kbd>+<kbd>CTRL</kbd>+<kbd>F1</kbd> to switch back to Brickman.

floydbloke, 2017-04-26:
Superb.  Thank you once again.

closed 2017-04-26 by floydbloke

seth10, 2017-04-27:
@floydbloke I'm glad you were able to get it working! @dlech, you're the man 👍