ev3dev/ev3dev #873: pip3 missing

fferri, 2017-04-24:
I need to install some python 3 packages, but pip3 is not there. I look for Debian jessie pip3 install instructions, and it say to just install python3-pip apt package, but it is not found.

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-03-28
Kernel version:     4.4.47-19-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2017-04-24:
1. Make sure your EV3 is connected to the Internet.
2. Always `sudo apt-get update` before `sudo apt-get <anything else>`

fferri, 2017-04-24:
Yes, I did `apt-get update` before trying to `apt-get install python3-pip`

dlech, 2017-04-24:
What is the exact error message you are getting?

fferri, 2017-04-24:
After repeating the install steps, it worked.
I don't have a trace of what failed the previous time, I'm sorry.

closed 2017-04-24 by fferri

yvcode, 2018-07-23:
i tryed 
`sudo apt-get install python3-pip`
and i cant install pip either, the error is:
`E: Unable to locate package python3-pip`
can you help??

WasabiFan, 2018-07-23:
Did you run `sudo apt-get update` first? Run that command then try again.

L-apps, 2019-04-14:
U need to use sudo

DaCodingBoss04, 2019-05-15:
I have the exact same problem. After connecting my EV3 to the internet and doing update and installing it, pip is not found.

balharash, 2021-05-27:
I am executing "sudo apt-get update"
robot@ev3dev:~$ sudo apt-get update
Ign:1 http://ftp.debian.org/debian stretch InRelease
Get:3 http://security.debian.org stretch/updates InRelease [53.0 kB]
Get:4 http://ftp.debian.org/debian stretch Release [118 kB]
Get:2 http://archive.ev3dev.org/debian stretch InRelease [8432 B]
Get:5 http://ftp.debian.org/debian stretch Release.gpg [2410 B]
Get:6 http://security.debian.org stretch/updates/contrib armel Packages [1524 B]
Get:7 http://security.debian.org stretch/updates/non-free armel Packages [5320 B]
Get:8 http://security.debian.org stretch/updates/main armel Packages [658 kB]
Get:9 http://archive.ev3dev.org/debian stretch/main armel Packages [82.7 kB]
Get:10 http://ftp.debian.org/debian stretch/non-free armel Packages [51.3 kB]
Get:11 http://ftp.debian.org/debian stretch/contrib armel Packages [39.5 kB]
Get:12 http://ftp.debian.org/debian stretch/main armel Packages [6879 kB]
0% [Working]^C

It does stuck...( as you can see in the output from the command)
I tried 2 different routes to internet to avoid any blocking, with no success.
Removed the packages under /var/lib/apt/list/partials,with no success.

Hope you can support here , thanks in advanced.