ev3dev/ev3dev #874: Color sensor not plugged (brick.h)

almsv, 2017-04-26:
Hello,

I am using ev3dev-c in Lego ev3 31313 robots.

I am trying to check if my color sensors are plugged in the ports I indicate, using the function from brick.h " sensor_is_plugged( POOL_T pool, INX_T type_inx );". This function returns a 0 indicating that the sensors are not plugged but I can see in the robots display in Device Browser/Sensors that I have a lego-ev3-color connected.

I have check if the port I indicate is the same, and it is the same. I don't know what is the problem. 
I am using the function like sensor_is_plugged( INPUT_2, LEGO_EV3_COLOR);

What can be the problem? and/or How can I solve it?

Thank you in advance.

dlech, 2017-04-26:
cc: @in4lio

in4lio, 2017-04-27:
Thank you @dlech !

Hello @almsv -- Unfortunately, I do not have a Color Sensor right now. I'll check the problem in a few days. Is the sensor visible in the `lego-sensor` directory:  `/sys/class/lego-sensor/`?

in4lio, 2017-05-04:
Hello again @almsv 

I brought the library to the actual state -- I found a few changes in the kernel that I missed, checked and fixed all examples. Now it works with the kernel 4.4.61-20-ev3dev-ev3. Thank you for reporting this problem.

closed 2020-08-17 by dlech