ev3dev/ev3dev #875: ev3dev controlled by rspi 3

sismo93, 2017-04-27:
hi every one 
i want to control my ev3  running with brickman with my rasperry pie 3 , but i' m lost ,i didn't find  any informations on bleutooth protocole and how we can send and receive de informations and then execute de code can anyone help me ?

jabrena, 2017-04-27:
Hi,

It is a complex question.
Brickman doesn't have that feature using some programming language, is possible to do it.

Take a look this video, the developer used python + MQTT.
https://www.youtube.com/watch?v=oIzKvRp9guU&feature=youtu.be

Juan Antonio

JorgePe, 2017-04-27:
Do you really need BT?
If using Wi-Fi (and pyhton) you can use RPyC and access EV3 resources from the RPi:
http://ev3dev-lang.readthedocs.io/projects/python-ev3dev/en/stable/rpyc.html

sismo93, 2017-04-28:
Hi

Thanks for responding so fast  for remote controle with rpyc i need a wifi dongle on my ev3 ? I did install every think i use thonny for pytnon  3.5 i did the installation but i was with a ssh connexion can u explaine me how i do after the installation to controled with bleutooth i use mobaxterm too

sismo93, 2017-04-28:
For the video thats wath i want but with 1 mindstorm and un rspi3 i didn't find de way he managed to do it can you help me

JorgePe, 2017-04-29:
Both methods (RPyC and MQTT) need TCP/IP so Wi-Fi dongle ir required (it is possible to use TCP/IP over Bluetooth but honestly don't know if ev3dev has it implemented).

For MQTT we have this tutorial:
http://www.ev3dev.org/docs/tutorials/sending-and-receiving-messages-with-mqtt/

You will need a MQTT broker (mosquitto), can be your RPi, your EV3, your PC... If you are running Windows on your PC, probably it's easier to use the  RPi as the broker.
You also need MQTT client (paho-mqtt) on the RPi and the ev3dev.

For RPyC we have this short documentation page:
http://ev3dev-lang.readthedocs.io/projects/python-ev3dev/en/stable/rpyc.html

RPyC and MQTT aren't quite the same - MQTT is just a way to pass messages between several systems but you need to create you own programs to generate and understand those messages. RPyC allows us to invoke on a python client all methods available on a remote python "server". So probably it will be easier to use RPyC. I'm using it this weekend at our LUG event to control a "wolf pack" of EV3 and BrickPi (Raspberry Pi) rovers from my Ubuntu laptop.

dlech, 2017-04-29:
> (it is possible to use TCP/IP over Bluetooth but honestly don't know if ev3dev has it implemented).

ev3dev can do TCP/IP over Bluetooth and USB, no Wi-Fi required.

sismo93, 2017-05-03:
oki oki thanks all for respondings so fast i will buy a wifi dongle , for rpyc i don' t need a programme i controlled directly from my thonny intrphase ?

sismo93, 2017-05-03:
whats kind of dongle wifi is supported by ev3dev

JorgePe, 2017-05-03:
> whats kind of dongle wifi is supported by ev3dev

All that are supported by EV3 native firmware, like Edimax.
http://www.ev3dev.org/docs/networking/

I've been using Edimax, Asus USB-N10 Nano and ThePiHut adapter.

sismo93, 2017-05-04:
thanks a lot ,i jhave juste a last question can a connecte throughts ssh  wit§h na bleuthoot connexion ?

JorgePe, 2017-05-04:
Yes!
http://www.ev3dev.org/docs/tutorials/using-bluetooth-tethering/

The tutorial is incomplete, it lacks information for Linux and Windows. For Ubuntu I just followed the guide for Mac OSX and it worked but for Windows it may be a little more difficult.

You need to configure Bluetooth and Tethering on your ev3dev. Then pair from your computer and accept the pairing from ev3dev.
You will now have IP address, 192.168.0.1
Now from computer side, you need to establish a connection to the ev3dev BT PAN network. On Ubuntu (17.04) it was easy, just went to Networks, there was a network with the hostname of my ev3dev ("ev3dev" is the default), i connect to it and it appeared a message at ev3dev scree, just did "Accept" and I got "ping 192.168.0.1" working from my computer.
ssh worked fine after that
RPyC worked fine also... never thought on using RPyC over BT, that's pretty amazing @dlech

sismo93, 2017-05-04:
but for windows it's the same procedure we don't need to do some configurations because when i connect via ssh he is deconnectings after 30 sec how can i resolve this problem ?

sismo93, 2017-05-04:
can you explaine me difference of the procedure if i am using windows ?

JorgePe, 2017-05-04:
Sorry, I don't have a windows system to test, all my systems are linux based and I lost my virtual machine a few weeks ago, need to rebuild it.
It's strange, once you get a connection it should be stable (mine is). So maybe it's not a problem of the procedure - do you have something in your windows event logs?

dlech, 2017-05-04:
If you are connecting to Windows, you probably don't want to use tethering on the EV3. Rather you probably want to enabled Internet Connection Sharing on Windows so that you can connect to the Internet from the EV3.

If you are directly connecting an EV3 and a Raspberry Pi (both running ev3dev), then you will need to enable tethering on one of the two devices in order to make the connection work.

sismo93, 2017-05-15:
Hi everyone it,s work with my laptop now thanks to everyone now i am trying
to do it's with my rspi3 can i control it with rpyc throught bleutooth like
i did with the laptop with my rspi running on raspbian

closed 2020-08-17 by dlech