ev3dev/ev3dev #876: Which SD card should I buy?

dlech, 2017-04-27:
I'm kind of surprised that I didn't find this question in the issues already.

Really, this is more of an announcement than an issue. I wrote a blog post that does a comparison of some currently available microSD cards. https://lechnology.com/2017/04/microsd-card-benchmark-for-ev3-and-raspberry-pi-3/

I suppose we ought to link to this from somewhere. More on that later...

JorgePe, 2017-04-28:
Nice post.
I've been using mostly SanDisk and lately Toshiba, mostly based on my perception of boot time (and of course availability as we don't have many brands here).
But a few weeks ago I used LeJOS (to show to my LUG president the RMI concept, although I prefer RPyC) and I used Toshiba 16 and 32 GB and later model seemed noticeable faster, I wonder if size matters.
Will try your script with my microSD cards.

dlech, 2017-04-29:
Yeah, the Toshiba has really impressive read speeds, but really terrible write speeds. It will be interesting to see if you get the same result.

JorgePe, 2017-05-29:
Tested 3 Toshiba Exceria, all microSD HC I, Class (10):

8 GB = M301-EA = UHS Speed Class 1, read Speed 48 MB/s
16 GB = M302-EA = UHS Speed Class 1, read Speed 90 MB/s
32 GB = M302-EA = UHS Speed Class 3, read Speed 90 MB/s
(yes, same code as the 16 GB)

hdparm -  dd write - iozone 4k rr - rw
8 GB: 13.79      -  6.7       - 2.42  -  0.54
16 GB: 14.49      -  6.6       - 2.39  -  0.14
32 GB: 14.45      -  7.4       - 2.41  -  0.17

Got even worst write speeds than you (same image, same script). But strangely, the 8 GB does random writes much better than the other two.
So size (or UHS-3 class) matters for sequential read and writes but almost not relevant.
I guess I'll start using SanDisk.

dlech, 2017-05-29:
Thanks for sharing. Yes. all of the ratings (class 10, UHS-3, etc.) only take into consideration sequential speeds, so are really not useful. However, there is a new rating called "Application Performance Class" abbriviated A1 and so on. This will be something to look for as manufactures start adopting it.

(p.s. I edited your comment to make a pretty table).

dlech, 2018-04-15:
Closing since this info is on the wiki https://github.com/ev3dev/ev3dev/wiki/Selecting-a-microSD-card

closed 2018-04-15 by dlech