ev3dev/ev3dev #877: Setting up an EV3 as a Bluetooth client for controlling BT peripherals?

kortschak, 2017-04-30:
I have a couple of BuWizz blocks that I was hoping to be able to use as additional motor/other controllers via Bluetooth. I have ask the BuWizz people whether they can provide an API/protocol for communicating with the blocks, but I also need some background about how to set this up from the client end on the EV3.

What is the best way to attack this given that BuWizz publish their protocol?

dlech, 2017-04-30:
It depends on the Bluetooth service that it uses. I'm guessing that it probably uses serial port emulation (RFCOMM), in which case you probably use a tty device to communicate with it. If you want to get fancy, you could use the bluez dbus api to set everything up.

dlech, 2017-04-30:
I had a look at the website. Looks like it uses BLE. You should check out @jorgepe's blog. He has some posts about the s-brick. I imagine it will be very similar.

Example: http://ofalcao.pt/blog/2014/sbrick-remote-control-2

kortschak, 2017-04-30:
Thanks. I'll check that out and hopefully BuWizz will get back to me with some protocol information. Here is what btmgnt had to say:

```
hci0 dev_found: 5C:F8:21:xx:xx:xx type LE Public rssi -38 flags 0x0000 
AD flags 0x06 
name HMSoft
```

kortschak, 2017-04-30:
For Go interaction, github.com/paypal/gatt works to connect to the brick. So now just a matter of figuring out the protocol.

JorgePe, 2017-04-30:
Hi!
Yes, BuWizz needs BLE, like the SBrick. You can use a BT 4.0 BLE USB dongle with ev3dev on EV3, it's now full supported thanks to David  :)
You can use shell scripts and gatttool from the BlueZ stack or you can use python with PyBluez.
But you need information from BuWizz about the GATT services and how to use it. Vengit (SBrick) were kind enough to make that kind of information public but some other companies insist on keeping it private (the Sphero Force Band is one of those, I'm still tinkering with it).
If you have a smart phone there are some tools that can help you, like Nordic nRF Connect.
There are also some BLE sniffers in the market, I have one, it helped with LEGO WeDO 2.0 before LEGO made the SDK available, but they don't do all the work for us (again, ForceBand is still a mistery to me).
Feel free to make any questions :)

dlech, 2017-08-12:
Is this still and issue? Can this be closed?

kortschak, 2017-08-12:
Yes, this can be closed. BuWizz don't respond to any kind of support request, let alone one asking for them to give information about open sourcing the protocol (which they claimed to intend doing).

closed 2017-08-12 by dlech