ev3dev/ev3dev #880: Show console on EV3 when running via SSH

danielhp02, 2017-05-07:
### What are you trying to do?
I would like to be able to see console output from the robot's LCD screen when I run code over SSH, similar to when code is run directly on the robot. I can't find any documentation on this. How can I do this?

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

JorgePe, 2017-05-07:
Not sure if this is the most recent information but on this thread you get some information about using console 6 (the equivalent to press Ctrl + Alt + F6 on a linux system) and returning to console 1 after:
https://github.com/rhempel/ev3dev-lang-python/issues/194

You can also write directly to the screen (also explained in the same thread).

dlech, 2017-05-07:
I like to do this from a ssh shell...

    sudo chvt 6
    sudo conspy

Then you can basically remote control the console on the EV3 screen.

FYI, `sudo chvt 6` does the same thing as <kbd>ALT</kbd>+<kbd>CTRL</kbd>+<kbd>F6</kbd> but doesn't require a keyboard plugged into the EV3

Kurtoid, 2017-06-06:
So how can one get back to brickman?

closed 2020-08-17 by dlech