ev3dev/ev3dev #881: Tensorflow on EV3- Installation from Source

EyeTreasure, 2017-05-08:
### What are you trying to do?

Tensorflow is easy to set up to run on 64bit systems and can run with CPU support only but can also be made to work on 32bit systems, if installed from source. It has been done for other 32bit systems before:

http://cudamusing.blogspot.co.uk/2015/11/building-tensorflow-for-jetson-tk1.html

### What did you expect to happen?

Regrettably I am not experienced enough to even attempt this. I suspect it will be tricky and would require a lot of tweaking to get everything working and build the right dependencies. I also suspect only very simple Tensorflow programms can be run and they may be really slow. Perhaps training could be done with laptop support and the trained network could then run from the robot.

### What actually happened?

It has not been tried so far. So we do not know. There is a first for everything. So if there is a brave pioneer out here please give it a go and make us all really happy!

### What hardware and software are you using (including version numbers)?

It would be the latest debian build on EV3 with python 3.4.

Many thanks

Berend

dlech, 2017-05-08:
Yeah, that looks pretty tricky. I would suggest [using Docker to cross-compile](http://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/). I think that @jabrena has some experience installing Java 8. He might be able to point out a reference for that.

Another thing to check before spending too much time on this would be to make sure that TensorFlow and all dependencies support running on armv5te (with soft float). If is fairly common these days for software to only support a minimum of armv7 (with hard float).

WasabiFan, 2017-05-08:
> Another thing to check before spending too much time on this would be to make sure that TensorFlow and all dependencies support running on armv5te (with soft float). If is fairly common these days for software to only support a minimum of armv7 (with hard float).

The only ARM platform that they officially support is armv7 (it's what is in the Jetson) as far as I know. So I'd bet that you will run into an incompatibility there.

jabrena, 2017-05-08:
Hi @EyeTreasure, the idea about `TensorFlow` is nice but the first step is complicated in a EV3 Brick. Using Debian repos, I couldn´t install Java 8 directly. For BrickPi+/PiStorms the installation of Java 8/9 is super easy but for EV3 Brick, you need to install in a manual way.

This is the issue open at the moment to install JDK 8 on EV3:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/221

JDK 7 is possible to be installed on EV3, but it is old.

I have tried with Zulu JVM too, but that binaries doesn´t run. The another path which I didn´t finish is a manual debian installation. This week, I will try in a brick again (I hope not crash the distro as usual with this manual installations :)). 

If you find the solution, try to share with the community. It could be very appreciated.

Cheers

Juan Antonio

jabrena, 2017-05-08:
**Note:**
Although the idea to install Tensor Flow could be delayed, if you have ideas to use with EV3Dev & Robotics, it could be pretty interesting. Fresh ideas are welcome. :)

EyeTreasure, 2017-05-09:
Thanks for all the feedback. So it looks we will have to wait for a new
version of the EV3 brick or use a Raspberry Pi to run it locally on the
brick. Alternatively one could have a remote version of Tensorflow on a
PC/Mac which is controlling the robot.

Berend

jabrena, 2017-05-09:
Hi @EyeTreasure, 

Maybe we could collaborate in this installer and include Tensor flow as an option for BrickPi + Raspi3
https://github.com/ev3dev-lang-java/installer

In the case of EV3 brick, the hardware is limited.
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/214

Cheers

EyeTreasure, 2017-05-10:
> ev3dev-lang-java/ev3dev-lang-java#214

Juan

One might be able to compile Bazel version *0.1.0* with the deprecated JDK
7. This is possibly something to look into.

https://bazel.build/versions/master/docs/install.html

Best regards

Berend

jabrena, 2017-05-10:
In that case, the flow for EV3, it could be developed too.
I remember that I installed JDK 7 from Debian packages in the first versions of EV3Dev and Installed maven and Gradle too but... Compile a HelloWorld take minutes...

jabrena, 2017-05-14:
Hi @EyeTreasure, if you continue with the idea to install Tensorflow, I can help you a bit with this installer.
https://github.com/ev3dev-lang-java/installer/blob/master/README.md

Maybe you will save some hours.

If at the end you install Tensor Flow manually, It could be a good idea to add a new module in the installer.
https://github.com/ev3dev-lang-java/installer/tree/master/modules

Cheers

EyeTreasure, 2017-05-15:
Thanks I will give it a go at some point. So you have been able to install
Java 8 manually after all?

Berend

jabrena, 2017-05-15:
No, I tried last weekend to install `openjdk-8-jre` manually:
https://packages.debian.org/jessie-backports/openjdk-8-jre

But in the step to install `ca-certificates-java` for `openjdk-8-jre-headless` I couldn't continue.
https://packages.debian.org/jessie-backports/ca-certificates-java

This is the process that I stored:
https://github.com/ev3dev-lang-java/installer/blob/master/modules/experimental/java-ev3-java8-installer.sh

I think that at the moment, it is better if you follow this module:
https://github.com/ev3dev-lang-java/installer/blob/master/modules/java.sh

Cheers

jabrena, 2017-05-16:
Hi @EyeTreasure.

Today, I tried again and I finally I could install Java 8 JRE:

Add `deb http://http.debian.net/debian jessie-backports main` at the end of the file: `/etc/apt/sources.list` and execute `sudo apt-get update`

later, execute the following command:

```
sudo apt install -t jessie-backports --no-install-recommends openjdk-8-jre-headless ca-certificates-java -y
```

After some time... you will java openjdk-8-jre in ev3.
https://packages.debian.org/jessie-backports/openjdk-8-jre-headless

Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/221

```
robot@ev3dev:/etc/apt$ update-alternatives --config java
There are 2 choices for the alternative java (providing /usr/bin/java).

  Selection    Path                                            Priority   Status
------------------------------------------------------------
* 0            /usr/lib/jvm/java-8-openjdk-armel/jre/bin/java   1063      auto mode
  1            /opt/ejdk1.8.0/linux_arm_sflt/jre/bin/java       1         manual mode
  2            /usr/lib/jvm/java-8-openjdk-armel/jre/bin/java   1063      manual mode

Press enter to keep the current choice[*], or type selection number: 
```

Cheers

Juan Antonio

jabrena, 2017-05-16:
Hi @EyeTreasure,

After doing a JVM benchmark, it is necessary to continue using `Oracle JRE 8`, the performance is better in general. Take a look this jvm benchmark:
https://github.com/ev3dev-lang-java/jvm-benchmark/blob/master/README.md

EyeTreasure, 2017-05-20:
I am a bit confused here. I followed your steps to install
java-8-openjdk-armel.
Installation seemed OK but there is no javac command available. Then you
suggest to use Oracle JRE 8 since it performs better on benchmarking then
openjdk but how do you install this on EV3 brick?

jabrena, 2017-05-20:
Hi @EyeTreasure, I will try to clarify.

If you installed `openjdk-8-jdk-headless`, you should have available the command `javac` to run in the terminal, if you don´t have that command available, maybe we have to review the installation. In my EV3Brick, I have available:

- Oracle Java 8 JRE
- OpenJDK Java 8 JRE
- OpenJDK Java 8 JDK

Although I could install them, the performance to compile Java programs inside of the brick is Poor, this is the reason to not recommend the compilation inside of the brick.

This week, I did this script:
https://github.com/ev3dev-lang-java/jvm-benchmark/blob/master/scripts/compile.sh

For a HelloWorld.java, http://introcs.cs.princeton.edu/java/11hello/HelloWorld.java.html, the brick invested 183 seconds to compile, 77 seconds in the execution with OpenJDK JRE 8 and 7 seconds for Oracle JRE 8. This is the reason for that recommendation.

My suggestion is to move the jar from `Tensorflow` previously compiled from another computer with more power and execute with Oracle JRE 8 that it has a much better performance on the brick than OpenJDK. This strategy is not necessary if you use a BrickPi because if you use Raspberry Pi the hardware is pretty different. The SOC from Raspberry Pi 3 has better performance than the SOC from EV3 Brick:

**2010 Chip:**

``` 
SoC: Sitara Processor AM1808
CPU: ARM9 300MHz
RAM: 16KB of Instruction Cache, 16KB of Data Cache ,8KB of RAM (Vector Table), 64KB of ROM
```

**2016 Chip:**

``` 
SoC: Broadcom BCM2837
CPU: 4× ARM Cortex-A53, 1.2GHz
RAM: 1GB LPDDR2 (900 MHz)
```

Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/214