ev3dev/ev3dev #882: BrickPi3, sensors and motors do not repond to commands

suhrm, 2017-05-08:
Hi I have just upgraded to the new BrickPi3 and I have updated the firmware from Dexter Industries and enabled the BrickPi3 overlay.

dmesg | grep brickpi gives me the following output

[    6.311811] brickpi3 spi0.1: Mfg: Dexter Industries
[    6.325175] brickpi3 spi0.1: Board: BrickPi3
[    6.339799] brickpi3 spi0.1: HW: 3.2.1
[    6.340283] brickpi3 spi0.1: FW: 1.4.0
[    6.344058] brickpi3 spi0.1: ID: C175E224514D355231202020FF070F2E

Which should suggest that everything is configured as intended.

However, when I do a test to see if the motors are functioning e.g.

echo 500 > /sys/class/tacho-motor/motor0/speed_sp

echo run-forever > /sys/class/tacho-motor/motor0/command 

nothing is happening. I have tried something similar with the ultrasonic distance sensor also with no success.

Is there someone who can assist me?

HW and SW config:

Image file:         ev3dev-jessie-rpi2-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-rpi2
Board:              Raspberry Pi 2 Model B Rev 1.1
Revision:           a01041
Brickman:           0.8.1
ev3devKit:          0.4.3
BrickPi3 FW:        1.4.0
         HW:        3.2.1

dlech, 2017-05-08:
What is `cat /sys/class/tacho-motor/motor0/address`? Does it match the port you have the motor plugged into?

What were the exact commands that you used for the ultrasonic sensor?

suhrm, 2017-05-08:
Yes it corresponds to the port.

For the ultrasonic sensor I use the following:

# setup the port S1 as a ev3-uart type
echo ev3-uart > /sys/class/lego-port/port0/mode
# set device type
echo lego-ev3-us > /sys/class/lego-port/port0/set_device

dlech, 2017-05-08:
Oops, I missed checking you kernel version the first time around. You need to upgrade your kernel. 

http://www.ev3dev.org/news/2017/04/15/kernel-release-cycle-20/

Also, some motor features will not be available unless you upgrade you BrickPi3 firmware as well.

suhrm, 2017-05-09:
That fixed it! Thanks.

closed 2017-05-09 by suhrm