ev3dev/ev3dev #883: BeagleBone Black EVB HDMI output?

dwalton76, 2017-05-13:
Has anyone tried connecting the micro HDMI port on their EVB to a monitor? I'm giving it a try but all output goes to the backlit screen, even on a reboot.  I thought maybe there would be a setting in `/boot/flash/uEnv.txt` but the only thing there is the ability to enable/disable the entire cape so I don't think I want to tweak anything there.

If the answer is "We've never looked at this and it would be a lot of work to get it going" that is fine, this isn't high priority.  Am doing a demo at a school and was hoping to be able to display to a monitor to make everything easier to see.

dlech, 2017-05-13:
All of the HDMI pins are used for the EVB, so it is not possible to use HDMI (same goes for the EMMC).

dwalton76, 2017-05-13:
Ack, thanks for the info

closed 2017-05-13 by dwalton76