ev3dev/ev3dev #884: Problems to install a Debian package. "FATAL -> Failed to fork."

jabrena, 2017-05-13:
### What are you trying to do?

I am trying to install a bluetooth library on EV3. Recently I detected that in EV3 brick, the process to install a Debian package take many time and I am not sure what is the source of the problem. In this case, this is the second time that I read the response: FATAL -> Failed to fork.

I think that time to build the tree of dependency is quite long. This fact doesn´t happen on BrickPi with Raspi 3. I suppose that the problem is the amount of memory but I remember that in previous releases, I didn´t receive that problem.

Any idea?

### What did you expect to happen?

Install the package without any problem

### What actually happened?

robot@ev3dev:~$ sudo apt-get install libbluetooth-dev
[sudo] password for robot: 
Reading package lists... Done
Building dependency tree        
Reading state information... Done
Suggested packages:
  pkg-config
The following NEW packages will be installed:
  libbluetooth-dev
0 upgraded, 1 newly installed, 0 to remove and 342 not upgraded.
FATAL -> Failed to fork.

### What hardware and software are you using (including version numbers)?

```
Image file:         ev3dev-jessie-ev3-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2017-05-13:
It sounds like you are out of memory. What is the output of `free`?

jabrena, 2017-05-13:
Hi @dlech, I tried to install manually but I think that you created a custom library for the project:

``` 
robot@ev3dev:~$ sudo dpkg -i libbluetooth-dev_5.23-2+b1_armel.deb          
(Reading database ... 26541 files and directories currently installed.)
Preparing to unpack libbluetooth-dev_5.23-2+b1_armel.deb ...
Unpacking libbluetooth-dev (5.23-2+b1) over (5.23-2+b1) ...
dpkg: dependency problems prevent configuration of libbluetooth-dev:
 libbluetooth-dev depends on libbluetooth3 (= 5.23-2+b1); however:
  Version of libbluetooth3:armel on system is 5.23-2ev3dev2.

dpkg: error processing package libbluetooth-dev (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 libbluetooth-dev
``` 

In that case, **how to install the library?** 
What is the changes on libbluetooth3:armel 5.23-2ev3dev2?

Juan Antonio

jabrena, 2017-05-13:
I am going to repeat the process to get the `free` result

dlech, 2017-05-13:
> In that case, how to install the library?

    sudo apt-get install -f

> What is the changes on libbluetooth3:armel 5.23-2ev3dev2?

https://github.com/ev3dev/bluez/blob/ev3dev-jessie/debian/changelog

jabrena, 2017-05-13:
Hi @dlech, I found a faster way without using apt-get on EV3:

```
robot@ev3dev:~$ sudo dpkg --force-depends-version -i libbluetooth-dev_5.23-2+b1_armel.deb                        
(Reading database ... 26541 files and directories currently installed.)
Preparing to unpack libbluetooth-dev_5.23-2+b1_armel.deb ...
Unpacking libbluetooth-dev (5.23-2+b1) over (5.23-2+b1) ...
dpkg: libbluetooth-dev: dependency problems, but configuring anyway as you requested:
 libbluetooth-dev depends on libbluetooth3 (= 5.23-2+b1); however:
  Version of libbluetooth3:armel on system is 5.23-2ev3dev2.

Setting up libbluetooth-dev (5.23-2+b1) ...
```

closed 2017-05-13 by jabrena