ev3dev/ev3dev #885: Docker Cross-compiling

Master-Matt, 2017-05-15:
### What are you trying to do?
Cross-compile C and C++ progs using docker.

### What did you expect to happen?
Programs should compile that have #include ".h's"

### What actually happened?
Programs only compile if I change them to ".c's"

### What hardware and software are you using (including version numbers)?

Latest docker version and image.

I was able to get the xv11lidar test program to compile via docker by changing the includes to .c.

I tried to use the dc.c example in the C language repo, but too many dependencies to change.
How do I make it pull in the needed code?
I was looking for a flag in [here](https://explainshell.com/explain/1/arm-linux-gnueabi-gcc-4.6), but I couldn't find what I was looking for there. Am I missing something simple?

dlech, 2017-05-15:
There are many ev3dev docker images. What is the exact name of the image?

Can you share your code?

Master-Matt, 2017-05-15:
It's the debian-jessie-cross image
This is my image id: d6ccb79c68c5
This is the error I get:
compiler@59b9d4b59d47:/src$ arm-linux-gnueabi-gcc -o main main.c
/tmp/ccn9SK0W.o: In function `main':
main.c:(.text+0xc8): undefined reference to `InitLaser'
main.c:(.text+0x14c): undefined reference to `CloseLaser'
/tmp/ccn9SK0W.o: In function `TextLaserData':
main.c:(.text+0x1c4): undefined reference to `ReadLaser'
/tmp/ccn9SK0W.o: In function `BinaryLaserData':
main.c:(.text+0x268): undefined reference to `ReadLaser'
collect2: error: ld returned 1 exit status

It looks like it's not linking the functions from the header files.

dlech, 2017-05-15:
I'm guessing you need to add a `-Isomething -lsomething`, but it is hard to say without seeing the code.

Master-Matt, 2017-05-15:
By code, do you mean the program I'm trying to compile?
I cloned [this repo](https://github.com/bmegli/xv11lidar-test) and am using it as my working directory.
The main in the command above is the main in that repo.

dlech, 2017-05-15:
> By code, do you mean the program I'm trying to compile?

Yes. It is hard to give advice if we don't know exactly what you are doing. :wink:

But, now that I know, I can tell you that you need to edit `Makefile` and replace the line...

    CC = gcc

...with...

    CC = arm-linux-gnueabi-gcc

Then run `make` and it should compile correctly (assuming you have undone any other changes you may have made).

Master-Matt, 2017-05-15:
Okay, That sort-of worked. I got this error after building:
compiler@cb229482b58d:/src$ make
make: Warning: File 'Makefile' has modification time 44620 s in the future
arm-linux-gnueabi-gcc -Wall -c main.c -o main.o
arm-linux-gnueabi-gcc  xv11lidar.o  main.o -Wall  -o xv11test
make: warning:  Clock skew detected.  Your build may be incomplete.
compiler@cb229482b58d:/src$ make
make: Warning: File 'Makefile' has modification time 44575 s in the future
make: Nothing to be done for 'default'.
make: warning:  Clock skew detected.  Your build may be incomplete.

dlech, 2017-05-15:
Interesting... apparently your clock in the docker image does not match the one on your host system.

try running `dpkg-reconfigure tzdata` in the docker image and select the correct timezone.

Master-Matt, 2017-05-15:
Okay weird. The timezone is set correctly, but the clock is about half an hour off.

dlech, 2017-05-15:
does `make` work now?

you can run 

    make clean
    make

to rebuild everything from scratch

Master-Matt, 2017-05-15:
This is what I get after Make clean:
compiler@17c312896034:/src$ make clean
make: Warning: File 'Makefile' has modification time 43414 s in the future
rm -f *.o
rm -f xv11test
make: warning:  Clock skew detected.  Your build may be incomplete.

dlech, 2017-05-15:
Are you using Docker Toolbox?

dlech, 2017-05-15:
as opposed to Docker for Windows

Master-Matt, 2017-05-15:
No I am using straight docker for both Windows and Mac

Master-Matt, 2017-05-15:
Well something just synced up. I installed ntp on the docker image a couple minutes ago. I wonder if that did it.

Master-Matt, 2017-05-15:
I'm going to close this now since I got it working. I need to get used to using makefiles.

closed 2017-05-15 by Master-Matt