ev3dev/ev3dev #886: How to stop the process bluetoothd?

jabrena, 2017-05-15:
I would like to save the maximum in the brick memory in the EV3 brick and I discovered that the bluetooth process is running but I didn´t enabled from Brickman.

How to disable this process in the best way?

```
root       331  0.0  3.2   5000  1920 ?        Ss   18:22   0:00 /usr/lib/bluetooth/bluetoothd
ntp        455  2.2  3.3   5940  1940 ?        Ss   18:23   6:15 /usr/sbin/ntpd -p /var/run/ntpd.pid -g -u 106:110
```

To stop the Brickman, what is the right way to stop? and how to start again?

``` 
robot@ev3dev:~$ ps aux
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         1  1.5  4.7  23916  2796 ?        Ss   18:22   4:16 /sbin/init splash
root         2  0.0  0.0      0     0 ?        S    18:22   0:00 [kthreadd]
root         3  0.6  0.0      0     0 ?        S    18:22   1:46 [ksoftirqd/0]
root         5  0.0  0.0      0     0 ?        S<   18:22   0:00 [kworker/0:0H]
root         7  1.4  0.0      0     0 ?        S    18:22   4:06 [rcu_preempt]
root         8  0.0  0.0      0     0 ?        S    18:22   0:00 [rcu_sched]
root         9  0.0  0.0      0     0 ?        S    18:22   0:00 [rcu_bh]
root        10  0.0  0.0      0     0 ?        S    18:22   0:00 [kdevtmpfs]
root        11  0.0  0.0      0     0 ?        S<   18:22   0:00 [netns]
root        12  0.0  0.0      0     0 ?        S<   18:22   0:02 [writeback]
root        13  0.0  0.0      0     0 ?        S<   18:22   0:00 [crypto]
root        14  0.0  0.0      0     0 ?        S<   18:22   0:00 [bioset]
root        15  0.0  0.0      0     0 ?        S<   18:22   0:00 [kblockd]
root        17  0.0  0.0      0     0 ?        S<   18:22   0:00 [rpciod]
root        18 22.9  0.0      0     0 ?        S    18:22  64:26 [kswapd0]
root        19  0.0  0.0      0     0 ?        S    18:22   0:00 [fsnotify_mark]
root        20  0.0  0.0      0     0 ?        S<   18:22   0:00 [nfsiod]
root        43  0.0  0.0      0     0 ?        S<   18:22   0:00 [bioset]
root        44  0.0  0.0      0     0 ?        S<   18:22   0:00 [bioset]
root        45  0.0  0.0      0     0 ?        S<   18:22   0:00 [bioset]
root        46  0.0  0.0      0     0 ?        S<   18:22   0:00 [bioset]
root        47  0.0  0.0      0     0 ?        S<   18:22   0:00 [bioset]
root        48  0.0  0.0      0     0 ?        S    18:22   0:00 [irq/56-spi_davi]
root        49  0.0  0.0      0     0 ?        S    18:22   0:00 [spi1]
root        50  0.0  0.0      0     0 ?        S    18:22   0:00 [irq/20-spi_davi]
root        51  6.2  0.0      0     0 ?        S    18:22  17:30 [spi0]
root        54  0.0  0.0      0     0 ?        S<   18:22   0:00 [deferwq]
root        56  0.0  0.0      0     0 ?        S<   18:22   0:00 [bioset]
root        57 12.2  0.0      0     0 ?        S    18:22  34:20 [mmcqd/0]
root        59  0.0  0.0      0     0 ?        S    18:22   0:01 [jbd2/mmcblk0p2-]
root        60  0.0  0.0      0     0 ?        S<   18:22   0:00 [ext4-rsv-conver]
root       112  0.0  0.0  10712     4 ?        Ss   18:22   0:01 /lib/systemd/systemd-udevd
root       142  0.0  0.0      0     0 ?        S<   18:22   0:00 [cfg80211]
root       188  0.0  0.0      0     0 ?        S<   18:22   0:00 [kworker/0:1H]
avahi      258  0.4  1.8   3844  1080 ?        Ss   18:22   1:10 avahi-daemon: running [ev3dev-2.local]
root       261  0.2  1.6   3820   988 ?        Ss   18:22   0:35 /lib/systemd/systemd-logind
root       266  0.0  0.0      0     0 ?        S<   18:22   0:00 [kworker/u3:0]
root       267  0.0  0.0      0     0 ?        S<   18:22   0:00 [hci0]
root       268  0.0  0.0      0     0 ?        S<   18:22   0:00 [hci0]
root       270  0.0  0.0      0     0 ?        S<   18:22   0:00 [kworker/u3:2]
message+   276  1.3  2.3   5808  1392 ?        Ss   18:22   3:39 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfi
avahi      283  0.0  0.0   3844     4 ?        S    18:22   0:00 avahi-daemon: chroot helper
root       289  0.0  0.0   2008     4 ?        S<   18:22   0:00 /usr/bin/hciattach -t 30 /dev/ttyS2 texas 2000000 flow nosleep
root       331  0.0  3.2   5000  1920 ?        Ss   18:22   0:00 /usr/lib/bluetooth/bluetoothd
root       332  2.3  4.1   7916  2428 ?        Ss   18:22   6:31 /usr/sbin/connmand -n
root       337  0.0  0.0   7916     4 ?        Ss   18:22   0:00 /usr/sbin/sshd -D
root       351  1.8  4.0   7032  2400 ?        Ss   18:23   5:17 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant
root       364  0.5  0.0      0     0 ?        S    18:23   1:36 [RTW_CMD_THREAD]
ntp        455  2.2  3.3   5940  1940 ?        Ss   18:23   6:15 /usr/sbin/ntpd -p /var/run/ntpd.pid -g -u 106:110
root       476  1.4  3.1  21024  1836 ?        Ss   18:24   4:02 /usr/sbin/nmbd -D
root       506  0.0  0.0  12172     4 ?        Ss   18:28   0:01 sshd: robot [priv]  
robot      512  0.0  0.0   4956     4 ?        Ss   18:28   0:00 /lib/systemd/systemd --user
robot      516  0.0  0.0  25148     4 ?        S    18:28   0:00 (sd-pam)         
robot      519  0.0  1.6  12172   968 ?        S    18:28   0:14 sshd: robot@pts/0   
robot      521  0.0  3.6   3564  2152 pts/0    Ss   18:28   0:00 -bash
root       637  0.0  0.0      0     0 ?        S    21:15   0:00 [kworker/u2:0]
root       665  0.0  0.0      0     0 ?        S    22:39   0:00 [kworker/0:1]
root       680  0.0  0.0      0     0 ?        S    22:54   0:00 [kworker/0:2]
root       681  0.0  0.0      0     0 ?        S    22:56   0:00 [kworker/u2:1]
root       682  0.0  0.0      0     0 ?        S    22:57   0:00 [kworker/u2:3]
root       688  0.1  9.6  12240  5636 ?        Ss   22:58   0:00 /lib/systemd/systemd-journald
root       702  0.2  0.0      0     0 ?        S    23:00   0:00 [kworker/0:0]
robot      729  0.0  2.9   3084  1748 pts/0    R+   23:02   0:00 ps aux
```

dlech, 2017-05-15:
sudo systemctl stop bluetooth
    sudo systemctl disable bluetooth

dlech, 2017-05-15:
same goes for `brickman`

dlech, 2017-05-15:
and any other service

jabrena, 2017-05-15:
Many thanks @dlech :D

jabrena, 2017-05-16:
Hi @dlech, 

is it possible to disable temporally the logging system?

```
/lib/systemd/systemd-journald
```

https://www.freedesktop.org/software/systemd/man/systemd-journald.service.html

I tried to kill the process or stop with `sudo systemctl stop systemd-journald` but the process always return as a Phoenix.

Juan Antonio

dlech, 2017-05-16:
Not sure. Ask google.

jabrena, 2017-05-16:
Yes, you had reason, i didn´t find but I found that if the process doesn´t store on disk, and the service is reloaded, the memory used by the service is reduced 60% +-

` sudo vi /etc/systemd/journald.conf`

```
Storage=none
#Storage=persistent
```

https://unix.stackexchange.com/questions/253203/how-to-tell-journald-to-re-read-its-configuration

Cheers

closed 2017-05-16 by jabrena