ev3dev/ev3dev #889: Brickman/Brickpi3 service failed

xyaa, 2017-05-21:
I have a RPi3 and a BrickPi3 with `ev3dev-jessie-rpi2-generic` but on boot brickpi3 and brickman.service failed

`dmesg|grep brickpi3`
```
[   22.581553] brickpi3 spi0.1: Mfg: Dexter Industries
[   22.584986] brickpi3 spi0.1: Board: BrickPi3
[   22.587917] brickpi3 spi0.1: HW: 3.2.1
[   22.623477] brickpi3 spi0.1: FW: 1.0.1
[   22.627389] brickpi3: probe of spi0.1 failed with error -22
```
`systemctl status brickman.service`
```
● brickman.service - Brick Manager
   Loaded: loaded (/lib/systemd/system/brickman.service; enabled)
   Active: failed (Result: exit-code) since sam. 2017-05-20 10:14:21 UTC; 24h ago
  Process: 349 ExecStart=/usr/sbin/brickman (code=exited, status=1/FAILURE)
 Main PID: 349 (code=exited, status=1/FAILURE)

mai 20 10:14:21 ev3dev systemd[1]: Started Brick Manager.
mai 20 10:14:21 ev3dev brickman[349]: ** (process:349): CRITICAL **: main.vala:39: Error setting driver
mai 20 10:14:21 ev3dev systemd[1]: brickman.service: main process exited, code=exited, status=1/FAILURE
mai 20 10:14:21 ev3dev systemd[1]: Unit brickman.service entered failed state.
```

**System info (from `ev3dev-sysinfo`)**                  
```                                                      
Image file:         ev3dev-jessie-rpi2-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-rpi2                
Board:              Raspberry Pi 3 Model B Rev 1.2       
Revision:           a02082                               
Brickman:           0.8.1                                
ev3devKit:          0.4.3                                
```                                                      

I think I missed something :/
Thx for help

dlech, 2017-05-21:
This sounds like two separate problems.

Brickman does not run on Raspberry Pi unless you have a secondary display, like PiStorms or an Adafruit TFT (which only works with older models of BrickPi, not BrickPi3). See #856.

I would suggest upgrading the BrickPi3 firmware. The current version is 1.4.1, I believe. Also, upgrade your ev3dev kernel or use the latest ev3dev snapshot image. http://www.ev3dev.org/news/2017/04/15/kernel-release-cycle-20/

xyaa, 2017-05-21:
Ah I understood that there was a difference between Brickman and BrickPi3. And I do not need brickman,  thanks :)

So it's a firmware, by the way, how do I install this new firmware? :S

dlech, 2017-05-22:
Unfortunately, there is not a way to upgrade the firmware using ev3dev. You will need to use the Dexter Industries Raspbian for Robots to upgrade the firmware.

xyaa, 2017-06-04:
I succeeded in installing the latest version of firmware :+1:

Thx for help :)

closed 2017-06-04 by xyaa