ev3dev/ev3dev #891: Executing a python file from Brickman File Browser

nmerekat, 2017-05-26:
### I'm new to python and to ev3dev. I have transferred a file, motor3.py, into the home/robot folder, using FileZilla. motor3.py is pasted below:

```
#!/usr/bin/env python

from ev3dev.auto import *
import time

def main():
	m = Motor(OUTPUT_A)
	m.run_timed(time_sp=3000, speed_sp=500)

if __name__ == '__main__':
	main()
```

Right now, when I establish an ssh connection, I can execute the program from Terminal. I first enter the line:

`robot@ev3dev:~$ chmod u+x motor3.py`

or 

`robot@ev3dev:~$ chmod +x motor3.py`

Ive tried both, and had success running motor3.py from Terminal using the line:

`robot@ev3dev:~$ python3 motor3.py`

 But I need to be able to execute this motor test program from the Brickman File Browser, and I have not been able to. In the File Browser, motor3.py does not have an * after it, which means it isn't executable? I made sure to set the permissions when in FileZilla, but I still do not see an asterisk after motor3.py

I've been reading through the issues page and I've included the "shebang" and wrapped my code with the if __name__ ... statement, but am I missing a step? How can I execute the above code directly from File Browser?

### What hardware and software are you using (including version numbers)?

dlech, 2017-05-26:
#!/usr/bin/env python

should be 

    #!/usr/bin/env python3

And if you are using Windows to edit the file, make sure you save it using Unix line endings, not Windows line endings.

nmerekat, 2017-05-26:
Thank you for the help, dlech! When I changed that line, the asterisk appeared and I was able to execute the code from File Browser. I am using a Mac, so I'm fine for the line endings. I can't believe I missed that... Again, thank you for spotting that!

closed 2017-05-26 by nmerekat