ev3dev/ev3dev #892: Doubt with the automatic process: apt.systemd.daily

jabrena, 2017-05-28:
Hi mates,

Recently, I noticed that in one EV3 Brick which I don´t turn off in some time:

```
robot@ev3dev:~$ uptime
21:27:54 up 12 days,  2:07,  1 user,  load average: 0.03, 0.16, 0.52
```
In some moments, the performance decreased and I don´t know the reason. Today I discovered the cause, take a look this processes:

```
robot@ev3dev:~$ ps aux
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root     15244  0.0  0.0   1848     0 ?        Ss   18:00   0:00 /bin/sh /usr/lib/apt/apt.systemd.daily update
root     15268 43.6 12.4  55288  7292 ?        R    18:00   6:16 apt-get check -qq
```

If you observe the usage of CPU and the memory is high.

- Why `apt` is updated every day?
- How to disable this service?

Many thanks in advance.

Juan Antonio

**Technical info:**

```
Image file:         ev3dev-jessie-ev3-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2017-05-29:
This is not something that was enabled on purpose in ev3dev, so it must be something that is installed by default in debian. I don't know how to disable it.

jabrena, 2017-05-29:
@dlech, I found the solution here:
https://unix.stackexchange.com/questions/315502/how-to-disable-apt-daily-service-on-ubuntu-cloud-vm-image

Cheers

Juan Antonio

dlech, 2017-05-29:
Cool. This sounds like something we should disable by default.

jabrena, 2017-05-29:
Yes, for Raspi, it is possible to maintain the default configuration but for EV3, the process reduce the performance a lot. When the system has 12-17Mb of free memory, the system has a good performance.

dlech, 2017-08-31:
Should be fixed in ev3dev-stretch now.

closed 2017-08-31 by dlech