ev3dev/ev3dev #893: runnig Hello world in a container

ValHaa, 2017-05-29:
### What are you trying to do?
Following the Instructions in the tutorial "Using Docker to Cross-Compile" 

using ./hello to run the program 

### What did you expect to happen?
"Hello world!" should be outputted 

### What actually happened?
error occures: "cannot execute binary file: Exec format error"

on the EV3 the program is running but not in the Container

### What hardware and software are you using (including version numbers)?

$ docker version
Client:
 Version:      17.04.0-ce
 API version:  1.28
 Go version:   go1.7.5
 Git commit:   4845c56
 Built:        Wed Apr  5 23:33:17 2017
 OS/Arch:      windows/amd64

Server:
 Version:      17.05.0-ce
 API version:  1.29 (minimum version 1.12)
 Go version:   go1.7.5
 Git commit:   89658be
 Built:        Thu May  4 21:43:09 2017
 OS/Arch:      linux/amd64
 Experimental: false

Using a win 7 64Bit system with docker toolbox

dlech, 2017-05-29:
It is probably not all that helpful to run your program in Docker since it can't simulate EV3 hardware.

But, if you really want to make it work, have a look at this: https://github.com/moul/docker-binfmt-register

Docker for Windows (which only runs on Win10 Pro) and Docker for Mac will "just work", but since you are using Docker Toolbox, it is not set up to do this out of the box.

The ev3dev cross-compiling docker images already have `qemu-user-static` present, so you don't have to do everything that the example that I linked does. I don't have Docker Toolbox installed anywhere to test but you can try running the following in your docker container (as root)...

If the directory `/proc/sys/fs/binfmt_misc` already exists, then you can skip this, otherwise...

     mount binfmt_misc -t binfmt_misc /proc/sys/fs/binfmt_misc

Then (note that `local` is missing from the path to `qemu-arm-static` compared to the example)...

    echo ':arm:M::\x7fELF\x01\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x28\x00:\xff\xff\xff\xff\xff\xff\xff\x00\xff\xff\xff\xff\xff\xff\xff\xff\xfe\xff\xff\xff:/usr/bin/qemu-arm-static:' > /proc/sys/fs/binfmt_misc/register

ValHaa, 2017-05-30:
thanks for the quick help :) i'll try this.

closed 2017-05-30 by ValHaa