ev3dev/ev3dev #894: Lego Mindstorms EV3 stuck at boot...

NV-GCK, 2017-05-31:
### What are you trying to do?
Run ev3dev on Lego Mindstorm EV3 bricks.

### What did you expect to happen?
I have several successfully booting EV3 units that are able to run ev3dev. 

### What actually happened?
I have two quarantined EV3 units that have both 'crashed' after trying to run bad python scripts. Said quarantined units are no longer able to run ev3dev. The units get stuck during the boot sequence (please refer to attached image.) 
- I have tried reburning the ev3dev image onto the micro SD cards. 
- I have tried changing micro SD cards from 32 GB to 16 GB. 
- I have tried changing FW version from 1.07H to 1.09H to 1.09E to 1.06H to no avail. 

### What hardware and software are you using (including version numbers)?
I am running Lego Mindstorms EV3.
Quarantined unit 1 FW was changed as described above.
Quarantined unit 2 FW is at 1.07H.

[image]

WasabiFan, 2017-05-31:
Although I likely can't suggest a fix for your immediate issue, I'll point out that you should be able to boot into the standard EV3 software by simply unplugging the SD card and turning on your EV3. That should at least indicate that there's nothing wrong with the hardware. You could try testing the normal SD card functionality of the LEGO software to demonstrate that the SD card reader is still fully functional.

Did you make sure to disconnect all motors, sensors, dongles and the USB client cable before booting? Sometimes those devices can cause issues, although I wouldn't expect it.

NV-GCK, 2017-05-31:
Hi WasabiFan, 
thank you for the prompt response. As per the ev3dev instructions, nothing (motors, sensors, USB cables/dongles etc.) was connected to the brick while we tried to boot up ev3dev. I did boot up the standard EV3 software, and also checked that I am able to view files in the SD card via the standard software file browser. However, I have not tried executing a program on either quarantine unit. I will try that next both connected to by PC and via a program saved on the SD card and will report back to this thread.

dlech, 2017-05-31:
FYI, it is actually getting stuck at `Waiting for root device /dev/mmcblk0p2...`. The random part is a background task that just finishes later.

It is odd that this seems to be related to the EV3 itself and not to an SD card. Can you try on of the stretch images from [here](https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/) and see if they get stuck too?

NV-GCK, 2017-05-31:
Thank you dlech, I will most definitely try the images you suggested and get back to you ASAP.

NV-GCK, 2017-05-31:
Hi dlech,
We tried the following images:

2016-08-13/  13-Aug-2016 05:31    -
2016-08-16/  16-Aug-2016 18:57    -
2016-08-18/  18-Aug-2016 00:28    -
2016-09-04/  04-Sep-2016 23:56    -
2016-09-22/  22-Sep-2016 19:09    -
2016-09-30/  30-Sep-2016 21:22    -
2017-01-17/  17-Jan-2017 05:29    -
2017-05-15/  15-May-2017 22:19    -

All of them hung in the exact same manner as described in the initial issue.

dlech, 2017-06-01:
Just to be sure, you tried a "stretch" image and not a "jessie" image? Just testing the most recent one is enough.

Do you still have the "bad python script"? I'm curious what it might have done to trigger this because I don't have any good guesses as to what might be going on.

NV-GCK, 2017-06-01:
dlech I am ever so sorry to have wasted your time on this. I have good reason to believe that fault was because of faulty card readers on the EV3 bricks. Wiggling the SD card in quarantine unit 2 caused it to boot intermittently. I have since reached out directly to Lego to get replacement bricks. Please go ahead and close this ticket, and ever so sorry again to have wasted your time. My bad.

closed 2017-06-01 by NV-GCK

WasabiFan, 2017-06-01:
That's really no problem -- we're happy to help! Hardware issues can be tough to track down.

Do you have any idea what caused the damage? It might be worth documenting the symptoms and what may have caused it.

NV-GCK, 2017-06-02:
I suspect a manufacturing defect but could not say for sure. I'm just delighted that Lego support is mailing me fresh bricks to replace the errant ones.

defue, 2018-09-28:
Experienced the same problem with a new educational EV3 brick. Tried 3 different Kingston microSD cards of various capacity and class and none of them worked. Tried Sundisk Ultra microSDXC 64GB and it worked.

dlech, 2018-09-28:
I would not be surprised if you find strange things happening with a 64GB card as you use it. The hardware is only capable of SDHC, which is 32GB max.

defue, 2018-09-28:
So far everything works fine. I was able update the ev3dev to the latest version without issues. Wifi dongle and visual studio code integration also works fine. I'll comment there if I face with any issue related to the sd card.

Timacrolsh, 2025-04-20:
heeeeelp please same problem