ev3dev/ev3dev #896: EVB crash

dwalton76, 2017-06-03:
I am running my rubiks cube solving robot and am hitting what I think is a kernel panic, the EVB just reboots while in the middle of moving a large motor motor.  I am running 

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-bone-generic-2017-02-11
Kernel version:     4.4.62-ti-rt-r102-20-ev3dev-bb.org
Board:              TI AM335x BeagleBone Black
Revision:           00c0
Brickman:           0.8.1
ev3devKit:          0.4.3
```

I do not see anything suspect in the logs after it reboots. How do I get console on the EVB? Or how else can I get some logs to get some ideas on what is happening?

This doesn't happen everytime but does happen fairly often (say 1 in every 3 rubiks cubes).

dlech, 2017-06-03:
If the kernel is crashing, the only way to see anything will be on the serial console.

http://blog.tonywall.co/2013/11/beaglebone-black-serial-debug-connection/

You will need a USB/Serial adapter. The EVB has a place where you can solder on a 6-pin female header on the bottom and a 3-pin male header on top to pass through UART0. If you don't have these, if you have some old jumpers, like you used to set IRQs with on old computers, you can use them to get access to the pins on the BeagleBone itself.

[image]

dwalton76, 2017-06-04:
Thanks....hmm it is going to be fun to get this on there with enough room to still get the EVB board on.

dlech, 2017-06-04:
It's not so bad. :wink:

[image]

dwalton76, 2017-06-09:
Good news and bad news.  The good news is I got console access, the bad news is it crashes but I get zero output to indicate what went wrong. It crashed right after the `route_to_quarter_turn 210`

```Writing PNG image to '/tmp/rubiks-side-F.png'.
2017-06-09 01:48:37,903 crane_cuber.py     INFO: elevate() from 0 to 3 took 535ms
2017-06-09 01:48:37,905 crane_cuber.py     INFO: round_to_quarter_turn 210 is already a multiple of 210

U-Boot SPL 2016.03-00001-g148e520 (Jun 06 2016 - 11:27:44)
Trying to boot from MMC
bad magic

U-Boot 2016.03-00001-g148e520 (Jun 06 2016 - 11:27:44 -0500), Build: jenkins-github_Bootloader-Builder-395

       Watchdog enabled
^MI2C:   ready
^MDRAM:  512 MiB
^MReset Source: Power-on reset has occurred.
^MMMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
ESC[37C^MUsing default environment
ESC[25C
Net:   <ethaddr> not set. Validating first E-fuse MAC
ESC[53C^Mcpsw, usb_ether
ESC[15C^MPress SPACE to abort autoboot in 2 seconds
```
Thoughts on what else to try?

dwalton76, 2017-06-09:
For archive purposes on how to get console on EVB I bought:

- https://www.amazon.com/gp/product/B00QT7LQ88/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1
- https://www.amazon.com/gp/product/B00PBZMN7C/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

and followed http://www.dummies.com/computers/beaglebone/how-to-connect-the-beaglebone-black-via-serial-over-usb/

dlech, 2017-06-09:
Are kernel console messages printing on the serial terminal at all (e.g. during boot)?

Did you have to pull the power to reboot? If so, this could be a deadlock rather than a crash. There are some kernel options you can enable to troubleshoot deadlocks if you want to compile your own kernel.

Also, try one of the stretch snapshot images. They have a 4.9 kernel, so may behave differently.

dwalton76, 2017-06-09:
> Are kernel console messages printing on the serial terminal at all (e.g. during boot)?

yep, I see the bootup output

> Did you have to pull the power to reboot? If so, this could be a deadlock rather than a crash. There are some kernel options you can enable to troubleshoot deadlocks if you want to compile your own kernel.

Na it reboots on its own

> Also, try one of the stretch snapshot images. They have a 4.9 kernel, so may behave differently.

ok will give that a try

dwalton76, 2017-06-09:
I installed https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2017-05-15/snapshot-ev3dev-stretch-bone-generic-2017-05-15.img.xz but it is not happy, when this runs

```
    def init_motors(self):

        for x in self.motors:
            if not x.connected:
                log.error("%s is not connected" % x) 
                sys.exit(1)
            x.reset()

        # Lower all the way down, then raise a bit, then lower back down.
        # We do this to make sure it is in the same starting spot each time.
        log.info("Initialize elevator %s" % self.elevator)
        self.elevator.run_forever(speed_sp=40, stop_action='brake')
        self.elevator.wait_until('running')
        self.elevator.wait_until('stalled')
        self.elevator.run_to_rel_pos(speed_sp=200, position_sp=-50)
        self.elevator.wait_while('running')
        self.elevator.run_forever(speed_sp=40, stop_action='hold')
        self.elevator.wait_until('running')
        self.elevator.wait_until('stalled')
        self.elevator.stop()
        self.elevator.reset()
        self.elevator.stop(stop_action='brake')

        if self.shutdown_event.is_set():
            return

        log.info("Initialize flipper %s" % self.flipper)
        self.flipper.run_forever(speed_sp=150, stop_action='hold')
        self.flipper.wait_until('running')
        self.flipper.wait_until('stalled')
        self.flipper.stop()
        self.flipper.reset()
        self.flipper.stop(stop_action='hold')
        self.flipper_at_init = True 

        if self.shutdown_event.is_set():
            return
```

The motors are really jittery
https://www.youtube.com/watch?v=ihOFbrRN9S8

dlech, 2017-06-09:
> Are kernel console messages printing on the serial terminal at all (e.g. during boot)?
> 
> yep, I see the bootup output

Just to be extra clear, you also see messages after boot is complete, for example if you plug or unplug a sensor?

If this is the case, then I'm not really sure how to troubleshoot.

> The motors are really jittery

At least it doesn't crash? 😆 

I guess we better open a new issue for this problem.

dwalton76, 2017-06-09:
> Just to be extra clear, you also see messages after boot is complete, for example if you plug or unplug a sensor?

yep...let me see if I can find a beaglebone black to borrow to rule out bad hardware

I'll file a new issue for the jittery motors

dwalton76, 2017-08-13:
I upgraded to
```
robot@ev3dev[~]# ev3dev-sysinfo
Image file:         ev3dev-stretch-bone-generic-2017-08-11
Kernel version:     4.9.41-ev3dev-1.4.0-bb.org
/usr/bin/ev3dev-sysinfo: line 68: warning: command substitution: ignored null byte in input
Board:              TI AM335x BeagleBone Black
Revision:           000c
Brickman:           0.9.0
ev3devKit:          0.5.1
robot@ev3dev[~]# 
```

and am still hitting the issue. This is on a different beaglebone black from when I first filed 896. I am starting to wonder if maybe my EVB has gone bad since I am the only one that is hitting this issue and it just hard crashes, I get nothing on the console.

dlech, 2017-08-13:
Can you post the full source code you are using so I can try to reproduce?

dlech, 2017-08-13:
Also, have you tried a different SD card?

dwalton76, 2017-08-14:
source is https://github.com/dwalton76/lego-crane-cuber

I'll see if I have another SD card...it is 128G to hold all of the rubiks cube tables so I may not have another one that big

dlech, 2017-08-14:
You wouldn't happen to have a simpler example to reproduce the problem would you? Otherwise, I'll try to figure it out.

dwalton76, 2017-08-14:
I've only seen it crash with this one robot...but then again it is the only robot I've touched in the last 9 months so not sure that helps much.  When it starts up it will init a few motors by turning them until they stall and then it waits for you to press the touch sensor.  At that point the motors start turning and that is when it crashes.

dwalton76, 2017-08-14:
You'll want large motors in A, C, and D and a medium motor in B

dlech, 2017-08-14:
I figured that out. :wink: Any tips on installing the development branch of ev3dev-lang-python? It looks like you are using a method that hasn't been packaged yet.

dwalton76, 2017-08-14:
ah yeah forgot about that part :) This should work
```
$ git clone https://github.com/rhempel/ev3dev-lang-python.git
$ cd ev3dev-lang-python
$ sudo python3 setup.py install
```

dlech, 2017-08-14:
Interesting... I'm seeing your webcam error now.

```
dmesg
...
[93952.121752] musb_ep_program 921: broken !rx_reinit, ep14 csr 0003
[93952.127907] musb_host_rx 1970: Rx interrupt with no errors or packet!
```

Apparently, your program does something to mess up the kernel (via the motor or sensor drivers most likely).

dlech, 2017-08-14:
Hmmm... after reboot, different USB errors...

```
[  274.002622] musb-hdrc musb-hdrc.1: VBUS_ERROR in a_host (89, <AValid), retry #1, port1 00000107
[  274.176122] usb 1-1: USB disconnect, device number 2
[  274.188925] musb-hdrc musb-hdrc.1: VBUS_ERROR in a_host (89, <AValid), retry #2, port1 00000105
[  274.670186] usb 1-1: new full-speed USB device number 4 using musb-hdrc
[  274.687302] musb-hdrc musb-hdrc.1: VBUS_ERROR in a_host (89, <AValid), retry #1, port1 00020115
[  274.802374] musb-hdrc musb-hdrc.1: VBUS_ERROR in a_wait_vrise (89, <AValid), retry #1, port1 00020515
[  274.918010] usb 1-1: new high-speed USB device number 5 using musb-hdrc
[  274.981177] musb-hdrc musb-hdrc.1: VBUS_ERROR in a_host (89, <AValid), retry #1, port1 00030105
[  275.082564] musb-hdrc musb-hdrc.1: VBUS_ERROR in a_wait_vrise (89, <AValid), retry #2, port1 00030105
[  275.171629] musb-hdrc musb-hdrc.1: VBUS_ERROR in a_wait_vrise (89, <AValid), retry #3, port1 00030105
```

Maybe the EVB doesn't have clean enough power for USB when the motors are running?

dlech, 2017-08-14:
Or maybe I need a bigger power supply?

dlech, 2017-08-14:
my camera doesn't work anymore :cry:

dlech, 2017-08-14:
still works on EV3 :relieved:

dwalton76, 2017-08-14:
weird...I have a powered USB hub at work I'll bring it home and see if that makes a difference.  At least I am not the only one seeing the camera issue now...I thought I was going crazy :)  I used EVB with this robot several months ago but it was before I had a lego webcam, I'll dig up my old webcam and see if it makes a difference.

As for what my program is doing that is causing issues, I wonder if it is the use of this `wait()` function somehow.
https://github.com/rhempel/ev3dev-lang-python/blob/develop/ev3dev/core.py#L885

I say that because it is a fairly recent (within the past 5 or 6 months) addition to ev3dev-lang-python and it is the only somewhat unusual thing I can think of off the top of my head that I do with the motors.  For the touch sensor I have a separate thread that I spin up to monitor the touch sensor, that isn't that out of the ordinary though.

dwalton76, 2017-08-16:
I'll see if I can narrow down what part of my program is the trigger for the crash

dlech, 2017-08-17:
I have a suspicion the crash is because of the USB webcam issue. It is sending bad USB packets, so that could certainly crash the kernel.

I am starting to wonder if this has damaged our webcams. Before I ran your program, my LEGO webcam worked find on BeagleBone. But now it no longer works on any BeagleBone. It still works on EV3 and RasberryPi, which is a relief. But, I have tried it on several BeagleBones (Green that has run the fatal program and Black that has not run the fatal program) and it no longer works on either. I also have another Logitech webcam that works on the same BeagleBones. So, I can conclude that the BeagleBone hardware is not damaged.

The error I'm seeing now in `dmesg` is

```
[   67.526400] STV06xx 1-1:1.0: URB packet length is smaller than the specified chunk length
[   67.631191] STV06xx 1-1:1.0: URB packet length is smaller than the specified chunk length
```

I guess the only thing to do now is find a virgin LEGO webcam and see if it still works on one of these BeagleBones.

dwalton76, 2017-08-17:
I've seen the crash when using a non-lego webcam but it did not crash nearly as often. I don't have hard data just gut feel. 

Interesting though..I wonder why ev3 doesn't have the same issue

dwalton76, 2017-09-04:
Getting back to this...I tried with a non-lego webcam and hit the same issue.  This is my webcam
```
robot@ev3dev[~]# lsusb
Bus 001 Device 004: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Bus 001 Device 003: ID 046d:0990 Logitech, Inc. QuickCam Pro 9000
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
robot@ev3dev[~]# 
```

So this does not appear to be webcam specific although it could be logitech specific since the lego webcams are `Logitech, Inc. QuickCam Web`.

dwalton76, 2017-09-04:
@dlech do you have a non-logitech webcam you can try? I do not

dlech, 2017-09-04:
I do, but I don't see how that helps.

Does the LEGO camera work on BeagleBone using the official BeagleBone SD card image?

dwalton76, 2017-09-04:
I just want to make sure this isn't somehow something logitech specific since that is one variable we have not yet changed.

I'll try the BeagleBone stretch image and see what I get.

dwalton76, 2017-09-09:
ok with just a beaglebone black (no usb hub, no EVB board) running https://debian.beagleboard.org/images/bone-debian-9.1-lxqt-armhf-2017-08-31-4gb.img.xz

I am unable to capture an image via fswebcam, I tried with both the lego webcam and another logitech webcam that I have.  Both hit this error:
```
root@beaglebone:/home/debian# fswebcam --config fswebcam.conf 
--- Opening /dev/video0...
Trying source module v4l2...
/dev/video0 opened.
No input was specified, using the first.
Adjusting resolution from 352x240 to 356x292.
[  889.333590] gspca_main: set alt 1 err -110
Error starting stream.
VIDIOC_STREAMON: Connection timed out
Unable to use mmap. Using read instead.
--- Capturing frame...
Timed out waiting for frame!
No frames captured.
root@beaglebone:/home/debian# 
```

This is weird

dlech, 2017-09-09:
That's kind of what I was suspecting. Maybe the BeagleBone just has buggy USB?

You could search around to see if anyone has reported a similar issue. You could also try older beaglebone images until you find one that works (if there is one). If you do find one, then we can probably figure out the change that introduced the problem.

dwalton76, 2017-09-09:
I compiled the latest fswebcam from github but it didn't make a difference :(

How about we leave this issue open for another month and I'll see if I can get time to bisect through the older beaglebone images to figure out when this broke.  If I don't get to it in a month though I'm ok if we close this and move on.  I think you are I are the only two people running ev3dev on an EVB anyway and we've already spent a lot of time on this one :)

dwalton76, 2018-02-21:
I offloaded the taking of pics via fswebcam to another machine (EVB connects to it via a tcp socket, says "take a pic" and gets the png file back as the reply) and even then my EVB still crashes :( Who knows maybe I have bad HW and it cannot be replaced so no easy way to narrow that down.

I am going to close this.  I think @dlech and I are the only two people using ev3dev with an EVB anyway...maybe it is time to drop EVB support?

closed 2018-02-21 by dwalton76