ev3dev/ev3dev #897: Is it possible to upgrade BrickPi3 if you have EV3Dev installed?

jabrena, 2017-06-04:
Hi,

This morning I read this news and the changelog for Raspi 2:
http://www.ev3dev.org/news/2017/04/15/kernel-release-cycle-20/
https://github.com/ev3dev/ev3dev-kpkg/blob/a90ed59ed7ece41c46c564653b4acb4a1dd4640c/ev3dev-ev3/changelog

And I would like to know if the process to upgrade the firmware for BrickPi is possible to run on EV3Dev.
https://github.com/DexterInd/BrickPi3/blob/master/Firmware/README.md

``` 
* Update to lego driver level 20:
    * Update for BrickPi3 firmware 1.4.x (thanks @sanjayseshan)
    * Fix stopping before run-to-*-pos is complete when stop_action is hold
```

With this update on BrickPi3 and using the latest EV3Dev version, it is possible to use the other stop actions ("hold", "coast" & "brake")?

dlech, 2017-06-04:
https://github.com/ev3dev/ev3dev/issues/889#issuecomment-302974273

As far as I can remember, you should be able to use all stop actions on all platforms.

jabrena, 2017-06-04:
Many thanks @dlech,

I will test this week.

Cheers

jabrena, 2017-06-05:
Good morning @dlech, 

I tested with BrickPi+ and PiStorms V2 with latest stable image:
http://www.ev3dev.org/news/2017/02/11/ev3dev-jessie-2017-02-11-release/

and you have reason, Stop modes run nice in every platform.

Cheers

Juan Antonio

jabrena, 2017-06-18:
Many thanks @dlech 

I could upgrade the firmware and add the support for BrickPi3.

Tomorrow, I will open a issue about some mode for GyroSensor.
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/329
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/188

Cheers

Juan Antonio

closed 2017-06-18 by jabrena